zephyrproject-rtos/zephyr #35721: Atmel sam0 Async and/or DMA may not work

nandojve, 2021-05-27:
**Describe the bug**
The #24873 made an important enhancement on Zephyr. Such big change may create eventual issues. It was detected that  Async and/or DMA features from Atmel SAM0 devices stopped to work after #24873. It is not exactly clear the why and require more investigation.

**To Reproduce**
Try run `tests/drivers/uart/uart_async_api` with `atsamr21_xpro` board, for instance.

**Expected behavior**
Both Async and DMA should work for all SAM0 devices. The `platform_exclude ` should be updated or removed on`testcase.yaml`, on all tests that uses that features for SAM0 devices to allow run those tests on the CI.

**Impact**
Features are not working.

CC: @galak @arvinf @stephanosio

galak, 2021-06-01:
@nandojve the issue that caused exclusion of atmel sam0 boards from `tests/drivers/uart/uart_async_api` was the fact that with uart async enabled it has to be enabled on ALL uart's and I don't think that is the case right now for how the sam0 boards are setup.

Do we know if we configure a board with only a single uart and dma if the test fails?

nandojve, 2021-06-09:
> @nandojve the issue that caused exclusion of atmel sam0 boards from tests/drivers/uart/uart_async_api was the fact that with uart async enabled it has to be enabled on ALL uart's and I don't think that is the case right now for how the sam0 boards are setup.

Yes sure, my intention is not forget to re-enable the test case after fix the issue.

> Do we know if we configure a board with only a single uart and dma if the test fails?

I added missing DMA fields at SERCOM0, as:
```
&sercom0 {
	...
	dmas = <&dmac 0 1>, <&dmac 1 2>;
	dma-names = "rx", "tx";
};
```
Then I build/flash hello world with `UART_ASYNC_API=y`
The result was: no characters on terminal.

```
Reading symbols from /home/gfbudke/zephyros/mainline/zephyr/build/zephyr/zephyr.elf...
Remote debugging using :3333
Info : accepting 'gdb' connection on tcp/3333
Debugger attaching: halting execution
Info : SWD DPIDR 0x0bc11477
Error: at91samr21g18a.cpu -- clearing lockup after double fault
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xfffffffe msp: 0x200058e0
force hard breakpoints
Info : SAMD MCU: SAMR21G18A (256KB Flash, 32KB RAM)
Warn : Prefer GDB command "target extended-remote 3333" instead of "target remote 3333"
Info : SWD DPIDR 0x0bc11477
Error: Failed to read memory at 0xfffffffe
0xfffffffe in ?? ()
Info : SWD DPIDR 0x0bc11477
Error: at91samr21g18a.cpu -- clearing lockup after double fault
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xfffffffe msp: 0x200058e0
Loading section rom_start, size 0xb0 lma 0x0
Loading section text, size 0x99e0 lma 0xb0
Loading section .ARM.exidx, size 0x8 lma 0x9a90
Loading section initlevel, size 0xa8 lma 0x9a98
Loading section devices, size 0x138 lma 0x9b40
Loading section sw_isr_table, size 0xe0 lma 0x9c78
Loading section log_const_sections, size 0x38 lma 0x9d58
Loading section log_backends_sections, size 0x20 lma 0x9d90
Loading section shell_area, size 0x30 lma 0x9db0
Loading section shell_root_cmds_sections, size 0x40 lma 0x9de0
Loading section device_handles, size 0x82 lma 0x9e20
Loading section rodata, size 0x1970 lma 0x9ea4
Loading section datas, size 0x148 lma 0xb814
Loading section log_dynamic_sections, size 0x1c lma 0xb95c
Loading section k_msgq_area, size 0x30 lma 0xb978
Loading section k_sem_area, size 0x18 lma 0xb9a8
Info : Padding image section 0 at 0x00009ea2 with 2 bytes
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (3883 ms). Workaround: increase "set remotetimeout" in GDB
Info : SWD DPIDR 0x0bc11477
Error: at91samr21g18a.cpu -- clearing lockup after double fault
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xfffffffe msp: 0x200058e0
Start address 0x000041a0, load size 47550
Transfer rate: 9 KB/sec, 2641 bytes/write.

(gdb) monitor reset
Info : SWD DPIDR 0x0bc11477
SWD DPIDR 0x0bc11477
Error: at91samr21g18a.cpu -- clearing lockup after double fault
at91samr21g18a.cpu -- clearing lockup after double fault
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xfffffffe msp: 0x200058e0
target halted due to debug-request, current mode: Handler HardFault
xPSR: 0x01000003 pc: 0xfffffffe msp: 0x200058e0
[... 61 lines cut ...]
TX_DONE timeout
 FAIL - test_long_buffers in 0.217 seconds
===================================================================
START - test_write_abort_setup
 PASS - test_write_abort_setup in 0.1 seconds
===================================================================
START - test_write_abort

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:365: test_write_abort: (k_sem_take(&tx_done, K_MSEC(100)) not equal to 0)
TX_DONE timeout
 FAIL - test_write_abort in 0.117 seconds
===================================================================
START - test_forever_timeout_setup
 PASS - test_forever_timeout_setup in 0.1 seconds
===================================================================
START - test_forever_timeout

    Assertion failed at WEST_TOPDIR/zephyr/tests/drivers/uart/uart_async_api/src/test_uart_async.c:433: test_forever_timeout: (k_sem_take(&tx_done, K_MSEC(100)) not equal to 0)
TX_DONE timeout
 FAIL - test_forever_timeout in 1.118 seconds
===================================================================
Test suite uart_async_test failed.
===================================================================
PROJECT EXECUTION FAILED
```

cfriedt, 2021-07-27:
@nandojve - it appears that you might have a fix in mind for this. Is that the case?

nandojve, 2021-07-27:
Hi @cfriedt ,

No, I don't have a solution. I simple noted regression when I was working on pinctrl.
I'm not an expert on SAM0, not sure if I can work on this.

nandojve, 2021-09-18:
@galak , see #38085 which may fix DMA/ASYNC issues with SAM0.

cfriedt, 2021-09-19:
@rocky134 - I think the `get_maintainers` should go away if you rebase. vaguely recall that from a separate pr.

rocky134, 2021-09-19:
@cfriedt rebase what? PR #38085 up to the current main head?

closed 2021-09-19 by cfriedt