zephyrproject-rtos/zephyr #35760: tests: subsys: canbus: isotp: implementation: fails on nucleo_g474re

ABOSTM, 2021-05-28:
**Describe the bug**
tests/subsys/canbus/isotp/implementation/  fails on nucleo_g474re.

This test is failed on main branch.
But even when reverting to original introduction of FDCAN on nucleo_g474re, 
#31061 with sha1 3b2fd1ca0d951ee06279afca0a8f565459d5550b
it is still failed

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p auto -b nucleo_g474re tests/subsys/canbus/isotp/implementation/
2. west flash
3.  See error

**Expected behavior**
test passed

**Logs and console output**
```
*** Booting Zephyr OS build v2.6.0-rc2-39-g94e8c2eb81f3  ***
Running test suite isotp
===================================================================
START - test_bind_unbind
 PASS - test_bind_unbind in 0.905 seconds
===================================================================
START - test_send_receive_net_sf
 PASS - test_send_receive_net_sf in 0.6 seconds
===================================================================
START - test_send_receive_net_blocks
 PASS - test_send_receive_net_blocks in 1.4 seconds
===================================================================
START - test_send_receive_net_single_blocks

E: Reception of next FC has timed out
    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:296: test_send_receive_net_single_blocks: (ret not equal to 0)
recv returned -14

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:61: send_complette_cb: (error_nr not equal to ISOTP_N_OK)
Sending failed (-2)
 FAIL - test_send_receive_net_single_blocks in 1.98 seconds
===================================================================
START - test_send_receive_sf

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:97: get_sf: (ret not equal to 1)
recv returned -14
 FAIL - test_send_receive_sf in 1.14 seconds
===================================================================
START - test_send_receive_blocks
E: No buffer for FF left

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:268: test_send_receive_blocks: (ret not equal to 0)
Binding failed (-11)
 FAIL - test_send_receive_blocks in 0.18 seconds
===================================================================
START - test_send_receive_single_block
E: No buffer for FF left

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:320: test_send_receive_single_block: (ret not equal to 0)
Binding failed (-11)
 FAIL - test_send_receive_single_block in 0.18 seconds
===================================================================
START - test_buffer_allocation
E: No buffer for FF left

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:393: test_buffer_allocation: (ret not equal to 0)
Binding failed (-11)
 FAIL - test_buffer_allocation in 0.18 seconds
===================================================================
START - test_buffer_allocation_wait
E: No buffer for FF left

    Assertion failed at WEST_TOPDIR/zephyr/tests/subsys/canbus/isotp/implementation/src/main.c:409: test_buffer_allocation_wait: (ret not equal to 0)
Binding failed (-11)
 FAIL - test_buffer_allocation_wait in 0.18 seconds
===================================================================
Test suite isotp failed.
===================================================================
PROJECT EXECUTION FAILED
```

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain gnuarmemb
 - Commit SHA 94e8c2eb81f390a9226a3b70fdb69a002642f274

ABOSTM, 2021-05-28:
@alexanderwachter Is there something I miss to pass the test, any loopback wire, tranceiver,  ...?
(I am not CAN expert)

str4t0m, 2021-05-28:
@ABOSTM no external fixture is required to run the test.
I ran the test locally with success on g0 and g4. As CAN_FD_MODE is enabled per default, #35738 may have fixed this.
Would you mind running the test on the latest main?

alexanderwachter, 2021-05-28:
We have issues with that tests ATM. There are race conditions in the testcase (context is reused before the work queue item was processed).
We are working on a fix.

ABOSTM, 2021-05-28:
@str4t0m I still reproduce the issue with latest main
@alexanderwachter ok thanks

ABOSTM, 2021-08-09:
@alexanderwachter,
Issue is still reproduced with zephyr master.
do you have some progress with the fix you are talking about ? any reference for the fix ?

erwango, 2021-09-13:
@alexanderwachter gentle ping.

legoabram, 2021-09-20:
At the very least, I can confirm that this is the case. There are also some issues related to corrupting the work queue stack by reinitializing the work item before it's completely done. I haven't had to time to work on a fix, but I think the cleanest fix is to modify the library so that anything in the context is fully deallocated or uninitialized before notifying the user that the operation is complete.

closed 2021-12-04 by github-actions[bot]