zephyrproject-rtos/zephyr #35772: Support C++ exceptions on NIOS2

galak, 2021-05-28:
Currently we don't support C++ exceptions on NIOS2:

subsys/cpp/Kconfig:
```
config EXCEPTIONS
        bool "Enable C++ exceptions support"
        depends on !NIOS2

```

They may work, but we can't test on qemu_nios2 due to limited flash.

stephanosio, 2021-05-28:
So I rearranged QEMU memory map and tested it -- it still fails:
```
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: nios2
        cpu->env.reset_addr:            0
        cpu->env.exception_addr:        400020
*** Booting Zephyr OS build v2.6.0-rc2-54-g066a7c2de51b  ***
version 201703
Running test suite libcxx_tests
===================================================================
START - test_array
 PASS - test_array in 0.0 seconds
===================================================================
START - test_vector
 PASS - test_vector in 0.0 seconds
===================================================================
START - test_make_unique
 PASS - test_make_unique in 0.0 seconds
===================================================================
START - test_exception
terminate called after throwing an instance of 'int'
terminate called recursively
exit
```

QEMU patch (to be applied on top of `0004-qemu-nios2-Add-Altera-MAX-10-board-support-for-Zephy.patch`:
```diff
diff --git a/hw/nios2/altera_10m50_zephyr.c b/hw/nios2/altera_10m50_zephyr.c
index 4b3465429f..2144a04efc 100644
--- a/hw/nios2/altera_10m50_zephyr.c
+++ b/hw/nios2/altera_10m50_zephyr.c
@@ -40,12 +40,12 @@
 #define ROM_SIZE               32

 #define RAM_BASE               0x400000
-#define RAM_SIZE               262144
+#define RAM_SIZE               33554432

-#define U16550_0_BASE          0x440000
+#define U16550_0_BASE          0xf8001600
 #define U16550_0_IRQ_IDX       1

-#define TIMER_0_BASE           0x440200
+#define TIMER_0_BASE           0xf8001440
 #define TIMER_0_FREQ           50000000
 #define TIMER_0_IRQ_IDX                2
```

Zephyr dts:
```diff
diff --git a/dts/nios2/nios2-qemu.dtsi b/dts/nios2/nios2-qemu.dtsi
index b1b498b266..8b03e2c752 100644
--- a/dts/nios2/nios2-qemu.dtsi
+++ b/dts/nios2/nios2-qemu.dtsi
@@ -16,14 +16,14 @@
                };
        };

-       flash0: flash@420000 {
+       flash0: flash@1400000 {
                compatible = "soc-nv-flash";
-               reg = <0x420000 0x20000>;
+               reg = <0x1400000 0x1000000>;
        };

        sram0: memory@400000 {
                compatible = "mmio-sram";
-               reg = <0x400000 0x20000>;
+               reg = <0x400000 0x1000000>;
        };

        soc {
@@ -41,9 +41,9 @@
                        status = "disabled";
                };

-               ns16550_uart: uart@440000 {
+               ns16550_uart: uart@f8001600 {
                        compatible = "ns16550";
-                       reg = <0x440000 0x400>;
+                       reg = <0xf8001600 0x400>;
                        interrupts = <1>;
                        clock-frequency = <50000000>;
                        label = "UART_0";
```

stephanosio, 2021-05-28:
Found the problem. It has to do with `gcc_except_table`.

tejlmand, 2021-05-28:
> QEMU patch (to be applied on top of `0004-qemu-nios2-Add-Altera-MAX-10-board-support-for-Zephy.patch`:

@stephanosio when using Zephyr-sdk 0.13.0-alpha, then Qemu on nios2 works (Zephyr-sdk 0.13 comes with Qemu6). 
But it seems broken in Zephyr-sdk 0.12.4.
https://github.com/zephyrproject-rtos/sdk-ng/releases

stephanosio, 2021-05-28:
> But it seems broken in Zephyr-sdk 0.12.4.

Technically, it shouldn't work due to the reason described here: https://github.com/zephyrproject-rtos/zephyr/issues/32448#issuecomment-849854848; but, if things just happen to fall in the right place, it just might work as it did for your case. I will submit a patch to sdk-ng to increase memory available for NIOS2 testing later (#35758).

stephanosio, 2021-05-28:
Meanwhile, I will submit a patch to fix NIOS2 C++ exception problem. The problem has been identified: https://github.com/zephyrproject-rtos/zephyr/issues/35772#issuecomment-850429980.

closed 2021-05-28 by galak