zephyrproject-rtos/zephyr #35778: PWM : STM32: TIMER INTERRUPT CALLBACK HANDLING

manoj153, 2021-05-28:
[image]

Possible to access any of the interrupt callback?
My goal is to signal `PWM cycle` has elapsed so that a consumer can keep track of the count.    

@erwango

erwango, 2021-05-28:
@manoj153 Thanks for raising this. At fisrt sight, this would require PWM API extention (either generic API or STM32 sepcific). This is not something we have in plans right now. We'll revise plans if more people are interested.

manoj153, 2021-05-28:
as start to make it work on STM32 specific, in which source file I am supposed to introduce these changes?

> @manoj153 Thanks for raising this. At fisrt sight, this would require PWM API extention (either generic API or STM32 sepcific). This is not something we have in plans right now. We'll revise plans if more people are interested.

erwango, 2021-05-31:
@manoj153 PWM STM32 driver: `drivers/pwm/pwm_stm32.c`

erwango, 2021-11-03:
@manoj153  is https://github.com/zephyrproject-rtos/zephyr/pull/39414 responding to this need ?

erwango, 2022-06-29:
No feedback > Closing.
Please re-open if required

adrienbruant, 2024-03-11:
I have implemented an extended PWM interface that allows registering a callback on update interrupt. This was needed to drive a step/dir interface for a stepper motor. The callback allows the application to count the number of steps taken, and to change the period and pulse to follow a velocity ramp.

Is there any chance to see an _extended_ PWM interface for STM32 be merged upstream? Have a look at https://github.com/zephyrproject-rtos/zephyr/commit/6dbff88b68d8ecebd98aecd7dccb0444d94d30d4 and let me know if this would be of interest to the zephy project. Maybe merging hardware specific feature is against the rules, but there is only so much we can do with the standard API.

erwango, 2024-03-11:
@adrienbruant It is possible to upstream an API extension, see https://github.com/zephyrproject-rtos/zephyr/pull/52980 as an example.
Regarding your exact need, it would be worth first checking on discord if there is any interest of having a generic api for this.

adrienbruant, 2024-03-11:
@erwango Thank you for the pointer.