zephyrproject-rtos/zephyr #35812: ESP32 Factory app partition is not bootable

maweicoltsmart, 2021-05-30:
**Describe the bug**

Make the following changes：

diff --git a/boards/xtensa/esp32/esp32.dts b/boards/xtensa/esp32/esp32.dts
index cb0f691b4c..92d5887ee3 100644
--- a/boards/xtensa/esp32/esp32.dts
+++ b/boards/xtensa/esp32/esp32.dts
@@ -57,6 +57,16 @@
        rts-pin = <7>;
        cts-pin = <8>;
        hw-flow-control;
+       /* QUECTEL BG9X */
+       quectel_bg9x: quectel_bg9x {
+               compatible = "quectel,bg9x";
+               label = "quectel,bg9x";
+               mdm-power-gpios = <&gpio0 4 GPIO_ACTIVE_LOW>;
+               mdm-reset-gpios = <&gpio0 4 GPIO_ACTIVE_LOW>;
+               /* mdm-dtr-gpios = <&gpio0 26 GPIO_ACTIVE_LOW>; */
+               /* mdm-wdisable-gpios = <&gpio0 29 GPIO_ACTIVE_LOW>; */
+               status = "disabled";
+       };
 };

 &i2c0 {
diff --git a/samples/net/gsm_modem/prj.conf b/samples/net/gsm_modem/prj.conf
index 390c3b55a4..7790da23c2 100644
--- a/samples/net/gsm_modem/prj.conf
+++ b/samples/net/gsm_modem/prj.conf
@@ -4,6 +4,7 @@ CONFIG_SERIAL=y
 # GSM modem support
 CONFIG_MODEM=y
 CONFIG_MODEM_GSM_PPP=y
+CONFIG_MODEM_QUECTEL_BG9X=y
 CONFIG_MODEM_GSM_UART_NAME="UART_1"

 # PPP networking support
@@ -11,7 +12,7 @@ CONFIG_NET_PPP=y
 CONFIG_NET_L2_PPP=y
 CONFIG_NET_NATIVE=y
 CONFIG_NETWORKING=y
-
+CONFIG_NET_SOCKETS=y
 CONFIG_NET_L2_PPP_TIMEOUT=10000

 # IPv4 enables PPP IPCP support

Compile and download to the target board。
The following error is reported when running:

rst:0x3 (SW_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7328
ho 0 tail 12 room 4
load:0x40078000,len:13212
load:0x40080400,len:4568
entry 0x400806f4
[0;32mI (60) boot: ESP-IDF 22e7576326-dirty 2nd stage bootloader[0m
[0;32mI (60) boot: compile time 11:22:27[0m
[0;32mI (60) boot: chip revision: 1[0m
[0;32mI (64) boot_comm: chip revision: 1, min. bootloader chip revision: 0[0m
[0;32mI (72) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (76) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (81) boot.esp32: SPI Flash Size : 4MB[0m
[0;32mI (85) boot: Enabling RNG early entropy source...[0m
[0;32mI (91) boot: Partition Table:[0m
[0;32mI (94) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (101) boot:  0 nvs              WiFi data        01 02 00002000 00006000[0m
[0;32mI (109) boot:  1 phy_init         RF data          01 01 00008000 00001000[0m
[0;32mI (117) boot:  2 factory          factory app      00 00 00010000 00200000[0m
[0;32mI (124) boot: End of partition table[0m
[0;32mI (128) boot_comm: chip revision: 1, min. application chip revision: 0[0m
[0;31mE (136) esp_image: image at 0x10000 segment count 17 exceeds max 16[0m
[0;31mE (143) boot: Factory app partition is not bootable[0m
[0;31mE (148) boot: No bootable app partitions in the partition table[0m
ets Jun  8 2016 00:22:57

sylvioalves, 2021-05-31:
@maweicoltsmart, I could reproduce in in here.  
I think those GPIO phandle used on DTS is not supported yet on ESP32. However, this issue is not yet related to it.
Would you try with the patch below and let me know?
Also, you might want to set `status = "okay";`.

```c
diff --git a/soc/xtensa/esp32/linker.ld b/soc/xtensa/esp32/linker.ld
index 4f09b154d8..1d964eb894 100644
--- a/soc/xtensa/esp32/linker.ld
+++ b/soc/xtensa/esp32/linker.ld
@@ -189,9 +189,10 @@ SECTIONS
     Z_LINK_ITERABLE(net_l2);
 #endif
 
-  } GROUP_DATA_LINK_IN(RAMABLE_REGION, ROMABLE_REGION)
+    . = ALIGN(4);
+    Z_LINK_ITERABLE(_static_thread_data);
 
-  Z_ITERABLE_SECTION_RAM(_static_thread_data, 4)
+  } GROUP_DATA_LINK_IN(RAMABLE_REGION, ROMABLE_REGION)
 
 #pragma push_macro("Z_ITERABLE_SECTION_RAM")
 #pragma push_macro("Z_ITERABLE_SECTION_RAM_GC_ALLOWED")
```

maweicoltsmart, 2021-06-01:
> @maweicoltsmart, I could reproduce in in here.
> I think those GPIO phandle used on DTS is not supported yet on ESP32. However, this issue is not yet related to it.
> Would you try with the patch below [...]

FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map
cmd.exe /C "cd . && C:\Users\Colt\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr\zephyr_prebuilt.elf  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=D:/ZephyrProject/code/zephyrproject/zephyr/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ethernet/libsubsys__net__l2__ethernet.a  zephyr/subsys/net/l2/ppp/libsubsys__net__l2__ppp.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/net/lib/conn_mgr/libsubsys__net__lib__conn_mgr.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/entropy/libdrivers__entropy.a  modules/mbedtls/libmodules__mbedtls.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -L"c:/users/colt/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0"  -LD:/ZephyrProject/code/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -TD:/ZephyrProject/code/zephyrproject/modules/hal/espressif/zephyr/adapter/src/linker/esp32.rom.alias.ld  -TD:/ZephyrProject/code/zephyrproject/modules/hal/espressif/zephyr/../components/esp_rom/esp32/ld/esp32.rom.ld  -TD:/ZephyrProject/code/zephyrproject/modules/hal/espressif/zephyr/../components/esp_rom/esp32/ld/esp32.rom.libgcc.ld  -TD:/ZephyrProject/code/zephyrproject/modules/hal/espressif/zephyr/../components/esp32/ld/esp32.peripherals.ld  -Wl,--wrap=k_calloc  -Wl,--wrap=k_malloc  -lnet80211  -lcore  -lpp  -lcoexist  -lphy  -lrtc  -lgcc  -LD:/ZephyrProject/code/zephyrproject/modules/hal/espressif/zephyr/../components/esp32  -LD:/ZephyrProject/code/zephyrproject/modules/hal/espressif/zephyr/../components/esp_wifi/lib/esp32  -Wl,--defsym,phy_printf=printf  -Wl,--defsym,net80211_printf=printf  -Wl,--defsym,coexist_printf=printf && cmd.exe /C "cd /D D:\ZephyrProject\code\zephyrproject\zephyr\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E echo ""
c:/users/colt/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: zephyr\zephyr_prebuilt.elf section `noinit' will not fit in region `dram0_1_seg'
c:/users/colt/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_1_seg' overflowed by 21472 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'D:\ZephyrProject\code\zephyrproject\zephyr\build'

maweicoltsmart, 2021-06-01:
> FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map
> cmd.exe /C "cd . && C:\Users\Colt.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe [...]

I enabled modem and wifi at the same time, and then it showed insufficient memory.

maweicoltsmart, 2021-06-01:
> @maweicoltsmart, I could reproduce in in here.
> I think those GPIO phandle used on DTS is not supported yet on ESP32. However, this issue is not yet related to it.
> Would you try with the patch below [...]

FAILED: zephyr/zephyr_prebuilt.elf zephyr/zephyr_prebuilt.map
cmd.exe /C "cd . && C:\Users\Colt\.espressif\tools\xtensa-esp32-elf\esp-2020r3-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe   zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj -o zephyr\zephyr_prebuilt.elf  -Wl,-T  zephyr/linker_zephyr_prebuilt.cmd  -Wl,-Map=D:/ZephyrProject/code/zephyrproject/zephyr/build/zephyr/zephyr_prebuilt.map  -Wl,--whole-archive  app/libapp.a  zephyr/libzephyr.a  zephyr/arch/common/libarch__common.a  zephyr/arch/arch/xtensa/core/libarch__xtensa__core.a  zephyr/lib/libc/minimal/liblib__libc__minimal.a  zephyr/lib/posix/liblib__posix.a  zephyr/subsys/net/libsubsys__net.a  zephyr/subsys/net/l2/ethernet/libsubsys__net__l2__ethernet.a  zephyr/subsys/net/l2/ppp/libsubsys__net__l2__ppp.a  zephyr/subsys/net/ip/libsubsys__net__ip.a  zephyr/subsys/net/lib/conn_mgr/libsubsys__net__lib__conn_mgr.a  zephyr/subsys/random/libsubsys__random.a  zephyr/drivers/gpio/libdrivers__gpio.a  zephyr/drivers/serial/libdrivers__serial.a  zephyr/drivers/entropy/libdrivers__entropy.a  modules/mbedtls/libmodules__mbedtls.a  -Wl,--no-whole-archive  zephyr/kernel/libkernel.a  zephyr/CMakeFiles/offsets.dir/./arch/xtensa/core/offsets/offsets.c.obj  -L"c:/users/colt/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0"  -LD:/ZephyrProject/code/zephyrproject/zephyr/build/zephyr  -lgcc  zephyr/arch/common/libisr_tables.a  -Wl,--gc-sections  -Wl,--build-id=none  -Wl,--sort-common=descending  -Wl,--sort-section=alignment  -Wl,-u,_OffsetAbsSyms  -Wl,-u,_ConfigAbsSyms  -nostdlib  -static  -no-pie  -Wl,-X  -Wl,-N  -Wl,--orphan-handling=warn  -TD:/ZephyrProject/code/zephyrproject/modules/hal/espressif/zephyr/adapter/src/linker/esp32.rom.alias.ld  -TD:/ZephyrProject/code/zephyrproject/modules/hal/espressif/zephyr/../components/esp_rom/esp32/ld/esp32.rom.ld  -TD:/ZephyrProject/code/zephyrproject/modules/hal/espressif/zephyr/../components/esp_rom/esp32/ld/esp32.rom.libgcc.ld  -TD:/ZephyrProject/code/zephyrproject/modules/hal/espressif/zephyr/../components/esp32/ld/esp32.peripherals.ld  -Wl,--wrap=k_calloc  -Wl,--wrap=k_malloc  -lnet80211  -lcore  -lpp  -lcoexist  -lphy  -lrtc  -lgcc  -LD:/ZephyrProject/code/zephyrproject/modules/hal/espressif/zephyr/../components/esp32  -LD:/ZephyrProject/code/zephyrproject/modules/hal/espressif/zephyr/../components/esp_wifi/lib/esp32  -Wl,--defsym,phy_printf=printf  -Wl,--defsym,net80211_printf=printf  -Wl,--defsym,coexist_printf=printf && cmd.exe /C "cd /D D:\ZephyrProject\code\zephyrproject\zephyr\build\zephyr && "C:\Program Files\CMake\bin\cmake.exe" -E echo ""
c:/users/colt/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: zephyr\zephyr_prebuilt.elf section `noinit' will not fit in region `dram0_1_seg'
c:/users/colt/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `dram0_1_seg' overflowed by 21472 bytes
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: 'C:\Program Files\CMake\bin\cmake.EXE' --build 'D:\ZephyrProject\code\zephyrproject\zephyr\build'

**I enabled modem and wifi at the same time, and then it showed insufficient memory.**

sylvioalves, 2021-06-01:
@maweicoltsmart Hi, ok, I see.

1. Could you confirm that you have successfully build and tested BG9X with the patch I shared?
2. Regarding BG9X and WiFi usage, yes, it will consume lots fo RAM. Trying reducing the HEAP size available for WiFi:
`CONFIG_HEAP_MEM_POOL_SIZE=32768`

I was able to build it in here:
```c
[358/358] Linking C executable zephyr/zephyr.elf
Memory region         Used Size  Region Size  %age Used
     iram0_0_seg:       84996 B       128 KB     64.85%
     irom0_0_seg:      277194 B    3342304 B      8.29%
     dram0_0_seg:       65416 B     180736 B     36.19%
     dram0_1_seg:      100784 B     113840 B     88.53%
     drom0_0_seg:       55152 B    4194272 B      1.31%
    rtc_iram_seg:          0 GB         8 KB      0.00%
    rtc_slow_seg:          0 GB         4 KB      0.00%
        IDT_LIST:          0 GB         8 KB      0.00%
```

sylvioalves, 2021-06-01:
@maweicoltsmart Hi, I'll PR the fix for flash segment issue.  
However, regarding the RAM overflow, I'll check with the team. Perhaps moving some stuff to DRAM0_0 is a way.

maweicoltsmart, 2021-06-02:
> I think those GPIO phandle used on DTS is not supported yet on ESP32. However, this issue is not yet related to it.

**In the esp32 specification, uart can use any gpio. Do you mean that currently Zephyr's esp32 serial driver only supports gpio0 pins as uart?**

sylvioalves, 2021-06-02:
@maweicoltsmart Dismiss my comment about the phandle usage. It is for bg9x modem. 
The RAM problem is really an issue. Can you try reducing BG9X RECV buffer. It is default 30k. Wondering if this should be placed into Kconfig instead of define.

`NET_BUF_POOL_DEFINE(mdm_recv_pool, MDM_RECV_MAX_BUF, MDM_RECV_BUF_SIZE, 0, NULL);`  

`quectel-bg9x.h`
```c
#define MDM_RECV_MAX_BUF		  10
#define MDM_RECV_BUF_SIZE		  512
```

shubhamkulkarni97, 2021-06-03:
@maweicoltsmart ESP32 has discrete RAM regions. The system heap allocator is enabled for the first DRAM region i.e. `dram0_0_seg`.

The second DRAM region (i.e. `dram0_1_seg`) is used for `noinit` data. To add free memory from `dram0_1_seg` to heap, we have setup an allocator for this region. If you are having issue with segment overflow, you can reduce the size of heap allocator from `dram0_1_seg` by reducing value of `ESP_HEAP_MEM_POOL_REGION_1_SIZE` config option in menuconfig.

I agree that this behaviour should be documented somewhere.

Hope this solves your issue related to `dram0_1_seg` overflow!

maweicoltsmart, 2021-06-03:
> @maweicoltsmart ESP32 has discrete RAM regions. The system heap allocator is enabled for the first DRAM region i.e. `dram0_0_seg`.
>
> The second DRAM region (i.e. `dram0_1_seg`) is used for `noinit` [...]

According to the solution you gave, the problem has been solved, thank you!

closed 2021-06-03 by maweicoltsmart

RicArch97, 2023-01-18:
Hi, 

I stumbled upon this issue while trying to figure out the in my eyes absurd high RAM usage on `dram0_1_seg`, leaving hardly anything for my application. 

Apparently `CONFIG_ESP_HEAP_MEM_POOL_REGION_1_SIZE` was set to `49152`, is there a specific reason for this?  How much can i lower this region size before it becomes unsafe?

I do not use the heap in my own application code, so i guess this depends on the drivers, of which the usage is unknown to me.