zephyrproject-rtos/zephyr #35813: Zephyr Native Posix Build Uses Linux Build Machine Headers out of Sandbox

bob2oneil, 2021-05-30:
I am creating a network enabled application, and my build targets are both native_posix_64 and qemu_riscv64.  I am able to compile for the QEMU board target.

However, for the native_posix_64 built, I get collisions with headers that are defined on my native Linux build machine rather than what I assumed was a cross compiled "sandbox", where for native POSIX, the cross compilation would not use anything native to my Linux distro, but entirely Zephyr alone.

As such, I am unable to compile properly as there is a mixture of Zephyr and native Linux.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -p always -b native_posix_64 ./

**Expected behavior**
A clean build within any errors and inclusion of native Linux build machine headers.

**Impact**
Showstopper

**Logs and console output**
home/robocom/zephyrproject/zephyr/include/net/socket_select.h:110: warning: "FD_SETSIZE" redefined
  110 | #define FD_SETSIZE ZSOCK_FD_SETSIZE
      | 
In file included from /usr/include/x86_64-linux-gnu/sys/types.h:179,
                 from /home/robocom/zephyrproject/zephyr/include/net/socket.h:25,
                 from ../src/config/sessionConfig.cpp:16:
/usr/include/x86_64-linux-gnu/sys/select.h:73: note: this is the location of the previous definition
   73 | #define FD_SETSIZE  __FD_SETSIZE
      | 
In file included from /home/robocom/zephyrproject/zephyr/include/net/socket.h:29,
                 from ../src/config/sessionConfig.cpp:16:
/home/robocom/zephyrproject/zephyr/include/net/socket_select.h:112:19: error: conflicting declaration of C function ‘int select(int, zsock_fd_set*, zsock_fd_set*, zsock_fd_set*, zsock_timeval*)’
  112 | static inline int select(int nfds, zsock_fd_set *readfds,

**Environment (please complete the following information):**
 - OS: Linux Ubuntu 20
 - Toolchain (e.g Zephyr 2.6.0 RC1)

**Additional context**
This is the inclusion of the native include header is troublesome:

from /usr/include/x86_64-linux-gnu/sys/types.h:179,

Project Configuration is as follows:

# Optionally force the file system to be recreated
#CONFIG_APP_WIPE_STORAGE=y

# fs_dirent structures are big.
CONFIG_MAIN_STACK_SIZE=2048

#CONFIG_FLASH_HAS_PAGE_LAYOUT=y
CONFIG_LOG=y
CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_LITTLEFS=y
CONFIG_FS_LOG_LEVEL_DBG=y

# CONFIG_SHELL=y
# CONFIG_SHELL_LOG_LEVEL_INF=y

# CONFIG_FILE_SYSTEM=y
# CONFIG_FILE_SYSTEM_SHELL=y

# Debug Options
CONFIG_DEBUG_OPTIMIZATIONS=y

# Disable logging using '%s' where strdup()s occur
CONFIG_LOG_DETECT_MISSED_STRDUP=n

# Log Buffer Size when not logging immediately
# CONFIG_LOG_BUFFER_SIZE=1024

CONFIG_POSIX_API=y

# Networking config
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_TCP=n
CONFIG_NET_SOCKETS=y
# CONFIG_NET_SOCKETS_POSIX_NAMES=y
# Network debug config
CONFIG_NET_LOG=y

------------

My default gcc include paths are as follows:

echo | gcc -E -Wp,-v -
ignoring nonexistent directory "/usr/local/include/x86_64-linux-gnu"
ignoring nonexistent directory "/usr/lib/gcc/x86_64-linux-gnu/9/include-fixed"
ignoring nonexistent directory "/usr/lib/gcc/x86_64-linux-gnu/9/../../../../x86_64-linux-gnu/include"
#include "..." search starts here:
#include <...> search starts here:
 /usr/lib/gcc/x86_64-linux-gnu/9/include
 /usr/local/include
 /usr/include/x86_64-linux-gnu
 /usr/include
End of search list.
# 1 "<stdin>"
# 1 "<built-in>"
# 1 "<command-line>"
# 31 "<command-line>"
# 1 "/usr/include/stdc-predef.h" 1 3 4
# 32 "<command-line>" 2
# 1 "<stdin>"

bob2oneil, 2021-05-30:
Also have collision on Linux header for select.h.

--------------------------------

In file included from /usr/include/x86_64-linux-gnu/sys/select.h:30,
                 from /usr/include/x86_64-linux-gnu/sys/types.h:179,
                 from /home/robocom/zephyrproject/zephyr/include/net/socket.h:25,
                 from ../src/config/sessionConfig.cpp:16:
/home/robocom/zephyrproject/zephyr/include/net/socket_select.h:124:19: note: to match this ‘(’
  124 | static inline int FD_ISSET(int fd, zsock_fd_set *set)

bob2oneil, 2021-05-31:
I am not positive what the proper behavior is for NATIVE POSIX development.  I looks like gcc in the output logs matches the gcc version on my Linux machine. Perhaps with that, it is using the default include paths prior to the Zephyr ones.  By contrast, should one expect that an X86 cross compilation should occur?  Should I change my environment/system configuration to prevent these default include files from being part of the build process?  Is there any ordering of the project include files (including stdio, stdint) that could be tripping this up?  

The full compile output is as follows:

 west build: making build dir /home/robocom/RCSUS/ddltconfig/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/robocom/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/robocom/RCSUS/ddltconfig
-- Zephyr version: 2.6.0-rc1 (/home/robocom/zephyrproject/zephyr), build: v2.6.0-rc1-61-g8eea9ebc3a88
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: native_posix_64
-- Cache files will be written to: /home/robocom/.cache/zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /home/robocom/zephyrproject/zephyr/boards/posix/native_posix/native_posix_64.dts
-- Generated zephyr.dts: /home/robocom/RCSUS/ddltconfig/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/robocom/RCSUS/ddltconfig/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/robocom/RCSUS/ddltconfig/build/zephyr/include/generated/device_extern.h
Parsing /home/robocom/RCSUS/ddltconfig/Kconfig
Loaded configuration '/home/robocom/zephyrproject/zephyr/boards/posix/native_posix/native_posix_64_defconfig'
Merged configuration '/home/robocom/RCSUS/ddltconfig/prj.conf'
Configuration saved to '/home/robocom/RCSUS/ddltconfig/build/zephyr/.config'
Kconfig header saved to '/home/robocom/RCSUS/ddltconfig/build/zephyr/include/generated/autoconf.h'

warning: POSIX_API (defined at lib/posix/Kconfig:12) was assigned the value 'y' but got the value
'n'. Check these unsatisfied dependencies: (!ARCH_POSIX) (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_POSIX_API.html and/or look up
POSIX_API in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: TEST_RANDOM_GENERATOR (defined at subsys/random/Kconfig:8) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: (!ENTROPY_HAS_DRIVER) (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_TEST_RANDOM_GENERATOR.html and/or look
up TEST_RANDOM_GENERATOR in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: The choice symbol NEWLIB_LIBC (defined at lib/libc/Kconfig:27) was selected (set =y), but
EXTERNAL_LIBC (defined at lib/libc/Kconfig:34) ended up as the choice selection. See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_NEWLIB_LIBC.html and/or look up
NEWLIB_LIBC in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/robocom/RCSUS/ddltconfig/build
-- west build: building application
[1/147] Preparing syscall dependency handling

[140/147] Building CXX object CMakeFiles/app.dir/src/config/sessionConfig.cpp.obj
FAILED: CMakeFiles/app.dir/src/config/sessionConfig.cpp.obj 
ccache /usr/bin/g++  -DBUILD_VERSION=v2.6.0-rc1-61-g8eea9ebc3a88 -DKERNEL -D_FORTIFY_SOURCE=2 -D_POSIX_C_SOURCE=200809 -D_XOPEN_SOURCE=600 -D_XOPEN_SOURCE_EXTENDED -D__ZEPHYR__=1 -I/home/robocom/zephyrproject/zephyr/include -Izephyr/include/generated -I/home/robocom/zephyrproject/zephyr/soc/posix/inf_clock -I/home/robocom/zephyrproject/zephyr/boards/posix/native_posix -I/home/robocom/zephyrproject/zephyr/subsys/net/lib/sockets/. -I/home/robocom/zephyrproject/zephyr/subsys/net/l2 -I/home/robocom/zephyrproject/modules/fs/littlefs -Og -fcheck-new -std=c++2a -Wno-register -Wno-volatile -fno-exceptions -fno-rtti -imacros /home/robocom/RCSUS/ddltconfig/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -fdiagnostics-color=always -imacros /home/robocom/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/robocom/RCSUS/ddltconfig=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/robocom/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/robocom/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -m64 -fPIC -include /home/robocom/zephyrproject/zephyr [... cut ...]
                 from /home/robocom/zephyrproject/zephyr/include/net/socket.h:25,
                 from ../src/config/sessionConfig.cpp:16:
/home/robocom/zephyrproject/zephyr/include/net/socket_select.h:134:20: error: expected unqualified-id before ‘)’ token
  134 | static inline void FD_SET(int fd, zsock_fd_set *set)
      |                    ^~~~~~
cc1plus: warning: unrecognized command line option ‘-Wno-volatile’
ninja: build stopped: subcommand failed.
FATAL ERROR: command exited with status 1: /usr/bin/cmake --build /home/robocom/RCSUS/ddltconfig/build

bob2oneil, 2021-05-31:
Here is the project kernel configuration as a ZIP compressed file.

[prj.5.31.21.zip](https://github.com/zephyrproject-rtos/zephyr/files/6572170/prj.5.31.21.zip)

aescolar, 2021-06-01:
Hi @bob2oneil 
Unfortunately `CONFIG_POSIX_API=y` is not supported in native_posix.
The conflicts you are seeing are the result of trying to use that.
This is not a bug, but a known limitation, that has been a long standing, lowish priority, feature request 
( #24685 ).

As extra info naive_posix does use your host default gcc and libC. That is as intended by design.

If this answers your question, feel free to close the issue.

bob2oneil, 2021-06-01:
Thanks for the quick attention to this issue. I explicitly set the project configuration with CONFIG_POSIX_API=n, and the problem still remains.  I still see Linux header files included in the Native Posix build.

The following are some updates to both the kernel configuration and the output. 

Any other ideas?

`#
# Copyright (c) 2019 Peter Bigot Consulting, LLC
#
# SPDX-License-Identifier: Apache-2.0
#
# Project Configuration
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_NEWLIB_LIBC=y
CONFIG_STD_CPP2A=y
CONFIG_CBPRINTF_FP_SUPPORT=y

# Random number generation
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y

CONFIG_HEAP_MEM_POOL_SIZE=32768

# Optionally force the file system to be recreated
#CONFIG_APP_WIPE_STORAGE=y

# fs_dirent structures are big.
CONFIG_MAIN_STACK_SIZE=2048

#CONFIG_FLASH_HAS_PAGE_LAYOUT=y
CONFIG_LOG=y
CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_LITTLEFS=y
CONFIG_FS_LOG_LEVEL_DBG=y

CONFIG_SHELL=y
CONFIG_SHELL_LOG_LEVEL_INF=y

CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_SHELL=y

# Debug Options
CONFIG_DEBUG_OPTIMIZATIONS=y

# Disable logging using '%s' where strdup()s occur
CONFIG_LOG_DETECT_MISSED_STRDUP=n

# Log Buffer Size when not logging immediately
# CONFIG_LOG_BUFFER_SIZE=1024

# Not supported for Native Posix
CONFIG_POSIX_API=n

# Networking config
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_TCP=n
CONFIG_NET_SOCKETS=y
# CONFIG_NET_SOCKETS_POSIX_NAMES=y
# Network debug config
CONFIG_NET_LOG=y
`

`robocom@scnuc-1:~/SUS/ddltconfig$ west build -p always -b native_posix_64 ./
-- west build: making build dir /home/robocom/SUS/ddltconfig/build pristine
[... 135 lines cut ...]
                 from /home/robocom/zephyrproject/zephyr/include/net/socket.h:25,
                 from ../src/config/sessionConfig.cpp:16:
/home/robocom/zephyrproject/zephyr/include/net/socket_select.h:129:20: error: expected unqualified-id before ‘)’ token
  129 | static inline void FD_CLR(int fd, zsock_fd_set *set)
      |                    ^~~~~~
In file included from /usr/include/x86_64-linux-gnu/sys/select.h:30,
                 from /usr/include/x86_64-linux-gnu/sys/types.h:179,
                 from /home/robocom/zephyrproject/zephyr/include/net/socket.h:25,
                 from ../src/config/sessionConfig.cpp:16:
/home/robocom/zephyrproject/zephyr/include/net/socket_select.h:134:20: error: expected unqualified-id before ‘void’
  134 | static inline void FD_SET(int fd, zsock_fd_set *set)
      |                    ^~~~~~
/home/robocom/zephyrproject/zephyr/include/net/socket_select.h:134:20: error: expected ‘)’ before ‘void’
/home/robocom/zephyrproject/zephyr/include/net/socket_select.h:134:20: error: expected ‘)’ before ‘void’
In file included from /usr/include/x86_64-linux-gnu/sys/select.h:123,
                 from /usr/include/x86_64-linux-gnu/sys/types.h:179,
                 from /home/robocom/zephyrproject/zephyr/include/net/socket.h:25,
                 from ../src/config/sessionConfig.cpp:16:
/home/robocom/zephyrproject/zephyr/include/net/socket_select.h:134:20: error: expected unqualified-id before ‘)’ token
  134 | static inline void FD_SET(int fd, zsock_fd_set *set)
      |                    ^~~~~~
cc1plus: warning: unrecognized command line option ‘-Wno-volatile’
[105/166] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/net_if.c.obj
ninja: build stopped: subcommand failed.
`

aescolar, 2021-06-01:
> I still see Linux header files included in the Native Posix build.

That is meant to happen

bob2oneil, 2021-06-01:
No reason on NEWLIB other than to ensure that I can build for both QEMU and Native Posix, presumably using the same project files .. that in generally being new to Zephyr.

Thanks for the tips.

I have tried again with NEWLIB set to external, but have the same build issues.

I am to understand that perhaps different kernel configs are required for different board targets?

When targeting posix, should I not be including #include <net/socket.h>, and just use
literally the platform version of say sockets, select(), poll(), etc.

I was thinking that all of the native Linux platform stuff would be sandboxed, so only content
within Zephyr proper would be used.  That could be a misconception on my park.

bob2oneil, 2021-06-01:
"""
robocom@scnuc-1:~/SUS/ddltconfig$ west build -p always -b native_posix_64 ./
-- west build: making build dir /home/robocom/SUS/ddltconfig/build pristine
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/robocom/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/robocom/SUS/ddltconfig
-- Zephyr version: 2.6.0-rc1 (/home/robocom/zephyrproject/zephyr), build: v2.6.0-rc1-61-g8eea9ebc3a88
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.5") found components: Interpreter 
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: native_posix_64
-- Cache files will be written to: /home/robocom/.cache/zephyr
-- Found dtc: /usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found toolchain: host (gcc/ld)
-- Found BOARD.dts: /home/robocom/zephyrproject/zephyr/boards/posix/native_posix/native_posix_64.dts
-- Generated zephyr.dts: /home/robocom/SUS/ddltconfig/build/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/robocom/SUS/ddltconfig/build/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/robocom/SUS/ddltconfig/build/zephyr/include/generated/device_extern.h
Parsing /home/robocom/SUS/ddltconfig/Kconfig
Loaded configuration '/home/robocom/zephyrproject/zephyr/boards/posix/native_posix/native_posix_64_defconfig'
Merged configuration '/home/robocom/SUS/ddltconfig/prj.conf'
Configuration saved to '/home/robocom/SUS/ddltconfig/build/zephyr/.config'
Kconfig header saved to '/home/robocom/SUS/ddltconfig/build/zephyr/include/generated/autoconf.h'

warning: TEST_RANDOM_GENERATOR (defined at subsys/random/Kconfig:8) was assigned the value 'y' but
got the value 'n'. Check these unsatisfied dependencies: (!ENTROPY_HAS_DRIVER) (=n). See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_TEST_RANDOM_GENERATOR.html and/or look
up TEST_RANDOM_GENERATOR in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/robocom/SUS/ddltconfig/build
-- west build: building application
[1/166] Preparing syscall dependency handling

[146/166] Building CXX object CMakeFiles/app.dir/src/config/sessionConfig.cpp.obj
FAILED: CMakeFiles/app.dir/src/config/sessionConfig.cpp.obj 
ccache /usr/bin/g++  -DBUILD_VERSION=v2.6.0-rc1-61-g8eea9ebc3a88 -DKERNEL -D_FORTIFY_SOURCE=2 -D_POSIX_C_SOURCE=200809 -D_XOPEN_SOURCE=600 -D_XOPEN_SOURCE_EXTENDED -D__ZEPHYR__=1 -I/home/robocom/zephyrproject/zephyr/include -Izephyr/include/generated -I/home/robocom/zephyrproject/zephyr/soc/posix/inf_clock -I/home/robocom/zephyrproject/zephyr/boards/posix/native_posix -I/home/robocom/zephyrproject/zephyr/lib/util/fnmatch/. -I/home/robocom/zephyrproject/zephyr/subsys/net/lib/sockets/. -I/home/robocom/zephyrproject/zephyr/subsys/net/l2 -I/home/robocom/zephyrproject/modules/fs/littlefs -Og -fcheck-new -std=c++2a -Wno-register -Wno-volatile -fno-exceptions -fno-rtti -imacros /home/robocom/SUS/ddltconfig/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -fdiagnostics-color=always -imacros /home/robocom/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/robocom/SUS/ddltconfig=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/robocom/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/robocom/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -m64 -fPIC -include /home/robocom/zephyrproject/zephyr/arch/posix/include/posix_cheats.h -fno-freestanding -MD -MT CMakeFiles/app.dir/src/config/sessionConfig.cpp.obj -MF CMakeFiles/app.dir/src/config/sessionConfig.cpp.obj.d -o CMakeFiles/app.dir/src/config/sessionConfig.cpp.obj -c ../src/config/sessionConfig.cpp
In file included from /home/robocom/zephyrproject/zephyr/include/net/socket.h:29,
                 from ../src/config/sessionConfig.cpp:16:
/home/robocom/zephyrproject/zephyr/include/net/socket_select.h:110: warning: "FD_SETSIZE" redefined
  110 | #define FD_SETSIZE ZSOCK_FD_SETSIZE
      | 
In file included from /usr/include/x86_64-linux-gnu/sys/types.h:179,
                 from /home/robocom/zephyrproject/zephyr/include/net/socket.h:25,
                 from ../src/config/sessionConfig.cpp:16:
/usr/include/x86_64-linux-gnu/sys/select.h:73: note: this is the location of the previous definition
   73 | #define FD_SETSIZE  __FD_SETSIZE
      | 
In file included from /home/robocom/zephyrproject/zephyr/include/net/socket.h:29,
                 from ../src/config/sessionConfig.cpp:16:
/home/robocom/zephyrproject/zephyr/include/net/socket_select.h:112:19: error: conflicting declaration of C function ‘int select(int, zsock_fd_set*, zsock_fd_set*, zsock_fd_set*, zsock_timeval*)’
  112 | static inline int select(int nfds, zsock_fd_set *readfds,
      |                   ^~~~~~
In file included from /usr/include/x86_64-linux-gnu/sys/types.h:179,
[... cut ...]
                 from /home/robocom/zephyrproject/zephyr/include/net/socket.h:25,
                 from ../src/config/sessionConfig.cpp:16:
/home/robocom/zephyrproject/zephyr/include/net/socket_select.h:134:20: error: expected unqualified-id before ‘)’ token
  134 | static inline void FD_SET(int fd, zsock_fd_set *set)
      |                    ^~~~~~
cc1plus: warning: unrecognized command line option ‘-Wno-volatile’
[155/166] Building C object zephyr/subsys/net/ip/CMakeFiles/subsys__net__ip.dir/net_context.c.obj
ninja: build stopped: subcommand failed.
""

aescolar, 2021-06-01:
>I am to understand that perhaps different kernel configs are required for different board targets?

That is not uncommon. You can set kconfig overlays for each board you want to treat differently. (you can find some examples of that in the tree, just look for samples with a folder with a board name, with some kconfig and/or dts overlays in them)

>When targeting posix, should I not be including #include <net/socket.h>

@jukkar are you able to help @bob2oneil ? Does he need to disable NET_SOCKETS_POSIX_NAMES in native_posix?  

> and just use literally the platform version of say sockets, select(), poll(), etc.

I doubt that would work..

> I was thinking that all of the native Linux platform stuff would be sandboxed,

Not really. You are building with the host libC. So name clashes are unfortunately easy to encounter.

aescolar, 2021-06-01:
@bob2oneil like this: (tripple back ticks == the backwards accent character) ` ``` `  your text ` ``` ` 

```your text ```
You can click on preview to see you how your text will render

bob2oneil, 2021-06-01:
Thanks, I will correct all future back ticks properly.

What do you recommend now for creating native POSIX socket applications running under Zephyr.

I am looking for BSD sockets, Linux poll() and select(), promiscuous sockets, runtime settable IP and MTUS, one real Ethernet interface (kernel configured) and one virtual Ethernet (9K MTU).

I want to build a bridge applications IPv4 between 2 interfaces.

I am ok with different kernel configurations for different board targets

jukkar, 2021-06-01:
> @jukkar are you able to help @bob2oneil ? Does he need to disable NET_SOCKETS_POSIX_NAMES and probably more in native_posix?

The networking sample applications are compiled all the time with native_posix and there is no need to disable `CONFIG_NET_SOCKETS_POSIX_NAMES`. You can try to compile various network applications like `samples/net/sockets/echo_server` and it compiles just fine
```
west build -d build -b native_posix_64 samples/net/sockets/echo_server
```

bob2oneil, 2021-06-01:
I am able to build the sample application without any event.  However, my application has a bunch more C++ class, uses some standard C/C++ headers, and I wonder if somehow "types.h" gets included or other platform headers that is causing the havoc.  I will do a strip down to see if this improves anything.  Would one of you fine gentleman be willing to test out my project to see if at least we are experiencing the same build issues?  I could ZIP and send in a private email.  I will post anything I find from my strip of standard header operations. Hopefully this work will be of benefit to the broader Zephyr community.

bob2oneil, 2021-06-01:
Thanks for the quick responses.  Attached is the current build issues as I strip away as much of the standard headers as possible.  Is there any advanced west argument that would provide more information on the include search paths, and what caused a certain standard header to be included?   For example, the errors occur as a function of including standard header "types.h", which are not directly included in the source code.

**/usr/include/x86_64-linux-gnu/sys/types.h:179**

---------------------------------------------------------

-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/gcc
-- Configuring done
-- Generating done
-- Build files have been written to: /home/robocom/RCSUS/ddltconfig/build
-- west build: building application
[1/167] Preparing syscall dependency handling

[160/167] Building CXX object CMakeFiles/app.dir/src/config/sessionConfig.cpp.obj
FAILED: CMakeFiles/app.dir/src/config/sessionConfig.cpp.obj 
ccache /usr/bin/g++  -DBUILD_VERSION=v2.6.0-rc1-61-g8eea9ebc3a88 -DKERNEL -D_FORTIFY_SOURCE=2 -D_POSIX_C_SOURCE=200809 -D_XOPEN_SOURCE=600 -D_XOPEN_SOURCE_EXTENDED -D__ZEPHYR__=1 -I/home/robocom/zephyrproject/zephyr/include -Izephyr/include/generated -I/home/robocom/zephyrproject/zephyr/soc/posix/inf_clock -I/home/robocom/zephyrproject/zephyr/boards/posix/native_posix -I/home/robocom/zephyrproject/zephyr/lib/util/fnmatch/. -I/home/robocom/zephyrproject/zephyr/subsys/net/lib/sockets/. -I/home/robocom/zephyrproject/zephyr/subsys/net/l2 -I/home/robocom/zephyrproject/modules/fs/littlefs -Og -fcheck-new -std=c++2a -Wno-register -Wno-volatile -fno-exceptions -fno-rtti -imacros /home/robocom/RCSUS/ddltconfig/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -fdiagnostics-color=always -imacros /home/robocom/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wpointer-arith -Wexpansion-to-defined -Wno-address-of-packed-member -Wno-unused-but-set-variable -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -fmacro-prefix-map=/home/robocom/RCSUS/ddltconfig=CMAKE_SOURCE_DIR -fmacro-prefix-map=/home/robocom/zephyrproject/zephyr=ZEPHYR_BASE -fmacro-prefix-map=/home/robocom/zephyrproject=WEST_TOPDIR -ffunction-sections -fdata-sections -m64 -fPIC -include /home/robocom/zephyrproject/zephyr/arch/posix/include/posix_cheats.h -fno-freestanding -MD -MT CMakeFiles/app.dir/src/config/sessionConfig.cpp.obj -MF CMakeFiles/app.dir/src/config/sessionConfig.cpp.obj.d -o CMakeFiles/app.dir/src/config/sessionConfig.cpp.obj -c ../src/config/sessionConfig.cpp
In file included from /home/robocom/zephyrproject/zephyr/include/net/socket.h:29,
                 from ../src/config/sessionConfig.cpp:16:
/home/robocom/zephyrproject/zephyr/include/net/socket_select.h:110: warning: "FD_SETSIZE" redefined
  110 | #define FD_SETSIZE ZSOCK_FD_SETSIZE
      | 
In file included from /usr/include/x86_64-linux-gnu/sys/types.h:179,
                 from /home/robocom/zephyrproject/zephyr/include/net/socket.h:25,
                 from ../src/config/sessionConfig.cpp:16:
/usr/include/x86_64-linux-gnu/sys/select.h:73: note: this is the location of the previous definition
   73 | #define FD_SETSIZE  __FD_SETSIZE

-------------------------------------------------

bob2oneil, 2021-06-01:
The referenced collision in types.h is the inclusion of sys/select with collides with the Zephyr implementation even with CONFIG_POSIX_API=n

```
/* It also defines `fd_set' and the FD_* macros for `select'.  */
# include <sys/select.h>
#endif /* Use misc.  */
```

bob2oneil, 2021-06-01:
```
In file included from /home/robocom/zephyrproject/zephyr/include/net/socket.h:29,
                 from ../src/config/sessionConfig.cpp:16:
/home/robocom/zephyrproject/zephyr/include/net/socket_select.h:112:19: error: conflicting declaration of C function ‘int select(int, zsock_fd_set*, zsock_fd_set*, zsock_fd_set*, zsock_timeval*)’
  112 | static inline int select(int nfds, zsock_fd_set *readfds,
      |                   ^~~~~~
In file included from /usr/include/x86_64-linux-gnu/sys/types.h:179,
                 from /home/robocom/zephyrproject/zephyr/include/net/socket.h:25,
                 from ../src/config/sessionConfig.cpp:16:
/usr/include/x86_64-linux-gnu/sys/select.h:101:12: note: previous declaration ‘int select(int, fd_set*, fd_set*, fd_set*, timeval*)’
  101 | extern int select (int __nfds, fd_set *__restrict __readfds,
```

bob2oneil, 2021-06-02:
Additional testing needs to be performed, but I was finally able to get this sample project to build under Native Posix 64 by changing the order of the entries in the project kernel configuration.  This might make some sense, or perhaps this change was anecdotal, and simply caused some other unexplained benefit.  I was building within west using the "-p always" flag to presumably sweep away and previous build artifacts and I deleted the entire 'build' subfolder a few times.

My working prj.conf is as follows. Perhaps this explanation rings a bell for you Zephyr experts that have generously commented on my postings.

I hope this will be of some benefit to other users that may have experienced the same behaviors for Native POSIX builds with collisions with Linux file system headers.

```
#
# Copyright (c) 2019 Peter Bigot Consulting, LLC
#
# SPDX-License-Identifier: Apache-2.0
#
# Project Configuration
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_STD_CPP2A=y
CONFIG_CBPRINTF_FP_SUPPORT=y

# Random number generation
CONFIG_ENTROPY_GENERATOR=y
CONFIG_TEST_RANDOM_GENERATOR=y

CONFIG_HEAP_MEM_POOL_SIZE=32768

# Optionally force the file system to be recreated
#CONFIG_APP_WIPE_STORAGE=y

# fs_dirent structures are big.
CONFIG_MAIN_STACK_SIZE=2048

#CONFIG_FLASH_HAS_PAGE_LAYOUT=y
CONFIG_LOG=y
CONFIG_FLASH=y
CONFIG_FLASH_MAP=y
CONFIG_FLASH_PAGE_LAYOUT=y
CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_LITTLEFS=y
CONFIG_FS_LOG_LEVEL_DBG=y

CONFIG_SHELL=y
CONFIG_SHELL_LOG_LEVEL_INF=y

CONFIG_FILE_SYSTEM=y
CONFIG_FILE_SYSTEM_SHELL=y

# Debug Options
CONFIG_DEBUG_OPTIMIZATIONS=y

# Disable logging using '%s' where strdup()s occur
CONFIG_LOG_DETECT_MISSED_STRDUP=n

# Log Buffer Size when not logging immediately
# CONFIG_LOG_BUFFER_SIZE=1024

# Networking config
CONFIG_NETWORKING=y
CONFIG_NET_IPV4=y
CONFIG_NET_IPV6=n
CONFIG_NET_TCP=n
CONFIG_NET_SOCKETS=y

# Not supported for Native Posix
CONFIG_POSIX_API=n
CONFIG_NET_SOCKETS_POSIX_NAMES=n

# Network debug config
CONFIG_NET_LOG=y
```

aescolar, 2021-06-02:
> my application has a bunch more C++ class, uses some standard C/C++ headers

Unfortunately that is a bad use case for native_posix.

> I was finally able to get this sample project to build under Native Posix 

Nice to hear. I guess we can close this issue/question by now then(?) (it can be always reopened if needed)

bob2oneil, 2021-06-02:
My thanks to aescolar and jukkar for helping out on this topic in a quick manner.  You guys are a credit to the open source community.

closed 2021-06-02 by bob2oneil