zephyrproject-rtos/zephyr #35816: timer: STM32: using hw timer for counting and interrupt callback

manoj153, 2021-05-31:
**Is your enhancement proposal related to a problem? Please describe.**

Generally, a general-purpose timer could be used to signal something accurately whenever ticks based kernel delay are less accurate 

**Describe the solution you'd like**
I think having a timer interrupt callback exposed, user can add non-blocking code as per their need in the cb. 

**Describe alternatives you've considered**

**Additional context**
having some api like :
`timer_start(&TIM_X, count_dir, target_count, void  *user_callback_func)`
`timer_stop(&TIM_X)`

the effort to add this feature hardware-specific is pretty easy, I am thinking how for Zephyr as a whole, is there a way like a dedicated subsys to handle this currently?

erwango, 2021-05-31:
@manoj153 can you check if counter API is what you actually need ?
cf `include/drivers/counter.h`

If that's the case, then what you miss is is the support for STM32 GP timer using this API.
For now this API is ported on RTC, but I understand accuracy will not be at expected level.

Also, please clarify link with your previous issue https://github.com/zephyrproject-rtos/zephyr/issues/35778 and close if possible so that we don't keep multiple tickets open for the same purpose. Thanks

manoj153, 2021-05-31:
> @manoj153 can you check if counter API is what you actually need ?
> cf `include/drivers/counter.h`
>
> If that's the case, then what you miss is is the support for STM32 GP timer using this API.
> For now [...]

@erwango this issue has no link with my previous one.

That was specific to PWM use case and this is related to timer countdown use cases for better accuracy. 

I will look at suggested counter.h

manoj153, 2021-05-31:
I had a requirement where needing microsecond accuracy to synchronise and where sleep and other kernel timing offering was simply not accurate. Nothing can get close to 1us

@erwango

stoberblog, 2021-08-20:
I too would like to use "counters" (ie hardware timers) on my stm32. I've used on Nordic MCUs and works fine, but there seems to be no support for STM32. How would one implement this while we wait for an official implementation?

manoj153, 2021-08-22:
@stoberblog  as of now, such timer instance is not officially supported by zephyr, counter is using the RTC tick as far I understand. 

I think having a counter driver to work with multiple defined timer will be ideal, timer setting could be passed down form DT, this is probably a feature request 
> I too would like to use "counters" (ie hardware timers) on my stm32. I've used on Nordic MCUs and works fine, but there seems to be no support for STM32. How would one implement this while we wait for an official implementation?
> 
> I'm going to have a try at implementing my own counter_ll_stm32_timer.c module tomorrow

FRASTM, 2021-09-06:
_"counter is using the RTC tick_"  Yes, that's what I realized too when trying to implement the set/get _top value ( counter: stm32: Missing implementation of set_top_value #37206). The RTC is not really convenient; beacuse it is  not well designed (in the stm32 but probably also other platforms) for counting other than hours and date. Even the WakeUp timer function is not enough, because depending on the RTC. An actual Timer like the the STM32 GP timer or LP timer is more suitable, with all the expected functions : alarm, reset, comparison.

erwango, 2021-11-03:
@manoj153  is https://github.com/zephyrproject-rtos/zephyr/pull/39414 responding to this need ?

erwango, 2022-01-28:
@manoj153 ping

manoj153, 2022-01-29:
> @manoj153 ping

@erwango what you need from me?

erwango, 2022-01-30:
I asked you a question on this point [here](https://github.com/zephyrproject-rtos/zephyr/issues/35816#issuecomment-959710933)

manoj153, 2022-01-31:
> I asked you a question on this point [here](https://github.com/zephyrproject-rtos/zephyr/issues/35816#issuecomment-959710933)

maybe using `counter_set_channel_alarm()` will satisfy the requirement, I need to study counter API a bit, will let know / close the issue once I have done assessment on my end

manoj153, 2022-07-05:
Yes, now I see there is counter driver addition for STM32,

The alarm function on the counter will satisfy the request.

p.s I should get back to zephyr and have fun again

closed 2022-07-05 by manoj153