zephyrproject-rtos/zephyr #35821: Filesystem subsystem logging

sw, 2021-05-31:
The filesystem subsystem generally logs errors (`LOG_ERR`) when operations fail.

This is good during development, but it interferes with `mcumgr fs` commands when logging is enabled on the shell. For example, when uploading a file, `fs_stat()` is called by the `fs_mgmt` module. If the file does not exist yet on the device (which is obviously not an error case), the error message "failed get file or dir stat" will be printed. This then confuses the mcumgr CLI which expects a CBOR/base64 encoded response. The upload fails with a timeout.

There are other cases where a non-existing file is not an error. The user of the fs API may simply want to detect the presence of a file.

So the first suggestion would be to turn these `LOG_ERR` calls into `LOG_WRN`.

But I then tried to disable logging for the fs subsystem. Logging is used as follows:

https://github.com/zephyrproject-rtos/zephyr/blob/3d39f72a88b3100ac83e7c6c285ae567aca640d0/subsys/fs/fs.c#L20-L22

I would expect to be able to turn off logging for that module by setting `CONFIG_FS_LOG_LEVEL_OFF=y`, but this does not work - the errors are still printed.

Is this because the value 0 for `OFF` is treated specially?
https://github.com/zephyrproject-rtos/zephyr/blob/3d39f72a88b3100ac83e7c6c285ae567aca640d0/include/logging/log.h#L368-L371

To reproduce:
# add `CONFIG_FS_LOG_LEVEL_OFF=y` to `samples/subsys/fs/littlefs/prj.conf`
# `west build -b native_posix_64 samples/subsys/fs/littlefs`
# `build/zephyr/zephyr.exe`

How to disable logging for the fs subsystem but leave it enabled for other modules?

sw, 2021-07-28:
Related: #35718 #36882

`CONFIG_FS_LOG_LEVEL_OFF=y` stills seems to be ignored in lastest main, though.

de-nordic, 2021-07-28:
@sw I think that the problem is with default log level overriding the fs log level selection.
The LOG_ macros, at some point, resolve to checking: `Z_LOG_CONST_LEVEL_CHECK(ll)`, where `ll` is module log level, and it seems that it selects at least `CONFIG_LOG_DEFAULT_LEVEL` log level, which means that even if you have selected lower valuer for FS, it will override it.

nordicjm, 2026-01-12:
> This then confuses the mcumgr CLI which expects a CBOR/base64 encoded response. The upload fails with a timeout.

Previously, output would be intermingled between MCUmgr and logs however this was fixed a long time ago by having printk go through logging, so this is no longer a problem as the full MCUmgr response will be sent by itself (there might also be logging output but it will be separate)