zephyrproject-rtos/zephyr #35824: ARM: Cortex-M: improvements for v2.7.0 release

ioannisg, 2021-05-31:
**Is your enhancement proposal related to a problem? Please describe.**
Enhancements for the Cortex-M architecture support

**List of open issues**

- [ ] Optimize SPLIM programming, potentially removing SPLIM clearing to zero
- [ ] Validate MSP in HardFault and NMI handlers
- [ ] Use labels for constants in assembly
- [ ] Investigate if barrier instructions are needed for semaphore operations
- [ ] Investigate if aligned access can be enforced, or at least set by default
- [ ] Use unprivileged load-store instructions in user mode assembly code
- [ ] Re-work the VTOR offset register for the different variants of Cortex-M
- [ ] Implement FPU support in QEMU and enable it in CI
- [ ] Clean up SoC-specific code from arch/arm/aarch32 and include/arch/arm/aarch32
- [ ] Switch arch_irq_lock to use the BASEPRI max instead of BASEPRI setting
- [ ] Configure non-banked faults in accordance with TF-M, if building with TF-M
- [ ] Store FP callee-saved context in stack instead of allocating a placeholder in kernel memory

theotherjimmy, 2022-01-12:
Do we have a reproducing case for the FP_SHARING bug? I would like to take a look into that soon.

carlescufi, 2024-02-22:
Assigned to @ithinuel. Feel free to close this one.

ithinuel, 2024-06-22:
@theotherjimmy I know it's been a while but do you remember if you had time to look into that at the time and if there was a reproducible issue ?