zephyrproject-rtos/zephyr #35848: File creation date time for FAT FS

rohand87, 2021-06-01:
Hello,

I am using the FAT FS sub system to create/read/write file on to an SD Card. The problem I am facing is when I read the SD Card on Windows, the file's "Date Modified" is showing `01-01-2016 00:00`

I have an a RTC on my board from where I can get the latest date time. My question is how do I configure the latest date-time into Zephyr so that the FAT FS automatically gets the correct time for file creation/modification? Or is there a way to update the file modification date?

**Environment**
- Zephyr 2.5.0
- Processor: STM32G4

Screenshot from my Windows PC
[image]

Thank you

de-nordic, 2021-06-18:
I will look into the issue.

ymfloresl, 2021-07-17:
I think changing _FS_NORTC to 0 within modules/fs/fatfs/include/ffconf.h. and adding the get_fattime() function using the RTC values would solve your issue.