zephyrproject-rtos/zephyr #35856: intel_adsp_cavs15: run semaphore testcases failed on ADSP

KangJianX, 2021-06-02:
**Describe the bug**
run tests/kernel/semaphore/semaphore/ failed on ADSP

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p intel_adsp_cavs15 --device-testing -T tests/kernel/semaphore/semaphore --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh,/home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem,/home/ztest/work/rimage/config,/home/ztest/work/rimage/build/rimage" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py,--no-history"

**Logs and console output**

> ===================================================================
START - test_sem_multi_take_timeout_diff_sem
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/semaphore/semaphore/src/main.c:1090: test_sem_multi_take_timeout_diff_sem: (retrieved_info.timeout == SEC2MS(i + 1) is false)
timeout did not occur properly
FAIL - test_sem_multi_take_timeout_diff_sem in 3.1 seconds
===================================================================

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain Zephyr-sdk-0.12.4
 - Commit ID c5e50ef0e219

LeiW000, 2021-08-03:
Please remove Stale label.

KangJianX, 2021-08-20:
It can not be reproduced in the latest code. So close this issue.

closed 2021-08-20 by KangJianX