zephyrproject-rtos/zephyr #35857: intel_adsp_cavs15: run msgq testcases failed on ADSP

KangJianX, 2021-06-02:
**Describe the bug**
run tests/kernel/msgq/msgq_usage/ failed on ADSP

**To Reproduce**
Steps to reproduce the behavior:
1. twister -p intel_adsp_cavs15 --device-testing -T tests/kernel/msgq/msgq_usage/ --west-flash="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/flash.sh,/home/ztest/work/zephyrproject/modules/audio/sof/keys/otc_private_key.pem,/home/ztest/work/rimage/config,/home/ztest/work/rimage/build/rimage" --device-serial-pty="/home/ztest/work/zephyrproject/zephyr/boards/xtensa/intel_adsp_cavs15/tools/adsplog.py,--no-history"

**Logs and console output**

> Running test suite msgq_usage
===================================================================
START - test_msgq_usage
Unknown message 0

**Environment (please complete the following information):**
 - OS: Fedora28
 - Toolchain: Zephyr-sdk-0.12.4
 - Commit ID: c5e50ef

LeiW000, 2021-08-03:
Please remove Stale label.

andyross, 2021-09-08:
Works for me on both cavs15 and cavs25 as of current main.  Retest?

KangJianX, 2021-09-14:
Blocked by #38446 . I will retest this one when #38446 is fixed.

nashif, 2021-10-06:
I can reproduce this, getting:

```
ups1:zephyr(main): sudo  boards/xtensa/intel_adsp_cavs15/tools/adsplog.py
Running test suite msgq_usage
===================================================================
START - test_msgq_usage
Unknown message 0
```

closed 2021-10-06 by nashif