zephyrproject-rtos/zephyr #35870: Build failure with gcc 11.x on native_posix

nashif, 2021-06-02:
**Describe the bug**
Warnings from new gcc turning into errors:

```
In file included from /home/nashif/Work/zephyrproject/zephyr/include/kernel_includes.h:39,
                 from /home/nashif/Work/zephyrproject/zephyr/include/kernel.h:17,
                 from /home/nashif/Work/zephyrproject/zephyr/include/zephyr.h:18,
                 from /home/nashif/Work/zephyrproject/zephyr/kernel/init.c:14:
/home/nashif/Work/zephyrproject/zephyr/include/sys/thread_stack.h:155:16: error: ignoring attribute ‘section (".noinit.\"WEST_TOPDIR/zephyr/kernel/init.c\".3")’ because it conflicts with previous ‘section (".noinit.\"WEST_TOPDIR/zephyr/kernel/include/kernel_internal.h\".0")’ [-Werror=attributes]
  155 |         struct z_thread_stack_element __kstackmem \
      |                ^~~~~~~~~~~~~~~~~~~~~~
/home/nashif/Work/zephyrproject/zephyr/kernel/init.c:61:1: note: in expansion of macro ‘K_KERNEL_STACK_ARRAY_DEFINE’
   61 | K_KERNEL_STACK_ARRAY_DEFINE(z_interrupt_stacks, CONFIG_MP_NUM_CPUS,
      | ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/nashif/Work/zephyrproject/zephyr/kernel/include/kernel_internal.h:136:36: note: previous declaration here
  136 | extern K_KERNEL_STACK_ARRAY_DEFINE(z_interrupt_stacks, CONFIG_MP_NUM_CPUS,
      |                                    ^~~~~~~~~~~~~~~~~~
/home/nashif/Work/zephyrproject/zephyr/include/sys/thread_stack.h:157:17: note: in definition of macro ‘K_KERNEL_STACK_ARRAY_DEFINE’
  157 |                 sym[nmemb][Z_KERNEL_STACK_LEN(size)]
      |                 ^~~
cc1: all warnings being treated as errors
gmake[2]: *** [zephyr/kernel/CMakeFiles/kernel.dir/build.make:134: zephyr/kernel/CMakeFiles/kernel.dir/init.c.obj] Error
1
gmake[1]: *** [CMakeFiles/Makefile2:2781: zephyr/kernel/CMakeFiles/kernel.dir/all] Error 2
gmake: *** [Makefile:103: all] Error 2

```

**To Reproduce**

On fedora 34 with gcc 11.x

Steps to reproduce the behavior:
1.  twister -T samples/philosophers/ -p native_posix
4. See error

niflostancu, 2021-06-29:
I've encountered this as well with my projects (using the ARM Cortex-M arch) and I would like it fixed.

The actual cause is that there are many separate files defining the `z_interrupt_stacks` extern symbol:
- [kernel/include/kernel_internal.h](https://github.com/zephyrproject-rtos/zephyr/blob/98f879bad65b5d593858dd281fdab8d7cd0d7f87/kernel/include/kernel_internal.h#L155);
- other includes / modules declaring the same symbol (e.g., _aarch32/cortex_m/stack.h_);

Both use the `K_KERNEL_STACK_ARRAY_DEFINE` macro, which calls upon the `__in_section_unique` macro generating a unique section string for each filename it's used.
The real solution would be to have a single header file declaring symbols using these file-named sections.

As workaround, I use `-Wno-attributes` or `-Wno-error=attributes` (when `-Werror` is desired) in my project cmake script:
```cmake
check_set_compiler_property(APPEND PROPERTY warning_base -Wno-attributes)
```

dcpleung, 2021-06-29:
#35893 should fix this.

closed 2021-07-22 by galak