zephyrproject-rtos/zephyr #35871: LPS22HH sensor reporting wrong pressure data

nashif, 2021-06-02:
**Describe the bug**

The pressure sensor reporting wrong data that is too high

**To Reproduce**
Steps to reproduce the behavior:
1.  west build -b frdm_k64f samples/shields/x_nucleo_iks01a3/ -- -DSHIELD=x_nucleo_iks01a
2. See data on console

```

HTS221: Temperature: 25.9 C
HTS221: Relative Humidity: 45.5%
LPS22HH: Temperature: 26.3 C
LPS22HH: Pressure:25822.956 kpa
STTS751: Temperature: 27.2 C
LIS2MDL: Magn (gauss): x: -0.864, y: -0.270, z: -0.330
LIS2MDL: Temperature: 4631.4 C
LIS2DW12: Accel (m.s-2): x: -0.459, y: 0.000, z: 9.648
LSM6DSO: Accel (m.s-2): x: 0.043, y: -0.273, z: 9.892
LSM6DSO: GYro (dps): x: 0.005, y: 0.021, z: 0.047
51:: lps22hh trig 10080
51:: lis2dw12 trig 10182
51:: lsm6dso acc trig 2
51:: lsm6dso gyr trig 2
51:: lsm6dso temp trig 0
```

**Expected behavior**

Value should be realistic

nashif, 2021-06-02:
also note:

`LIS2MDL: Temperature: 4631.4 C`

avisconti, 2021-06-03:
@nashif 
Anas, can you possibly set the priority of this bug? For me it is low, but I'll let you decide. Anyway, I'll start asap.

avisconti, 2021-06-03:
> also note:
> 
> `LIS2MDL: Temperature: 4631.4 C`

@nashif 
This is LIS2MDL, can you please open a separate bug for it?

nashif, 2021-06-03:
see #35910

PR #35907 (2021-06-03): Fix issue with LPS22HH sensor reporting wrong pressure data
Fixes #35871
commit 4e8c809d (2021-06-03): drivers/sensor: lps22hh: Fix the raw to kPa sample conversion

The lps22hh 24 bit raw sample is left aligned, which means that
it needs to be right-shifted by 8 before applying conversion.
Moreover the conversion has been simplified for clarity.

Fix #35871