zephyrproject-rtos/zephyr #35873: Log on Intel ADSP / SOF repeats

lyakh, 2021-06-02:
**Describe the bug**
Running adsplog.py generates output with fragments repeating multiple times.

**To Reproduce**
1. Launch adsplog.py with output redirected to a file
2. In a different shell prompt perform actions, creating output, e.g. play audio, wait for the driver to suspend the DSP, play audio again, etc.
3. Stop logging
4. Search for some of timestamps, appearing closer to the end of the file, they will repeat multiple times

**Expected behavior**
The log should accumulate without repetition.

**Impact**
Logging usefulness is reduced a lot

**Logs and console output**
Search for 2338793 in [sof.log](https://github.com/zephyrproject-rtos/zephyr/files/6584213/sof.log)

**Environment (please complete the following information):**
Linux + SOF, observed on Apollolake (UP2) and Tigerlake

lyakh, 2021-06-02:
@andyross could you have a look, please?

lyakh, 2021-09-07:
The bug is still there, as also confirmed by @kv2019i Let's keep this issue open until either adsplog.py is fixed or we switch over to sof-logger

andyross, 2021-09-08:
Is this really still happening?  As I read it, current SOF builds with USE_PRINTK, so it's just forwarding to printk(), which has its own synchronization as long as CONFIG_PRINTK_SYNC is defined (which it is).  So it's logging the same way the upstream test suite is, and that's just not showing anything like this.  I'm wondering if maybe you have a stale adsplog.py or something?

andyross, 2021-10-20:
Management is eager to get this closed.  Is this really still happening?  Any idea on how to exercise outside a full SOF environment?  Again, it's just plain not happening anywhere in the Zephyr tests, on either of the two hardware platforms I have, nor on the SOF binaries at least as far as device initialization.  Frankly the logging as it stands right now seems rock solid and it's hard to imagine this kind of race (or whatever) sneaking in on SOF only.  Maybe there's a hidden configuration difference with the way you're building or a kconfig collision or something?

closed 2022-01-05 by github-actions[bot]