zephyrproject-rtos/zephyr #35880: PSA tests run indefinitely when CONFIG_TFM_IPC=y

akiannillo, 2021-06-02:
**Describe the bug**
I tried to run the tf-m PSA test with CONFIG_TFM_IPC=y.
It simply reboots the test suite over and over, indefinitely.

Run with `west build samples/tfm_integration/tfm_psa_test/ -p -b mps2_an521_nonsecure -t run -- -DCONFIG_TFM_PSA_TEST_STORAGE=y`

It can be solved increasing the watchdog timer in `../modules/tee/tfm/psa-arch-tests/api-tests/platform/targets/tgt_dev_apis_tfm_an521/target.cfg` but not sure if it's a solution or a workaround.

related to #31910

ioannisg, 2021-06-02:
Hi @akiannillo, I believe this is the intended behavior of the sample.

```
The sample reboots after 5 seconds to demonstrate rebooting with TF-M.
The sys_reboot call is routed to TF-M, since the nonsecure app is not allowed
to perform the reboot directly.
```

ioannisg, 2021-06-02:
Yeap, so I suggest we close this.

closed 2021-07-27 by cfriedt