zephyrproject-rtos/zephyr #35890: Build system ignores explicit ZephyrBuildConfiguration_ROOT variable

palchak-google, 2021-06-02:
**Describe the bug**
CMake's `find_package` function allows users to set a specifically named variable to specify where to search for a specific package. Zephyr's build system uses CMake's `find_package` functionality to locate the optional ZephyrBuildConfiguration package. However, Zephyr's build system ignores the value of the ZephyrBuildConfiguration_ROOT variable if it is set. This prevents users from providing an explicit location for the ZephyrBuildConfiguration package.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a file named ZephyrBuildConfig.cmake outside of any West workspace.
2. Invoke CMake for any of the Zephyr sample projects, adding the additional CMake option "-DZephyrBuildConfiguration_ROOT=/path/to/folder/containing/package_file"

**Expected behavior**
Build system should find ZephyrBuildConfig.cmake in the folder named "/path/to/folder/containing/package_file" and load it, thus customizing the Zephyr build.

**Actual behavior**
Build system ignores the ZephyrBuildConfiguration_ROOT variable when searching for the ZephyrBuildConfiguration package, does not find any package by said name, and quietly proceeds.

**Impact**
Our company generally requires that third-party code (in this case Zephyr and all dependencies) be maintained under a //third_party/<vendor_name>/* directory structure. This means that the location of the West workspace for our projects is //third_party/zephyr/.west, the location of the Zephyr code is //third_party/zephyr/zephyr, and the location of first-party build-system code is //cmake. We are thus unable to use build Zephyr without either maintaining local modifications to Zephyr's build system or break the strict directory-based distinction between between first-party and third-party code.

**Additional context**
This problem is easily remedied by changing the `find_package(ZephyrBuildConfiguration ....)` call in boilerplate.cmake to add the value of the ZephyrBuildConfiguration_ROOT variable to the list of search paths.

palchak-google, 2021-06-02:
Actually, a better solution is to replace the following line in boilerplate.cmake:

```
find_package(ZephyrBuildConfiguration NAMES ZephyrBuild PATHS ${ZEPHYR_BASE}/../* QUIET NO_DEFAULT_PATH NO_POLICY_SCOPE)
```

with this:

```
find_package(ZephyrBuildConfiguration
  NAMES ZephyrBuild
  PATHS ${ZEPHYR_BASE}/../*
  QUIET NO_POLICY_SCOPE
  NO_SYSTEM_ENVIRONMENT_PATH
  NO_CMAKE_PATH
  NO_CMAKE_SYSTEM_PATH
  NO_CMAKE_ENVIRONMENT_PATH
  NO_CMAKE_PACKAGE_REGISTRY
  NO_CMAKE_SYSTEM_PACKAGE_REGISTRY
)
```

This change would enable searching in paths specified in the `<PackageName>_ROOT` CMake variable and the `<PackageName>_ROOT` environment variable while still excluding all other implicit search paths.

tejlmand, 2021-06-03:
@palchak-google fixed here: https://github.com/zephyrproject-rtos/zephyr/pull/35902

PR #35902 (2021-06-04): cmake: support ZephyrBuildConfiguration_ROOT for package location

Fixes: #35890

The current implementation of Zephyr build configuration CMake package
only allows the build configuration package to be placed inside a
Zephyr workspace.

This commit extends the usability by allowing users to locate the
Zephyr build configuration CMake package outside the Zephyr workspace
and then refer to the package using
`-DZephyrBuildConfiguration_ROOT=<path>`
`set(ZephyrBuildConfiguration_ROOT <path>)`

This allows users greater flexibility in their workspace layouts.
commit 489a69c0 (2021-06-04): cmake: support ZephyrBuildConfiguration_ROOT for package location

Fixes: #35890

The current implementation of Zephyr build configuration CMake package
only allows the build configuration package to be placed inside a
Zephyr workspace.

This commit extends the usability by allowing users to locate the
Zephyr build configuration CMake package outside the Zephyr workspace
and then refer to the package using
`-DZephyrBuildConfiguration_ROOT=<path>`
`set(ZephyrBuildConfiguration_ROOT <path>)`

This allows users greater flexibility in their workspace layouts.
closed 2021-06-04 by galak