zephyrproject-rtos/zephyr #35896: frdm_k64f: build failure missing dt-bindings/clock/kinetis_sim.h: No such file or directory

hakehuang, 2021-06-03:
**Describe the bug**
missing dtb clock file
dt-bindings/clock/kinetis_sim.h: No such file or directory

**To Reproduce**
Steps to reproduce the behavior:
cd samples/hello_world
1. mkdir build; cd build
2. cmake -DBOARD=frdm_k64f ..
3. make
4. See error

**Expected behavior**
build pass

**Impact**
kinetis support

**Logs and console output**
```
ubuntu@ubuntu-OptiPlex-7050:/home/shared/disk/zephyr_project/zephyr_test/zephyr/samples/net/gptp$ west build -b frdm_k64f
-- west build: generating a build system
Including boilerplate (Zephyr base): /home/shared/disk/zephyr_project/zephyr_test/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/shared/disk/zephyr_project/zephyr_test/zephyr/samples/net/gptp
-- Zephyr version: 2.6.0-rc3 (/home/shared/disk/zephyr_project/zephyr_test/zephyr), build: v2.6.0-rc3-16-g366a64cb1e7b
-- Found Python3: /usr/bin/python3.8 (found suitable exact version "3.8.0") found components: Interpreter
-- Found west (found suitable version "0.10.1", minimum required is "0.7.1")
-- Board: frdm_k64f
-- Cache files will be written to: /home/ubuntu/.cache/zephyr
-- Using toolchain: zephyr 0.12.4 (/home/ubuntu/zephyr-sdk)
-- Found dtc: /home/ubuntu/zephyr-sdk/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/shared/disk/zephyr_project/zephyr_test/zephyr/boards/arm/frdm_k64f/frdm_k64f.dts
In file included from /home/shared/disk/zephyr_project/zephyr_test/zephyr/dts/arm/nxp/MK64FN1M0VLL12.dtsi:7,
                 from /home/shared/disk/zephyr_project/zephyr_test/zephyr/boards/arm/frdm_k64f/frdm_k64f.dts:5,
                 from <command-line>:
/home/shared/disk/zephyr_project/zephyr_test/zephyr/dts/arm/nxp/nxp_k6x.dtsi:5:10: fatal error: dt-bindings/clock/kinetis_sim.h: No such file or directory
    5 | #include <dt-bindings/clock/kinetis_sim.h>
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
CMake Error at /home/shared/disk/zephyr_project/zephyr_test/zephyr/cmake/dts.cmake:157 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/shared/disk/zephyr_project/zephyr_test/zephyr/cmake/app/boilerplate.cmake:554 (include)
  /home/shared/disk/zephyr_project/zephyr_test/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:24 (include)
  /home/shared/disk/zephyr_project/zephyr_test/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:35 (include_boilerplate)
  CMakeLists.txt:5 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /home/shared/disk/zephyr_project/cmake-3.19.4-Linux-x86_64/bin/cmake -DWEST_PYTHON=/usr/bin/python3.8 -B/home/shared/disk/zephyr_project/zephyr_test/zephyr/samples/net/gptp/build -S/home/shared/disk/zephyr_project/zephyr_test/zephyr/samples/net/gptp -GNinja -DBOARD=frdm_k64f

```

**Environment (please complete the following information):**
 - OS: (e.g. Linux)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used: v2.6.0-rc3-16-g366a64cb1e7b

hakehuang, 2021-06-03:
@MaureenHelm , @dleach02

galak, 2021-06-03:
Can't reproduce this, https://github.com/zephyrproject-rtos/zephyr/blob/main/include/dt-bindings/clock/kinetis_sim.h exists so feels like something local to your environment.

hakehuang, 2021-06-03:
@galak  you are right, it is my lock repo issue, thank for quick check.

closed 2021-06-03 by hakehuang