zephyrproject-rtos/zephyr #35908: Stopping DHCP with network interface goes down leaves networking state in a broken state

thedjnK, 2021-06-03:
**Describe the bug**
We are developing an application which uses MQTT over ethernet with an interface that uses DHCP to get an address - hardware is an nRF5340-based module with enc424j600 ethernet driver. The application is runtime configurable for DHCP or static addressing, we are currently testing DHCP whereby it starts up, enables DHCP on an interface then makes a MQTT connection which fails and disconnects (expected) after which we remove the ethernet cable which then disables DHCP. At this point, it seems the networking stack falls apart (during the call to net_dhcpv4_stop), after adding debug it seems that this function actually never returns. As a result, issuing `net iface` at this point correctly shows that the network is down, however if the cable is plugged back in and the command is re-ran, it shows a very odd output where DHCP is still renewing, there is no main IP but it still has a gateway set, as per:
```
Interface 0x20002238 (Ethernet) [1]
===================================
Link addr : E8:EB:1B:CE:AB:9D
MTU       : 1500
Ethernet capabilities supported:
        10 Mbits
        100 Mbits
IPv4 unicast addresses (max 1):
        <none>
IPv4 multicast addresses (max 1):
        <none>
IPv4 gateway : 192.168.1.1
IPv4 netmask : 255.255.255.0
DHCPv4 lease time : 86400
DHCPv4 renew time : 43200
DHCPv4 server     : 192.168.1.1
DHCPv4 requested  : 192.168.1.2
DHCPv4 state      : renewing
DHCPv4 attempts   : 0
```
No management events seem to fire after this point to the user application

**To Reproduce**
This is a bit of a complex application but the overall design of the application is that we add handlers for NET_EVENT_DNS_SERVER_ADD, NET_EVENT_DNS_SERVER_DEL, NET_EVENT_IF_UP, NET_EVENT_IF_DOWN, NET_EVENT_IPV4_CMD_DHCP_START, NET_EVENT_IPV4_CMD_DHCP_BOUND, NET_EVENT_IPV4_CMD_DHCP_STOP events, start DHCP near application start using net_dhcpv4_start and call net_dhcpv4_stop in the NET_EVENT_IF_DOWN callback and trigger it by removing the ethernet cable. This is the point where the stack breaks. Stack sizes have been greatly increased to see if that was an issue and the net_mgmt event queue size has been increased to 8 which has has no impact.

**Expected behavior**
For the interface's IP and network information to clear correctly and for the networking stack to continue working

**Impact**
Showstopper

**Environment (please complete the following information):**
 - OS: Windows 10
 - Toolchain: (GNU Arm Embedded Toolchain 9-2020-q2-update) 9.3.1
 - Version: nRF connect SDK 1.5.1 which is using zephyr-v2.4.0-3444-gbe068d1193bd

jukkar, 2021-06-03:
I do not have an easy way to test MQTT atm, but that should not be relevant thing here as DHCP is running in lower layer.
So I did a small test with echo-server like this:
```
cd samples/net/sockets/echo_server
cmake -Bbuild . -DBOARD=frdm_k64f -DCONFIG_NET_DHCPV4=y
make -Cbuild -j9 flash
```
and then start to monitor the console. The Ethernet cable is connected and I see this:
```
** Booting Zephyr OS build v2.6.0-rc3-28-g410f07c099dd  ***

[00:00:00.005,000] <inf> net_config: Initializing network
[00:00:00.005,000] <inf> net_config: Waiting interface 1 (0x20002730) to be up...
[00:00:03.001,000] <inf> net_config: Interface 1 (0x20002730) coming up
[00:00:03.001,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:03.001,000] <inf> net_config: IPv4 address: 192.0.2.1
[00:00:03.001,000] <inf> net_config: Running dhcpv4 client...
[00:00:03.105,000] <inf> net_config: IPv6 address: 2002:5b9c:41a0:0:4:9fff:fe36:8eb5
[00:00:03.106,000] <inf> net_echo_server_sample: Run echo server
[00:00:03.106,000] <inf> net_echo_server_sample: Network connected
[00:00:03.106,000] <inf> net_echo_server_sample: Starting...
[00:00:03.106,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv6)...
[00:00:03.106,000] <inf> net_echo_server_sample: Waiting for TCP connection on port 4242 (IPv4)...
[00:00:03.107,000] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv6)...
[00:00:03.107,000] <inf> net_echo_server_sample: Waiting for UDP packets on port 4242 (IPv4)...
[00:00:07.006,000] <inf> net_dhcpv4: Received: 192.168.178.41
[00:00:07.006,000] <inf> net_config: IPv4 address: 192.168.178.41
[00:00:07.006,000] <inf> net_config: Lease time: 864000 seconds
[00:00:07.006,000] <inf> net_config: Subnet: 255.255.255.0
uart:~$ net iface 

Interface 0x20002730 (Ethernet) [1]
===================================
Link addr : 02:04:9F:36:8E:B5
MTU       : 1480
Flags     : NO_AUTO_START,IPv4,IPv6
Ethernet capabilities supported:
        Virtual LAN
        Auto negotiation
        10 Mbits
        100 Mbits
IPv6 unicast addresses (max 3):
        fe80::4:9fff:fe36:8eb5 autoconf preferred infinite
        2001:db8::1 manual preferred infinite
        2002:5b9c:41a0:0:4:9fff:fe36:8eb5 autoconf preferred
IPv6 multicast addresses (max 4):
        ff02::1
        ff02::1:ff36:8eb5
        ff02::1:ff00:1
IPv6 prefixes (max 2):
        2002:5b9c:41a0::/64
IPv6 default router :
        fe80::9ec7:a6ff:fe5e:4735
IPv6 hop limit           : 255
IPv6 base reachable time : 30000
IPv6 reachable time      : 18991
IPv6 retransmit timer    : 0
IPv4 unicast addresses (max 1):
        192.168.178.41 DHCP preferred
IPv4 multicast addresses (max 1):
        <none>
IPv4 gateway : 192.168.178.1
IPv4 netmask : 255.255.255.0
DHCPv4 lease time : 864000
DHCPv4 renew time : 432000
DHCPv4 server     : 192.168.178.1
DHCPv4 requested  : 192.168.178.41
DHCPv4 state      : bound
DHCPv4 attempts   : 1
```
then I disconnect the cable:
```
[00:00:38.005,000] <inf> eth_mcux: ETH_0 link down
[00:00:38.005,000] <inf> net_echo_server_sample: Network disconnected
uart:~$ net iface

Interface 0x20002730 (Ethernet) [1]
===================================
Interface is down.
uart:~$ net iface 1

Interface 0x20002730 (Ethernet) [1]
===================================
Interface is down.
Link addr : 02:04:9F:36:8E:B5
MTU       : 1480
Flags     : NO_AUTO_START,IPv4,IPv6
Ethernet capabilities supported:
        Virtual LAN
        Auto negotiation
        10 Mbits
        100 Mbits
IPv6 unicast addresses (max 3):
        fe80::4:9fff:fe36:8eb5 autoconf preferred infinite
        2001:db8::1 manual preferred infinite
        2002:5b9c:41a0:0:4:9fff:fe36:8eb5 autoconf preferred
IPv6 multicast addresses (max 4):
        ff02::1
        ff02::1:ff36:8eb5
        ff02::1:ff00:1
IPv6 prefixes (max 2):
        2002:5b9c:41a0::/64
IPv6 default router :
        fe80::9ec7:a6ff:fe5e:4735
IPv6 hop limit           : 255
IPv6 base reachable time : 30000
IPv6 reachable time      : 18991
IPv6 retransmit timer    : 0
IPv4 unicast addresses (max 1):
        192.168.178.41 DHCP preferred
IPv4 multicast addresses (max 1):
        <none>
IPv4 gateway : 192.168.178.1
IPv4 netmask : 255.255.255.0
DHCPv4 lease time : 864000
DHCPv4 renew time : 432000
DHCPv4 server     : 192.168.178.1
DHCPv4 requested  : 192.168.178.41
DHCPv4 state      : renewing
DHCPv4 attempts   : 0
```
and then reconnect it:
```
[00:00:53.007,000] <inf> eth_mcux: ETH_0 enabled 100M full-duplex mode.
[00:00:53.007,000] <inf> net_echo_server_sample: Network connected
[00:00:53.107,000] <inf> net_config: IPv6 address: 2002:5b9c:41a0:0:4:9fff:fe36:8eb5
[00:00:53.107,000] <inf> net_config: IPv6 address: 2002:5b9c:41a0:0:4:9fff:fe36:8eb5
uart:~$ net iface 1

Interface 0x20002730 (Ethernet) [1]
===================================
Link addr : 02:04:9F:36:8E:B5
MTU       : 1480
Flags     : NO_AUTO_START,IPv4,IPv6
Ethernet capabilities supported:
        Virtual LAN
        Auto negotiation
        10 Mbits
        100 Mbits
IPv6 unicast addresses (max 3):
        fe80::4:9fff:fe36:8eb5 autoconf preferred infinite
        2001:db8::1 manual preferred infinite
        2002:5b9c:41a0:0:4:9fff:fe36:8eb5 autoconf preferred
IPv6 multicast addresses (max 4):
        ff02::1
        ff02::1:ff36:8eb5
        ff02::1:ff00:1
IPv6 prefixes (max 2):
        2002:5b9c:41a0::/64
IPv6 default router :
        fe80::9ec7:a6ff:fe5e:4735
IPv6 hop limit           : 255
IPv6 base reachable time : 30000
IPv6 reachable time      : 18991
IPv6 retransmit timer    : 0
IPv4 unicast addresses (max 1):
        192.168.178.41 DHCP preferred
IPv4 multicast addresses (max 1):
        <none>
IPv4 gateway : 192.168.178.1
IPv4 netmask : 255.255.255.0
DHCPv4 lease time : 864000
DHCPv4 renew time : 432000
DHCPv4 server     : 192.168.178.1
DHCPv4 requested  : 192.168.178.41
DHCPv4 state      : bound
DHCPv4 attempts   : 1
```
Is this workflow similar as what you are doing?

thedjnK, 2021-06-03:
Yes, I'll give it a try on one of the small networking sample apps and see if I can reproduce it on that.

closed 2021-08-31 by github-actions[bot]