zephyrproject-rtos/zephyr #35911: shield sample sensorhub does not produce and meaningful data

nashif, 2021-06-03:
**Describe the bug**

```
���[00:00:00.002,000] <dbg> LIS2DW12.lis2dw12_set_power_mode: Apply default Power Mode
[00:00:00.004,000] <inf> LSM6DSO: Initialize device LSM6DSO
[00:00:00.008,000] <inf> LSM6DSO: chip id 0x6c
[00:00:00.017,000] <inf> LSM6DSO: shub: start sensorhub for LSM6DSO
```

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b frdm_k64f samples/shields/x_nucleo_iks01a3/sensorhub/  -- -DSHIELD=x_nucleo_iks01a
2. west flash
3. See output, no sensor data at all

**Expected behavior**
expecting some sensor data

avisconti, 2021-06-03:
@nashif 
Did you put the x_nucleo_iks01a3 shield in Mode2 (sensorhub)?
See info on how to configure jumpers here:
https://docs.zephyrproject.org/latest/boards/shields/x_nucleo_iks01a3/doc/index.html

nashif, 2021-06-03:
aha, switched now and I am getting output, looks like I am on the surface of the sun :-)

```
LIS2DW12: Accel (m.s-2): x: -0.306, y: 0.077, z: 9.571
LSM6DSO: Accel (m.s-2): x: 0.000, y: 0.000, z: 0.000
LSM6DSO: GYro (dps): x: 0.000, y: 0.000, z: 0.000
LSM6DSO: Magn (gauss): x: 4.363, y: 536871436.025, z: 46801.000
LSM6DSO: Temperature: 44231.9 C
LSM6DSO: Pressure:46352.000 kpa
16:: lsm6dso acc trig 0
16:: lsm6dso gyr trig 0
16:: lsm6dso temp trig 0
```

The above with latest main which fixes other sensor issues.

avisconti, 2021-06-03:
OK, I'll take a look at it

avisconti, 2021-06-03:
> aha, switched now and I am getting output, looks like I am on the surface of the sun :-)
>
> ```
> LIS2DW12: Accel (m.s-2): x: -0.306, y: 0.077, z: 9.571
> LSM6DSO: Accel (m.s-2): x: 0.000, y: 0.000, z: [...]

No, wait, I know the issue!
Values are bad because I think the irq-gpios for lsm6dso in DT is wron for that particular board (i.e. frdm_k64f). Does it make sense to you? So, you may want to check the rq_gpios definitions in standard mode (which seems to work) and in sensorhub mode.

nashif, 2021-06-03:
ok, makes sense. lets close it then.

closed 2021-06-03 by nashif