zephyrproject-rtos/zephyr #35926: Shell tab-completion with more than two levels of nested dynamic commands fails

berendo, 2021-06-03:
**Describe the bug**
Shell commands declared with `SHELL_DYNAMIC_CMD_CREATE()` can further define another level of dynamic commands by setting the `subcmd` of the dynamic entries they return to another command set declared with `SHELL_DYNAMIC_CMD_CREATE()`. Tab-completion with such a deeper nested dynamic command structure fails because in such a scenario [`z_shell_find_cmd()`](https://github.com/zephyrproject-rtos/zephyr/blob/a02704bacf468147d37ad9bb4b67c3a24dd94d02/subsys/shell/shell_utils.c#L287-L302) gets called with the same memory area being pointed to with `parent` (parent command entry, which was returned by a previous dynamic command lookup) and `dloc` (temporary dynamic command lookup entry). As `z_shell_find_cmd()` iterates through the command choices of the next level, the `subcmd` field of the `parent` is overwritten, causing further iteration to jump deeper into the dynamic command lookup.

**To Reproduce**
Building Zephyr with the shell subsystem enabled (with tab-completion support) and declaring a simple set of commands such as:
```
#include <shell/shell.h>

static void bar3_get(size_t idx, struct shell_static_entry *entry);

SHELL_DYNAMIC_CMD_CREATE(bar3, bar3_get);

static void bar3_get(size_t idx, struct shell_static_entry *entry)
{
    static char *names[] = {
        "bar3_1",
        "bar3_2",
        "bar3_3"
    };
    if (idx < sizeof(names)/sizeof(names[0])) {
        entry->syntax = names[idx];
        entry->subcmd = NULL;
    }
    else {
        entry->syntax = NULL;
    }
}

static void bar2_get(size_t idx, struct shell_static_entry *entry);

SHELL_DYNAMIC_CMD_CREATE(bar2, bar2_get);

static void bar2_get(size_t idx, struct shell_static_entry *entry)
{
    static char *names[] = {
        "bar2_1",
        "bar2_2",
        "bar2_3"
    };
    if (idx < sizeof(names)/sizeof(names[0])) {
        entry->syntax = names[idx];
        entry->subcmd = &bar3;
    }
    else {
        entry->syntax = NULL;
    }
}

static void bar1_get(size_t idx, struct shell_static_entry *entry);

SHELL_DYNAMIC_CMD_CREATE(bar1, bar1_get);

static void bar1_get(size_t idx, struct shell_static_entry *entry)
{
    static char *names[] = {
        "bar1_1",
        "bar1_2",
        "bar1_3"
    };
    if (idx < sizeof(names)/sizeof(names[0])) {
        entry->syntax = names[idx];
        entry->subcmd = &bar2;
    }
    else {
        entry->syntax = NULL;
    }
}

SHELL_CMD_REGISTER(foo, &bar1, "", NULL);
```
is sufficient to illustrate the problem. With the above nested dynamic commands declared, the following tab-completion fails:
```
uart:~$ foo bar1_1 bar2_2 [TAB]
```
What would be expected is for `bar3_1`, `bar3_2` and `bar3_3` to be offered as choices, but nothing comes up. However, if the following is tab-completed:
```
uart:~$ foo bar1_1 bar2_1 [TAB]
```
the correct choices are displayed. This is because `bar2_1` is the first entry returned from the second level of dynamic command lookup and it immediately matches the input, so the collateral damage of `subcmd` being overwritten while `z_shell_find_cmd()` is executing is masked.

**Expected behavior**
Shell command tab-completion should work regardless of how many levels of commands are declared and the nesting of dynamic commands.

**Impact**
Significant annoyance.

**Logs and console output**
Snippets provded above.

**Environment (please complete the following information):**
IUbuntu Linux 20.04, Zephyr SDK 0.12.4 and latest `master`.

**Additional context**
This might be readily fixable by having `z_shell_find_cmd()` keep a copy of the incoming `parent` entry before it starts iterating on that entry's `subcmd`.

berendo, 2021-06-03:
One could argue that the source of the issue is one more function call up, in [`z_shell_get_last_command()`](https://github.com/zephyrproject-rtos/zephyr/blob/a02704bacf468147d37ad9bb4b67c3a24dd94d02/subsys/shell/shell_utils.c#L327) where the two parameters passed down to `z_shell_find_cmd()` (`entry` and `dloc`) can alias to the storage and the former is used as the anchor to the current command level and the latter is a temporary entry buffer.

jakub-uC, 2021-06-04:
@berendo Thank you very much for finding this bug. I did not know that anyone is using dynamic commands, especially with dynamic subcommands ;-).

jakub-uC, 2021-06-04:
btw. If you wish to test the example using your code please either add a command handler or add in dynamic handlers:

```
 entry->handler = NULL;
```

closed 2021-06-04 by galak

berendo, 2021-06-04:
@jakub-uC I see that @galak merged the PR associated with this issue, but I want to make sure you saw my notes here https://github.com/zephyrproject-rtos/zephyr/pull/35951#issuecomment-854845754. Regardless, thank you for the quick response.

berendo, 2021-06-06:
Please ignore my previous comment re: the alternate fix proposal - your fix looks good 👍.

PR #35951 (2021-06-04): shell: fix tab for dynamic commands
The tabulator handler creates a single structure if it is handling dynamic
commands. If the currently processed dynamic command has a dynamic
subcommand they both share the same structure.
As a result tabulation operation may result in undefined behaviour.

As a solution, a new structure was introduced to keep subcommand
information.

Fixes #35926.
commit 4a85ecac (2021-06-04): shell: fix tab for dynamic commands

The tabulator handler creates a single structure if it is handling
dynamic commands. If the currently processed dynamic command has a
dynamic subcommand they both share the same structure.
As a result tabulation operation may result in undefined behaviour.

As a solution, a new structure was introduced to keep subcommand
information.

Fixes #35926.