zephyrproject-rtos/zephyr #35930: nRF Dongle as BLE Central Unstable Connectivity at Long-ish Range

jhardyswym, 2021-06-04:
I am developing a project using an nRF52840 dongle as a BLE central to connect and subscribe to a single GATT characteristic on up to 10 remote BLE sensors. 

At this point, everything works as intended and I am able to successfully connect and subscribe to notifications on 10 concurrently. The only problem is when one of the sensors is outside of a certain range (AKA when the RSSI drops too low). The application for this project is going to be entirely indoors (i.e houses), thus my testing setups are as well, and I start having issues when the sensors are placed a few rooms away.

When the RSSI drops below ~ -75 dBm, connecting and subscribing become hit-and-miss, and frequent disconnects occur. I am using the same app flow as the "central_hr" sample, but persisting the connection and adding a whitelist on top of it. 

General flow is Scan->Ad of whitelist device found ->stop scan ->connect to device-> subscribe -> Restart Scan

Obviously Bluetooth in general is pretty short-range, but not being able to persist a stable connection in the next room is pretty bad, and I'm trying to figure out if it's a hardware or firmware issue. 

Here's my prj.conf:

`### System ###
CONFIG_MINIMAL_LIBC=n
CONFIG_NEWLIB_LIBC=y
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y

# Some command handlers require a large stack.
CONFIG_SYSTEM_WORKQUEUE_STACK_SIZE=8192
CONFIG_MAIN_STACK_SIZE=4096
CONFIG_HEAP_MEM_POOL_SIZE=4096
# Allow settings to be stored in the back-end (i.e. Bluetooth information)
CONFIG_SETTINGS=y
CONFIG_SETTINGS_RUNTIME=y
CONFIG_SETTINGS_NONE=y
CONFIG_REBOOT=y
CONFIG_WATCHDOG=y

### Hardware ###
CONFIG_GPIO=y

### Bluetooth ###
CONFIG_BT=y
CONFIG_BT_GATT_DYNAMIC_DB=y
CONFIG_BT_CTLR_COMPANY_ID=0x0077
CONFIG_BT_MAX_CONN=11
CONFIG_BT_HCI=y
CONFIG_BT_CTLR=y
CONFIG_BT_LL_SW_SPLIT=y

# CONFIG_BT_DEBUG_SMP=y
# Allow for large Bluetooth data packets.
# CONFIG_BT_L2CAP_TX_MTU=252
# CONFIG_BT_L2CAP_RX_MTU=252
# CONFIG_BT_RX_BUF_LEN=260
CONFIG_BT_CTLR_TX_PWR_PLUS_8=y
CONFIG_BT_USER_PHY_UPDATE=y
CONFIG_BT_CTLR_PHY_CODED=y
CONFIG_BT_L2CAP_DYNAMIC_CHANNEL=y
# BLE Central settings
CONFIG_BT_CENTRAL=y
CONFIG_BT_GATT_CLIENT=y

# Enable Long connection timeout (10 seconds)
CONFIG_BT_CREATE_CONN_TIMEOUT=10

CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y
CONFIG_CLOCK_CONTROL_NRF_K32SRC_250PPM=y
### DFU ###

# Enable mcumgr.
CONFIG_MCUMGR=y
# Ensure an MCUboot-compatible binary is generated.
CONFIG_BOOTLOADER_MCUBOOT=y

# USB backend is serial device
CONFIG_MCUMGR_SMP_UART=y
# Choose USB serial device for backend (CDC_ACM)
CONFIG_UART_MCUMGR_ON_DEV_NAME="CDC_ACM_1"

# Enable most core commands.
CONFIG_MCUMGR_CMD_IMG_MGMT=y
CONFIG_MCUMGR_CMD_OS_MGMT=y
CONFIG_MCUMGR_CMD_STAT_MGMT=y

# Enable flash operations.
CONFIG_FLASH=y

# Required by the `taskstat` command.
CONFIG_THREAD_MONITOR=y

# Enable statistics and statistic names.
CONFIG_STATS=y
CONFIG_STATS_NAMES=y

CONFIG_MCUBOOT_UTIL_LOG_LEVEL_WRN=y

CONFIG_SHELL=y

### Debugging ###
# Logging adds ~15kB to binary size
CONFIG_LOG=y

# Printk is significantly smaller than logging
CONFIG_PRINTK=n
CONFIG_USE_SEGGER_RTT=y

# Enable console
CONFIG_CONSOLE=y
CONFIG_UART_CONSOLE=y
CONFIG_RTT_CONSOLE=y
CONFIG_SHELL_PROMPT_UART=""
CONFIG_STDOUT_CONSOLE=y
CONFIG_SHELL_VT100_COLORS=n
CONFIG_BOARD_HAS_NRF5_BOOTLOADER=n
CONFIG_USB=y
CONFIG_USB_DEVICE_STACK=y
CONFIG_USB_DEVICE_PRODUCT="Zephyr USB console sample"
CONFIG_USB_COMPOSITE_DEVICE=y
CONFIG_USB_CDC_ACM=y
CONFIG_USB_CDC_ACM_DEVICE_COUNT=2
CONFIG_USB_UART_CONSOLE=y
CONFIG_UART_INTERRUPT_DRIVEN=y
CONFIG_UART_LINE_CTRL=y
CONFIG_UART_SHELL_ON_DEV_NAME="CDC_ACM_0"
CONFIG_SHELL_BACKEND_SERIAL_INIT_PRIORITY=51`

I have tried various other configurations, but none of them seem to fix the issue.

Is there more connection management I need to do at the app level to maintain multiple connections? When enabling BT Debug logs, I never seem to get errors, and just to take the sensors out of the equation, my phone (Oneplus 7 Pro) has no problem connecting to them using nRF Connect at greater ranges.

Is there anything I can do in software to bump up the usable range?

cvinayak, 2021-06-04:
> CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y
CONFIG_CLOCK_CONTROL_NRF_K32SRC_250PPM=y

Any reason for using the internal RC oscillator for the 32KHz source? nRF52840 dongle has a 32KHz crystal, is it damaged in your dongle?

Have you tried using a nRF52840 DK instead for your application?

Plese provide details as to building, flashing, reproducing, commit revisions, toolchain, OS environment etc. (as requested in the github issue template)

jhardyswym, 2021-06-04:
OS Environment: Ubuntu 18.04
ZEPHYR_TOOLCHAIN_VARIANT: zephyr
Build using MCUBoot with a signed application
Flashed over SWD w/ a Nordic Dev kit using `west flash --hex-file build/zephyr/zephyr.signed.hex`

Console Output:
```
00> Ad Found in whitelist [60:A4:23:C9:94:63] rssi: -67
00> 
00> Connected to Sensor 60:A4:23:C9:94:63
00> 
00> Subscribed to state char on 60:A4:23:C9:94:63
00> 
00> Done Configuring Characteristics on 60:A4:23:C9:94:63
00> 
00> Ad Found in whitelist [DD:AC:7B:AE:7B:73] rssi: -71
00> 
00> Connected to Sensor DD:AC:7B:AE:7B:73
00> 
00> Subscribed to state char on DD:AC:7B:AE:7B:73
00> 
00> Done Configuring Characteristics on DD:AC:7B:AE:7B:73
00> 
00> Ad Found in whitelist [D6:F1:7E:2D:B3:2A] rssi: -67
00> 
00> Connected to Sensor D6:F1:7E:2D:B3:2A
00> 
00> Subscribed to state char on D6:F1:7E:2D:B3:2A
00> 
00> Done Configuring Characteristics on D6:F1:7E:2D:B3:2A
00> 
00> Disconnected From Sensor DD:AC:7B:AE:7B:73 reason:0x8
00> 
00> Ad Found in whitelist [60:A4:23:D4:66:53] rssi: -72
00> 
00> Connected to Sensor 60:A4:23:D4:66:53
00> 
00> Subscribed to state char on 60:A4:23:D4:66:53
00> 
00> Done Configuring Characteristics on 60:A4:23:D4:66:53
00> 
00> Ad Found in whitelist [DD:AC:7B:AE:7B:73] rssi: -52
00> 
00> Connected to Sensor DD:AC:7B:AE:7B:73
00> 
00> Subscribed to state char on DD:AC:7B:AE:7B:73
00> 
00> Done Configuring Characteristics on DD:AC:7B:AE:7B:73
00> 
00> Ad Found in whitelist [D4:16:27:EB:9A:F0] rssi: -70
00> 
00> Connected to Sensor D4:16:27:EB:9A:F0
00> 
00> Subscribed to state char on D4:16:27:EB:9A:F0
00> 
00> Done Configuring Characteristics on D4:16:27:EB:9A:F0
00> 
00> Ad Found in whitelist [D4:EB:07:A7:5F:D6] rssi: -73
00> 
00> Connected to Sensor D4:EB:07:A7:5F:D6
00> 
00> Subscribed to state char on D4:EB:07:A7:5F:D6
00> 
00> Done Configuring Characteristics on D4:EB:07:A7:5F:D6
00> 
00> Ad Found in whitelist [60:A4:23:D4:66:53] rssi: -73
00> 
00> [00:08:02.350,433] <wrn> bt_conn: Found valid connection in connected state
00> Disconnected From Sensor 60:A4:23:D4:66:53 reason:0x8
00> 
00> Ad Found in whitelist [60:A4:23:D4:66:53] rssi: -76
00> 
00> Connected to Sensor 60:A4:23:D4:66:53
00> 
00> Subscribed to state char on 60:A4:23:D4:66:53
00> 
00> Done Configuring Characteristics on 60:A4:23:D4:66:53
00> 
00> Disconnected From Sensor D4:16:27:EB:9A:F0 reason:0x8
00> 
00> Ad Found in whitelist [D4:16:27:EB:9A:F0] rssi: -78
00> 
00> Connected to Sensor D4:16:27:EB:9A:F0
00> 
00> Subscribed to state char on D4:16:27:EB:9A:F0
00> 
00> Done Configuring Characteristics on D4:16:27:EB:9A:F0
```
Is there a way to get my zephyr commit version?

cfriedt, 2021-07-27:
@cvinayak - we're unsure how to classify this one - Bug ? Enhancement?

cvinayak, 2021-07-28:
@jhardyswym Please switch to using a nRF52840 Development Kit, which can also have a external antenna connected if need be.

@cfriedt I do not see any implementation issue here.

cfriedt, 2021-09-14:
Closing due to inactivity / unable to reproduce.

closed 2021-09-14 by cfriedt