zephyrproject-rtos/zephyr #35931: bluetooth: controller assertion in ull_master.c

JordanYates, 2021-06-04:
**Describe the bug**
A rare Bluetooth ULL assertion has been observed on Zephyr v2.5.
On Zephyr v2.5 the assertion is at line 720.
On Zephyr v2.6 the assertion is at line 868.

`ticker_start` in `ull_master_setup` is returning `TICKER_STATUS_FAILURE` instead of the expected `TICKER_STATUS_SUCCESS` or `TICKER_STATUS_BUSY`.

The application triggering the asserts is performing the following actions:
* Running a basic connection cycle
    * Initiate a connection to a remote device
    * Perform discovery on the device
    * Write to a characteristic on the device
    * Disconnect from the device
* Starting a Bluetooth advertising set (that runs for 1 or 2 transmits) every few seconds
* Run Bluetooth scanning for 1 second after advertising

The connection cycle is not synchronised with the advertising & scanning, so I would guess that the problem is caused by a rare edge condition when the two actions overlap.

**To Reproduce**
I am still in the process of reproducing this error.
This issue is currently about raising awareness that this assertion can fail.

**Expected behaviour**
Assertions should be checking against impossible conditions, as normal errors should just be handled.
If `ticker_start` returning `TICKER_STATUS_FAILURE` is a fatal error condition, then is shouldn't be possible to get that result under normal operation.

**Impact**
Hard fault when this condition occurs, triggering undesired reboots.

**Logs and console output**
```
ASSERTION FAIL [(ticker_status == 0) || (ticker_status == 2)] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_master.c
[22:06:14.102,020] <err> os: r0/a1:  0x00000003  r1/a2:  0x20003ba0  r2/a3:  0x20003810
[22:06:14.102,020] <err> os: r3/a4:  0x200016e4 r12/ip:  0x00000000 r14/lr:  0x0000876f
[22:06:14.102,020] <err> os:  xpsr:  0x4100001b
[22:06:14.102,020] <err> os: Faulting instruction address (r15/pc): 0x0000877a
[22:06:14.102,020] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[22:06:14.102,050] <err> os: Fault during interrupt handling
[22:06:14.102,050] <err> os: Current thread: 0x20003ae0 (idle 00)
```

**Environment (please complete the following information):**
 - Zephyr v2.5

**Additional context**

#35476 described failures in an identical assertion check in `lll.c`.
The assertion in that case was added after the v2.5 release, so the question is "Is the strict failure check also wrong in this case?"

JordanYates, 2021-06-04:
Caught the error again, the sequence before the assertion was:
1. Connection created and failed to establish
2. Connection retry started
3. Bluetooth advertising started (Period = 30 - 60ms) @ Time T
4. Assertion @ Time (T + 40ms)

So it seems likely that the assertion itself is related to the advertising TX.

```
[00:50:21.964,721] <inf> uc_bt_gatt: Connected to D8:97:90:00:04:59
[00:50:22.216,766] <inf> uc_bt_gatt: Disconnected from D8:97:90:00:04:59 62
<< Connection retry initiated here somewhere
[00:50:22.864,349] <inf> uc_bt_adv: Advertising started
ASSERTION FAIL [(ticker_status == 0) || (ticker_status == 2)] @ WEST_TOPDIR/zephyr/subsys/bluetooth/controller/ll_sw/ull_master.[00:50:22.905,303] <err> os: r0/a1:  0x00000003  r1/a2:  0x2000db80  r2/a3:  0x200028d0
[00:50:22.905,303] <err> os: r3/a4:  0x00000006 r12/ip:  0x00000000 r14/lr:  0x0000876f
[00:50:22.905,303] <err> os:  xpsr:  0x4100001b
[00:50:22.905,303] <err> os: Faulting instruction address (r15/pc): 0x0000877a
[00:50:22.905,334] <err> os: >>> ZEPHYR FATAL ERROR 3: Kernel oops on CPU 0
[00:50:22.905,334] <err> os: Fault during interrupt handling
[00:50:22.905,334] <err> os: Current thread: 0x20003ae0 (idle 00)
```

cvinayak, 2021-06-04:
At anytime, from the ULL HIGH execution context, a maximum of 3 ticker operations (or 4 ticker operations in case of flash driver being enabled) could be enqueued. The assert conveys that the ULL HIGH operations queue is full.

> On Zephyr v2.6 the assertion is at line 868.

Is this issue reproducible in v2.6.0-rc1 ?

> Additional context
>
> #35476 described failures in an identical assertion check in lll.c.
The assertion in that case was added after the v2.5 release, so the question is "Is the strict failure check also wrong in this case?"

No, strict failure check in LLL was a regression in the v2.6 development cycle and does not impact <=2.5 release.

> Environment (please complete the following information):
>
>    Zephyr v2.5

Are you using the flash driver in your application? There have been few (related/unrelated to this issue) critical bug fixes backported into v2.5-branch, do use the v2.5-branch and revert back if the issue is still observed.

JordanYates, 2021-06-04:
> Is this issue reproducible in v2.6.0-rc1 ?

Unfortunately I haven't completed updating our libraries to v2.6 yet so I can't test that yet.

> Are you using the flash driver in your application?

The flash driver is enabled but it is only used at startup in this application.

carlescufi, 2021-06-08:
Setting priority as high since this is an assertion failure.

cvinayak, 2021-06-09:
@JordanYates are you on top of the v2.5-branch?

JordanYates, 2021-06-09:
> @JordanYates are you on top of the v2.5-branch?

No, just the v2.5 release. I wasn't even aware of the v2.5-branch until a few days ago.

I did reduce the frequency of the assertion by changing some code flow so that function calls don't conflict with each other (e.g. scan / connect), but I still see it occasionally.

I have been tracking down other bugs recently, but I will try to give v2.5-branch a go tomorrow.

carlescufi, 2021-06-14:
@JordanYates whenever you have the time to test this would be most helpful.
@cvinayak do you have any clues yet?

JordanYates, 2021-06-15:
I updated my app to crash relatively frequently, and then rebased it on `v2.5-branch`.
I haven't seen any crashes over ~20 hours of runtime, so I'm happy to close this.

For sanity I did instrument the `ticker_start` function and it was showing different behaviour after the rebase, so that is most likely the fix.

closed 2021-06-15 by JordanYates

cvinayak, 2021-06-16:
@JordanYates Thank you, for verifying the v2.5-branch. 

I am not able to definitely  point which one of the commits between 6031f04519 and dabf237584 has solved the assert, but you may feel free to re-open this issue in the future if need be.

joseppc, 2022-06-28:
I am seeing this with a modified version of the central_hr sample running on a nrf51_blenano, which tries to connect to a board running the peripheral_hr sample.
Using the tip of v3.1-branch (at the time of this writing 4ab83425e0ee37786163be128e0c3fb8ad46ddc0)