zephyrproject-rtos/zephyr #35945: SPI4 on nRF5340 not working when using k_sleep() in main

alexsven, 2021-06-04:
**Describe the bug**
When building a release build with no serial, no debug etc. I've noticed that SPI4 won't work if there is a k_sleep() in main. I know this is a very special edge case, but it could be part of a larger issue. 

The bug can be seen using a logic analyzer connected to:
MOSI: P0.09
MISO: P0.10
SCK: P0.08
CS: P0.011

When the k_sleep is present, only the CS-pin toggles

When swapping out SPI4 with SPI1/SPI2/SPI3 everything works, if I use a k_busy_wait() instead of a k_sleep() it also works. And if any sort of debug features are enabled (like serial, debug, log etc) it also works. 

**To Reproduce**
I've attached a very simple sample here so that it can be reproduced with a nRF5340DK (I used a 0.11.0 version)
[spi_fail.txt](https://github.com/zephyrproject-rtos/zephyr/files/6597702/spi_fail.txt) <-- rename to spi_fail.patch
So all is needed is to apply the patch, run:
`west build -b nrf5340dk_nrf5340_cpuapp`
and
`west flash`

I'm testing this in zephyr hash ebfe9be01f, but I have also verified on latest master (e2068d10bc73aca743584205f8bfdfc3c2b3dc18). If using latest master CONFIG_OBJECT_TRACING=n must be removed from prj.conf

koffes, 2021-08-05:
AFAIK this issue is still present. Please remove stale label.

carlescufi, 2021-09-08:
This is valid, running this internally first.

carlescufi, 2021-09-22:
A workaround PR will be posted soon by @anangl.

anangl, 2021-09-24:
After #37632 has been merged, I can no longer reproduce this issue using main. Specifically, commit 2a199afac3c825d00d142d49a1c011eb6a8f0d52 seems to be preventing it. When I revert this commit, the issue appears again.
@alexsven or @koffes could you confirm this?

carlescufi, 2021-09-27:
Closing this as per @anangl's last comment. Please reopen if this reappears.

closed 2021-09-27 by carlescufi