zephyrproject-rtos/zephyr #35964: shell_uart hangs when putting UART into PM_LOW_POWER_STATE

maximevince, 2021-06-04:
**Describe the bug**
The shell thread can hang when putting the UART into PM_LOW_POWER_STATE, before it has finished outputting all characters on the physical UART. (At least on an nRF52)

It seems the shell thread is stuck in `shell_pend_on_txdone()`, polling for the `SHELL_SIGNAL_TXDONE` signal which is never coming, since the UART driver was put into LOW_POWER_STATE.

**To Reproduce**
Steps to reproduce the behavior:
1. Print something to the **uart shell** (printk works fine)
2. Put `uart0` in low power state
3. Go to sleep for some time
4. Put `uart0` into active state again

The code I used for this:
```C
int cmd_sleep(const struct shell *shell, size_t argc, char **argv)
{
    shell_print(shell, "About to enter sleep with UART off. k_sleep in %us\n", SLEEP_S);
    device_set_power_state(uart0, DEVICE_PM_LOW_POWER_STATE, NULL, NULL);

    k_sleep(K_SECONDS(SLEEP_S));

    device_set_power_state(uart0, DEVICE_PM_ACTIVE_STATE, NULL, NULL);
    shell_print(shell, "Awake again!\n");
    return 0;
}
```

I noticed the uart driver is receiving bytes again after wakeup, but those are never parsed by the shell thread, since that one's hanging indefinitely.

**Expected behavior**
I expect the shell to accept commands again after putting the uart driver in active state again (`DEVICE_PM_ACTIVE_STATE`).

**Impact**
Cannot use shell and deep sleep at the same time. 

**Logs and console output**

This is the shell thread backtrace, in case that helps:
```
#0  arch_swap (key=key@entry=0) at /home/vinz/Work/git/snt_zephyr/zephyr/arch/arm/core/aarch32/swap.c:53
#1  0x0002aeec in z_swap_irqlock (key=0) at /home/vinz/Work/git/snt_zephyr/zephyr/kernel/include/kswap.h:153
#2  z_swap (key=..., lock=0x2000545c <kheap.system_heap>) at /home/vinz/Work/git/snt_zephyr/zephyr/kernel/include/kswap.h:164
#3  z_pend_curr (lock=lock@entry=0x2000545c <kheap.system_heap>, key=..., wait_q=wait_q@entry=0x20009eb8 <shell_uart_stack+1784>, timeout=...) at /home/vinz/Work/git/snt_zephyr/zephyr/kernel/sched.c:766
#4  0x0002b95e in z_impl_k_poll (events=0x200028e8 <shell_uart_ctx+728>, num_events=num_events@entry=1, timeout=...) at /home/vinz/Work/git/snt_zephyr/zephyr/kernel/poll.c:289
#5  0x0002e084 in k_poll (timeout=..., num_events=1, events=<optimized out>) at zephyr/include/generated/syscalls/kernel.h:1012
#6  shell_pend_on_txdone (shell=shell@entry=0x36818 <shell_uart>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.c:361
#7  0x0002e1c6 in z_shell_write (shell=0x36818 <shell_uart>, data=0x20002920 <shell_uart_out_buffer>, length=1) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.c:390
#8  0x0002e1d4 in z_shell_print_stream (user_ctx=<optimized out>, data=<optimized out>, len=<optimized out>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.c:398
#9  0x0002dba4 in z_shell_fprintf_buffer_flush (sh_fprintf=sh_fprintf@entry=0x38c70 <shell_uart_fprintf>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_fprintf.c:46
#10 0x00009092 in z_shell_fprintf_fmt (sh_fprintf=0x38c70 <shell_uart_fprintf>, fmt=fmt@entry=0x9093 <z_shell_fprintf_fmt+30> "\347\255\333\002", args=..., args@entry=...) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_fprintf.c:39
#11 0x0002e044 in z_shell_raw_fprintf (ctx=<optimized out>, fmt=0x3e108 "%s") at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.h:24
#12 0x0000935a in z_shell_vt100_color_set (shell=shell@entry=0x36818 <shell_uart>, color=color@entry=SHELL_VT100_COLOR_GREEN) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.c:431
#13 0x0002e22c in z_shell_vfprintf (shell=shell@entry=0x36818 <shell_uart>, color=<optimized out>, fmt=0x3e108 "%s", fmt@entry=0x0 <i2c_dump_msgs>, args=..., args@entry=...) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.c:455
#14 0x0002e252 in z_shell_fprintf (shell=shell@entry=0x36818 <shell_uart>, color=color@entry=SHELL_VT100_COLOR_GREEN, fmt=0x3e108 "%s") at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.c:478
#15 0x000094aa in print_prompt (shell=shell@entry=0x36818 <shell_uart>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.c:339
#16 0x0002e360 in z_shell_print_prompt_and_cmd (shell=shell@entry=0x36818 <shell_uart>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.c:349
#17 0x00008bb8 in state_set (state=SHELL_STATE_ACTIVE, shell=0x36818 <shell_uart>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell.c:96
#18 state_collect (shell=shell@entry=0x36818 <shell_uart>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell.c:955
#19 0x0002daaa in shell_process (shell=0x36818 <shell_uart>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell.c:1391
#20 0x0002d65c in shell_signal_handle (shell=shell@entry=0x36818 <shell_uart>, sig_idx=sig_idx@entry=SHELL_SIGNAL_RXRDY, handler=handler@entry=0x2da5b <shell_process>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell.c:1226
#21 0x00008f00 in shell_thread (shell_handle=0x36818 <shell_uart>, arg_log_backend=<optimized out>, arg_log_level=0x4 <i2c_dump_msgs+4>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell.c:1280
#22 0x0002c1b8 in z_thread_entry (entry=0x8e99 <shell_thread>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /home/vinz/Work/git/snt_zephyr/zephyr/lib/os/thread_entry.c:29
```

**Environment: **
 - OS: Linux
 - Toolchain: Zephyr SDK
 - Zephyr version: v2.4.99-ncs2

**Additional context**
I figured that if I can flush the uart somehow, it might work.
`uart_poll_out()` seems to flush the uart internally (by waiting for the irq lock).

So, this works:
```C
int cmd_sleep(const struct shell *shell, size_t argc, char **argv)
{
    shell_print(shell, "About to enter sleep with UART off. k_sleep in %us\n", SLEEP_S);
    /* Force UART flush */
    uart_poll_out(uart0, '\n');
    device_set_power_state(uart0, DEVICE_PM_LOW_POWER_STATE, NULL, NULL);
    
    k_sleep(K_SECONDS(SLEEP_S));
    
    device_set_power_state(uart0, DEVICE_PM_ACTIVE_STATE, NULL, NULL);
    shell_print(shell, "Awake again!\n");
    return 0;
}
```
But this doesn't fell like a decent solution :)
What if another thread want to write to the shell just before my calling `DEVICE_PM_LOW_POWER_STATE` ?

jakub-uC, 2021-06-07:
@nordic-krch 
Do you think it is a good solution to put UART to sleep mode without deinitializing the shell first?

nordic-krch, 2021-06-07:
Can you try to run it on latest version. Just before 2.6 release there were couple of fixes in nrf uarte driver that should handle power downs.

maximevince, 2021-06-09:
@nordic-krch: oh shoot! It just tried on 2.6.x and that does seem to fix the case where:
```C
    /* first print to shell */
    shell_print(shell, "About to enter sleep with UART off. k_sleep in %us\n", SLEEP_S);
    /* now disable uart, go to sleep, re-enable uart */
    pm_device_state_set(uart0, PM_DEVICE_STATE_LOW_POWER, NULL, NULL);
    k_sleep(K_SECONDS(SLEEP_S));
    pm_device_state_set(uart0, PM_DEVICE_STATE_ACTIVE, NULL, NULL);
```

It still hangs the shell thread indefinitely if you do this:
```C
    /* first disable uart */
    pm_device_state_set(uart0, PM_DEVICE_STATE_LOW_POWER, NULL, NULL);
    /* try to print to shell */
    shell_print(shell, "About to enter sleep with UART off. k_sleep in %us\n", SLEEP_S);
    /* go to sleep, re-enable uart */
    k_sleep(K_SECONDS(SLEEP_S));
    pm_device_state_set(uart0, PM_DEVICE_STATE_ACTIVE, NULL, NULL);
```

But that makes me wonder:
* Probably it should not be possible to hang the shell thread like this
* Trying to print to the shell, after disabling the uart is of course not "smart", but what if this is happening in different threads?
* And, as @jakub-uC said: do we need to disable the shell first? And how do I do this? (Couldn't find documentation on this)

carlescufi, 2021-06-15:
@jakub-uC @nordic-krch any more input to the above?

maximevince, 2021-06-17:
Additionally, I seem to have the same issue with RTT:

```C
#0  shell_pend_on_txdone (shell=shell@entry=0x44e40 <shell_rtt>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.c:369
#1  0x0003c2d2 in z_shell_write (shell=0x44e40 <shell_rtt>, data=0x20004eb4 <shell_rtt_out_buffer>, length=4) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.c:401
#2  0x0003c2e0 in z_shell_print_stream (user_ctx=<optimized out>, data=<optimized out>, len=<optimized out>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.c:409
#3  0x0003bbb8 in z_shell_fprintf_buffer_flush (sh_fprintf=sh_fprintf@entry=0x4a394 <shell_rtt_fprintf>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_fprintf.c:46
#4  0x0001312a in z_shell_fprintf_fmt (sh_fprintf=0x4a394 <shell_rtt_fprintf>, fmt=fmt@entry=0x1312b <z_shell_fprintf_fmt+30> "\347\301\273\003", args=..., args@entry=...) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_fprintf.c:39
#5  0x0003c13c in z_shell_raw_fprintf (ctx=<optimized out>, fmt=0x4a344 "\033[%d%c") at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.h:24
#6  0x0001321e in z_shell_op_cursor_horiz_move (shell=shell@entry=0x44e40 <shell_rtt>, delta=<optimized out>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.c:22
#7  0x00013376 in z_shell_cmd_line_erase (shell=shell@entry=0x44e40 <shell_rtt>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell_ops.c:341
#8  0x0003b5c0 in shell_log_process (shell=0x44e40 <shell_rtt>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell.c:1119
#9  0x0003b682 in shell_signal_handle (shell=shell@entry=0x44e40 <shell_rtt>, sig_idx=sig_idx@entry=SHELL_SIGNAL_LOG_MSG, handler=handler@entry=0x3b59f <shell_log_process>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell.c:1230
#10 0x00012fa2 in shell_thread (shell_handle=0x44e40 <shell_rtt>, arg_log_backend=<optimized out>, arg_log_level=0x4 <i2c_dump_msgs+4>) at /home/vinz/Work/git/snt_zephyr/zephyr/subsys/shell/shell.c:1292
#11 0x00039eb0 in z_thread_entry (entry=0x12f31 <shell_thread>, p1=<optimized out>, p2=<optimized out>, p3=<optimized out>) at /home/vinz/Work/git/snt_zephyr/zephyr/lib/os/thread_entry.c:29
#12 0xaaaaaaaa in ?? ()
```

Where the shell thread is now waiting for `shell_pend_on_txdone(const struct shell *shell)` to complete.

closed 2021-08-31 by github-actions[bot]