zephyrproject-rtos/zephyr #35971: ehl_crb: test_nop  is failing under tests/kernel/common/

yerabolu, 2021-06-04:
**Describe the bug**
kernel.common.nop & kernel.common.nano64.nop tests are failing. 

What have you tried to diagnose or workaround this issue?

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W -p ehl_crb --device-testing --device-serial /dev/ttyUSB0 -T tests/kernel/common/ --west-flash=/home/ztest/EHL_X86_PXE.sh -j 1 -vv
2. See error

**Expected behavior**
All the tests should pass. 

**Logs and console output**
```
*** Booting Zephyr OS build v2.6.0-rc3-33-ga02704bacf46  (delayed boot 500ms) ***
Running test suite common
===================================================================
START - test_bootdelay
PASS - test_bootdelay in 0.1 seconds
===================================================================
START - test_irq_offload
PASS - test_irq_offload in 0.1 seconds
===================================================================
START - test_byteorder_memcpy_swap
PASS - test_byteorder_memcpy_swap in 0.1 seconds
===================================================================
START - test_byteorder_mem_swap
PASS - test_byteorder_mem_swap in 0.1 seconds
===================================================================
START - test_sys_get_be64
PASS - test_sys_get_be64 in 0.1 seconds
===================================================================
START - test_sys_put_be64
PASS - test_sys_put_be64 in 0.1 seconds
===================================================================
START - test_sys_get_be48
PASS - test_sys_get_be48 in 0.1 seconds
===================================================================
START - test_sys_put_be48
PASS - test_sys_put_be48 in 0.1 seconds
===================================================================
START - test_sys_get_be32
PASS - test_sys_get_be32 in 0.1 seconds
===================================================================
START - test_sys_put_be32
PASS - test_sys_put_be32 in 0.1 seconds
===================================================================
START - test_sys_get_be24
PASS - test_sys_get_be24 in 0.1 seconds
===================================================================
START - test_sys_put_be24
PASS - test_sys_put_be24 in 0.1 seconds
===================================================================
START - test_sys_get_be16
PASS - test_sys_get_be16 in 0.1 seconds
===================================================================
START - test_sys_put_be16
PASS - test_sys_put_be16 in 0.1 seconds
===================================================================
START - test_sys_get_le16
PASS - test_sys_get_le16 in 0.1 seconds
===================================================================
START - test_sys_put_le16
PASS - test_sys_put_le16 in 0.1 seconds
===================================================================
START - test_sys_get_le24
PASS - test_sys_get_le24 in 0.1 seconds
===================================================================
START - test_sys_put_le24
PASS - test_sys_put_le24 in 0.1 seconds
===================================================================
START - test_sys_get_le32
PASS - test_sys_get_le32 in 0.1 seconds
===================================================================
START - test_sys_put_le32
PASS - test_sys_put_le32 in 0.1 seconds
===================================================================
START - test_sys_get_le48
PASS - test_sys_get_le48 in 0.1 seconds
===================================================================
START - test_sys_put_le48
PASS - test_sys_put_le48 in 0.1 seconds
===================================================================
START - test_sys_get_le64
PASS - test_sys_get_le64 in 0.1 seconds
===================================================================
START - test_sys_put_le64
PASS - test_sys_put_le64 in 0.1 seconds
===================================================================
START - test_atomic
PASS - test_atomic in 0.1 seconds
===================================================================
START - test_threads_access_atomic
PASS - test_threads_access_atomic in 0.22 seconds
===================================================================
START - test_bitfield
twiddling bits....
PASS - test_bitfield in 0.3 seconds
===================================================================
START - test_bitarray_declare
PASS - test_bitarray_declare in 0.1 seconds
===================================================================
START - test_bitarray_set_clear
PASS - test_bitarray_set_clear in 0.1 seconds
===================================================================
START - test_bitarray_alloc_free
Testing bit array alloc and free with predefined patterns
Testing bit array alloc and free with divisor 1
Testing bit array alloc and free with divisor 2
Testing bit array alloc and free with divisor 4
Testing bit array alloc and free with divisor 8
Testing bit array alloc and free with divisor 16
Testing bit array alloc and free with divisor 32
Testing bit array alloc and free with divisor 64
Testing bit array interval alloc and free
PASS - test_bitarray_alloc_free in 0.44 seconds
===================================================================
START - test_bitarray_region_set_clear
Testing bit array region bit tests
Testing bit array region bit manipulations
PASS - test_bitarray_region_set_clear in 0.8 seconds
===================================================================
START - test_printk
22 113 10000 32768 40000 22
p 112 -10000 -32768 -40000 -22
0x1 0x01 0x0001 0x00000001 0x0000000000000001
0x1 0x 1 0x   1 0x       1
42 42 0042 00000042
-42 -42 -042 -0000042
42 42   42       42
42 42 0042 00000042
255     42    abcdef        42
68719476735 -1 18446744073709551615 ffffffffffffffff
0xcafebabe 0xbeef 0x2a
PASS - test_printk in 0.33 seconds
===================================================================
START - test_timeout_order
PASS - test_timeout_order in 0.127 seconds
===================================================================
START - test_clock_uptime
PASS - test_clock_uptime in 0.12 seconds
===================================================================
START - test_clock_cycle
PASS - test_clock_cycle in 0.3 seconds
===================================================================
START - test_version
PASS - test_version in 0.1 seconds
===================================================================
START - test_multilib
PASS - test_multilib in 0.1 seconds
===================================================================
START - test_thread_context
PASS - test_thread_context in 0.31 seconds
===================================================================
START - test_errno
PASS - test_errno in 0.1 seconds
===================================================================
START - test_ms_time_duration
PASS - test_ms_time_duration in 0.182 seconds
===================================================================
START - test_bounds_check_mitigation
PASS - test_bounds_check_mitigation in 0.1 seconds
===================================================================
START - test_nop
time k_cycle_get_32() takes 284 cycles
arch_nop() takes 0 cycles
Assertion failed at WEST_TOPDIR/zephyr/tests/kernel/common/src/irq_offload.c:173: test_nop: (diff > 0 is false)
arch_nop() takes 0 cpu cycles
FAIL - test_nop in 0.22 seconds
===================================================================
Test suite common failed.
===================================================================
PROJECT EXECUTION FAILED

```

**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain: SDK 12.4
 - Commit SHA: a02704bacf46

yerabolu, 2021-07-08:
@enjiamai: Could you take a look at this issue?

enjiamai, 2021-07-08:
@yerabolu yes, I will submit a fix today, thanks for the reminder.

PR #36846 (2021-08-07): tests: common: skip arch_nop testing on ehl_crb board
Using the NOP instructions to do timing control on the physical board such as
x86 ehl_crb, up squared and intel ADSP board, that doesn't work. It seems like
it can only be used for instruction alignment purposes. We skip this test on this
board because it's not meaningful.

Fixes #35971
commit ee327803 (2021-08-07): tests: common: skip arch_nop testing on some physical x86 board

Using the NOP instructions to do timing control on some physical board
such as ehl_crb, up_squared and intel adsp board, that doesn't work.
It seems like it can only be used for instruction alignment purposes.
We skip this test on this board because it's not meaningful.

Fixes #35971
closed 2021-08-07 by cfriedt