zephyrproject-rtos/zephyr #35978: ESP32 SPI send data hangup

lgl88911, 2021-06-05:
**Describe the bug**
When using esp32 spi to drive st7789v, there will be hangup when sending data

**To Reproduce**
Steps to reproduce the behavior:
1. config st7789v for eps32
esp32.dts
```
&spi3 {
	#address-cells = <1>;
	#size-cells = <0>;
	status = "okay";
	miso-pin = <19>;
	mosi-pin = <23>;
	sclk-pin = <18>;
	csel-pin = <5>;

	/* Sitronix ST7789V LCD */
	st7789v: st7789v@1 {
		compatible = "sitronix,st7789v";
		reg = <1>;
		spi-max-frequency = <80000>; /* 8MHz */
		label = "ST7789V";
		cmd-data-gpios = <&gpio0 22 GPIO_ACTIVE_LOW>;	/* DET */
		reset-gpios = <&gpio0 17 GPIO_ACTIVE_LOW>;	/* RESX reset */
		width = <240>;
		height = <240>;
		x-offset = <0>;
		y-offset = <0>;
		vcom = <0x19>;
		gctrl = <0x35>;
		vrhs = <0x12>;
		vdvs = <0x20>;
		mdac = <0x00>;
		gamma = <0x01>;
		colmod = <0x05>;
		lcm = <0x2c>;
		porch-param = [0c 0c 00 33 33];
		cmd2en-param = [5a 69 02 01];
		pwctrl1-param = [a4 a1];
		pvgam-param = [D0 04 0D 11 13 2B 3F 54 4C 18 0D 0B 1F 23];
		nvgam-param = [D0 04 0C 11 13 2C 3F 44 51 2F 1F 1F 20 23];
		ram-param = [00 F0];
		rgb-param = [CD 08 14];
	};
};
```
prj.conf
```
CONFIG_SPI=y
CONFIG_ST7789V=y
```
2. west build -p -b esp32 zephyrproject/zephyr/samples/drivers/display/
3. west flash
4. After running, you will see that the spi will hangup when writing data

**Expected behavior**
SPI of ESP32 work normal.

**Logs and console output**
```
▒▒*** Booting Zephyr OS build v2.6.0-rc1-300-g6ce0f2ee6606  ***
[00:00:00.448,210] <dbg> display_st7789v: st7789v_reset_display: Resetting display
[00:00:00.479,707] <inf> esp32_spi: CS control inhibited (no GPIO device)
[00:00:00.479,971] <dbg> esp32_spi: transceive: CONFIG First
[00:00:00.481,856] <dbg> esp32_spi: spi_context_buffers_setup: tx_bufs 0x3ffe5250 - rx_bufs (nil) - 1
[00:00:00.491,502] <dbg> esp32_spi: spi_context_buffers_setup: current_tx 0x3ffe5258 (1), current_rx (nil) (0), tx buf/len 0x3ffe5260/1, rx buf/len (nil)/0
[00:00:00.506,001] <dbg> esp32_spi: spi_context_update_tx: tx buf/len (nil)/0
[00:00:00.518,406] <dbg> esp32_spi: spi_context_buffers_setup: tx_bufs 0x3ffe5270 - rx_bufs (nil) - 1
[00:00:00.523,018] <dbg> esp32_spi: spi_context_buffers_setup: current_tx 0x3ffe5278 (1), current_rx (nil) (0), tx buf/len 0x3ffe5280/1, rx buf/len (nil)/0
[... 149 lines cut ...]
Check the code at this time and you will find waiting in spi_esp32_transfer
```
while (!spi_hal_usr_is_done(hal)) {
/* nop */
}
```
As far as I understand it, spi_hal_usr_is_done checks hardware registers. Does this mean that the hardware working status of ESP32 is abnormal?
```
static inline bool spi_ll_usr_is_done(spi_dev_t *hw)
{
     return hw->slave.trans_done;
}
```

When I simply use the following sending process to simulate SPI, there will be no hangup
```
while(1){
send 1
send 4
send 1
send 4
send 1
send 480
}
```

sylvioalves, 2021-07-12:
@lgl88911 Hi, are you able to re-test this with latest main branch?

JonasNorling, 2021-08-12:
I ran into a similar case while trying to drive a ST7735S display with ESP32, and found this bug report.
For me, it appears that SPI often locks up while or after trying to send more than 64 bytes in a transaction. That doesn't always happen, but once it happens it's repeatable with the same software build. In any case, only zeros get transmitted after the first 64 bytes.

A hacky workaround that makes it work for me is to limit the transfer size in the SPI driver:
```
diff --git a/drivers/spi/spi_esp32_spim.c b/drivers/spi/spi_esp32_spim.c
index e848136758..360192716e 100644
--- a/drivers/spi/spi_esp32_spim.c
+++ b/drivers/spi/spi_esp32_spim.c
@@ -63,6 +63,7 @@ static int IRAM_ATTR spi_esp32_transfer(const struct device *dev)
        spi_hal_dev_config_t *hal_dev = &data->dev_config;
        spi_hal_trans_config_t *hal_trans = &data->trans_config;
        size_t chunk_len = spi_context_max_continuous_chunk(&data->ctx);
+       chunk_len = MIN(chunk_len, 64);
 
        /* clean up and prepare SPI hal */
        memset((uint32_t *) hal->hw->data_buf, 0, sizeof(hal->hw->data_buf));
```

This happens on 2.6.0 and on today's main (zephyr-v2.6.0-1955-ga42890fb39).

sylvioalves, 2021-08-12:
@JonasNorling Are you able to re-test that using 72 instead of 64? The max chunk size is limited to 72 in fact. Let me know if you are able to that so I can submit a PR.

PR #37643 (2021-08-12): drivers: spi: set max chunk len for esp32
Max SPI chunk len was missing from the
implementation, causing SPI to hang up in some
conditions.

Fixes #35978
Fixes #37213