zephyrproject-rtos/zephyr #35986: POSIX: multiple definition of posix_types

manoj153, 2021-06-05:
**Describe the bug**
A clear and concise description of what the bug is.
Multiple declaration/conflict between `/sys/_pthreadtypes.h` and `posix_types.h`
I was trying to import and compile an external c source, a posix compliant library to a zephyr project

External library repo: https://github.com/digidotcom/xbee_ansic_library/

What have you tried to diagnose or workaround this issue?
Identify which header file is the conflict. Described above

**To Reproduce**
Clone the external library to the project directory, configure it to compile along the project in top-level cmakelist

**Expected behavior**
Successful build

**Impact**
Build error 

**Logs and console output**
```
il.c.obj.d -o CMakeFiles/app.dir/src/xbee_ansic_library/src/mbedtls/mbedtls_util.c.obj   -c ../src/xbee_ansic_library/src/mbedtls/mbedtls_util.c
In file included from /home/manoj/zephyrproject/zephyr/include/posix/time.h:61,
                 from ../src/xbee_ansic_library/include/mbedtls/platform.h:73,
                 from ../src/xbee_ansic_library/src/mbedtls/mbedtls_util.c:40:
/home/manoj/zephyrproject/zephyr/include/posix/posix_types.h:45:3: error: conflicting types for 'pthread_attr_t'
   45 | } pthread_attr_t;
      |   ^~~~~~~~~~~~~~
In file included from /home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/types.h:223,
                 from /home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:61,
                 from ../src/xbee_ansic_library/include/mbedtls/platform.h:71,
                 from ../src/xbee_ansic_library/src/mbedtls/mbedtls_util.c:40:
/home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_pthreadtypes.h:75:3: note: previous declaration of 'pthread_attr_t' was here
   75 | } pthread_attr_t;
      |   ^~~~~~~~~~~~~~
In file included from /home/manoj/zephyrproject/zephyr/include/posix/time.h:61,
                 from ../src/xbee_ansic_library/include/mbedtls/platform.h:73,
                 from ../src/xbee_ansic_library/src/mbedtls/mbedtls_util.c:40:
/home/manoj/zephyrproject/zephyr/include/posix/posix_types.h:47:15: error: conflicting types for 'pthread_t'
   47 | typedef void *pthread_t;
      |               ^~~~~~~~~
In file included from /home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/types.h:223,
                 from /home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:61,
                 from ../src/xbee_ansic_library/include/mbedtls/platform.h:71,
                 from ../src/xbee_ansic_library/src/mbedtls/mbedtls_util.c:40:
/home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_pthreadtypes.h:32:20: note: previous declaration of 'pthread_t' was here
   32 | typedef __uint32_t pthread_t;            /* identify a thread */
      |                    ^~~~~~~~~
In file included from /home/manoj/zephyrproject/zephyr/include/posix/time.h:61,
                 from ../src/xbee_ansic_library/include/mbedtls/platform.h:73,
                 from ../src/xbee_ansic_library/src/mbedtls/mbedtls_util.c:40:
/home/manoj/zephyrproject/zephyr/include/posix/posix_types.h:58:3: error: conflicting types for 'pthread_mutex_t'
   58 | } pthread_mutex_t;
      |   ^~~~~~~~~~~~~~~
In file included from /home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/types.h:223,
                 from /home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:61,
                 from ../src/xbee_ansic_library/include/mbedtls/platform.h:71,
                 from ../src/xbee_ansic_library/src/mbedtls/mbedtls_util.c:40:
/home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_pthreadtypes.h:154:20: note: previous declaration of 'pthread_mutex_t' was here
  154 | typedef __uint32_t pthread_mutex_t;      /* identify a mutex */
      |                    ^~~~~~~~~~~~~~~
In file included from /home/manoj/zephyrproject/zephyr/include/posix/time.h:61,
                 from ../src/xbee_ansic_library/include/mbedtls/platform.h:73,
                 from ../src/xbee_ansic_library/src/mbedtls/mbedtls_util.c:40:
/home/manoj/zephyrproject/zephyr/include/posix/posix_types.h:62:3: error: conflicting types for 'pthread_mutexattr_t'
   62 | } pthread_mutexattr_t;
      |   ^~~~~~~~~~~~~~~~~~~
In file included from /home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/types.h:223,
                 from /home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:61,
                 from ../src/xbee_ansic_library/include/mbedtls/platform.h:71,
                 from ../src/xbee_ansic_library/src/mbedtls/mbedtls_util.c:40:
/home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_pthreadtypes.h:169:3: note: previous declaration of 'pthread_mutexattr_t' was here
  169 | } pthread_mutexattr_t;
      |   ^~~~~~~~~~~~~~~~~~~
In file included from /home/manoj/zephyrproject/zephyr/include/posix/time.h:61,
                 from ../src/xbee_ansic_library/include/mbedtls/platform.h:73,
                 from ../src/xbee_ansic_library/src/mbedtls/mbedtls_util.c:40:
/home/manoj/zephyrproject/zephyr/include/posix/posix_types.h:67:3: error: conflicting types for 'pthread_cond_t'
   67 | } pthread_cond_t;
      |   ^~~~~~~~~~~~~~
In file included from /home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/types.h:223,
                 from /home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:61,
                 from ../src/xbee_ansic_library/include/mbedtls/platform.h:71,
                 from ../src/xbee_ansic_library/src/mbedtls/mbedtls_util.c:40:
/home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_pthreadtypes.h:176:20: note: previous declaration of 'pthread_cond_t' was here
  176 | typedef __uint32_t pthread_cond_t;       /* identify a condition variable */
      |                    ^~~~~~~~~~~~~~
In file included from /home/manoj/zephyrproject/zephyr/include/posix/time.h:61,
                 from ../src/xbee_ansic_library/include/mbedtls/platform.h:73,
                 from ../src/xbee_ansic_library/src/mbedtls/mbedtls_util.c:40:
/home/manoj/zephyrproject/zephyr/include/posix/posix_types.h:70:3: error: conflicting types for 'pthread_condattr_t'
   70 | } pthread_condattr_t;
      |   ^~~~~~~~~~~~~~~~~~
In file included from /home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/types.h:223,
                 from /home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h:61,
                 from ../src/xbee_ansic_library/include/mbedtls/platform.h:71,
                 from ../src/xbee_ansic_library/src/mbedtls/mbedtls_util.c:40:
/home/manoj/zephyr-sdk-0.12.0/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_pthreadtypes.h:186:3: note: previous declaration of 'pthread_condattr_t' was here
  186 | } pthread_condattr_t;         /* a condition attribute object */
      |   ^~~~~~~~~~~~~~~~~~
ninja: build stopped: subcommand failed.

```

XBee subdirectory `CMakeLists.txt`

```
cmake_minimum_required(VERSION 3.13.1)
project(project_info)
set(TARGET_SCOPE app PRIVATE)

target_compile_definitions(${TARGET_SCOPE} -D POSIX)
target_compile_definitions(${TARGET_SCOPE} -D MBEDTLS_NO_PLATFORM_ENTROPY)

set(INCLUDES    include
                ports/posix/
                
)

target_include_directories(${TARGET_SCOPE} ${INCLUDES})

file(GLOB SRCS_FILES
        "ports/posix/*.c"
        "src/*/*.c"
        )

target_sources(${TARGET_SCOPE} ${SRCS_FILES})

```

**Environment (please complete the following information):**
 - OS: `Ubuntu 20.04.2 LTS`
 - Toolchain: `Zephyr SDK `
 - Commit SHA or Version used: `0.12.0` 

**Additional context**
Add any other context about the problem here.

manoj153, 2021-06-20:
any updates? keen to learn or get some hints what's going on atleast, I can try experimenting for a fix on my end

stephanosio, 2021-06-21:
cc @pfalcon

stephanosio, 2021-06-21:
Reviewing the newlib source code, `pthread_t` and its friends should only be defined by the newlib when `defined(_POSIX_THREADS) || __POSIX_VISIBLE >= 199506`:
https://github.com/zephyrproject-rtos/newlib-cygwin/blob/b1fe4401fdbd0860e0b91227219b15d2e0142b78/newlib/libc/include/sys/_pthreadtypes.h#L21

As far as `tests/posix/common`, which covers pthread, is concerned, it is run with `CONFIG_NEWLIB_LIBC=y` and not having any issues. Neither `_POSIX_THREADS` nor `__POSIX_VISIBLE` is defined in this particular test, making me wonder where/how they are defined in your project.

Also, your `CMakeFiles.txt` does not look right -- it is not using the proper Zephyr primitives.

You should either be incorporating that library as a directory under `lib/` in the Zephyr repository (not recommended since we are moving non-essential libraries out to modules), or as a Zephyr module (recommended).

See the `CMakeLists.txt` files under the relevant directories as to how they should be written (e.g. https://github.com/zephyrproject-rtos/zephyr/blob/e62c3c533a37a4e7a413500a6375a350306e1e0f/lib/gui/lvgl/CMakeLists.txt for `lib/` approach).

stephanosio, 2021-08-13:
As per the assessment above, this does not seem to be a Zephyr-side issue.

Closing due to lack of response. Please re-open this issue if you think otherwise.

closed 2021-08-13 by stephanosio

belussi, 2022-10-17:
Hi. I'm facing exactly the same problem.

If you enable CONFIG_POSIX_API=y compiler sees POSIX structures from the external library and from native support. The problem is that those structures are different.

I have run that test locally and POSIX is always taken from the native support:

`ding C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/pthread_barrier.c.obj
[67/136] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/pthread_mutex.c.obj
[68/136] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/pthread_sched.c.obj
[69/136] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/sleep.c.obj
[70/136] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/clock.c.obj
[71/136] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/pthread_key.c.obj
[72/136] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/semaphore.c.obj
[73/136] Building C object zephyr/lib/posix/CMakeFiles/lib__posix.dir/timer.c.obj`

If I use GNU ARM toolchain and use C++ Standard Library I get the same output as @manoj153.

Amen.

PatrickM-ZS, 2022-11-14:
I'm also running into this issue. Trying to build external library with CMake ExternalProject, passing along Zephyr build parameters, doesn't seem to work well when both newlib and posix api support are enabled. Any guidance from Zephyr devs or hints for further debugging are welcome.

> As far as `tests/posix/common`, which covers pthread, is concerned, it is run with `CONFIG_NEWLIB_LIBC=y` and not having any issues. Neither `_POSIX_THREADS` nor `__POSIX_VISIBLE` is defined in this particular test, making me wonder where/how they are defined in your project.

These are set internally by newlib headers (sys/features.h). The only way I can successfully omit that is with -nostdinc, which is not what I want (I need the other newlib headers, just not the posixtypes).

Maybe a fix is coming up, noticed this PR: https://github.com/zephyrproject-rtos/zephyr/pull/52087

belussi, 2022-11-15:
@PatrickM-ZS Would it be possible to share your prj.conf so I can compare it with my C++ configuration?

cfriedt, 2022-11-15:
I think this is a valid issue. Not sure why it was closed (maybe Stale?). I'm hoping to resolve the Newlib / zephyr POSIX header issue within the next week.

PrzeBrudny, 2022-11-16:
Hi I have tried what @belussi suggested in #52087 and commented conflicted typas in _pthreadtypes.h -> but failed on linking zephyr_pre0.elf with 2k+ errors:

```
[build] Consolidate compiler generated dependencies of target zephyr_pre0
[build] [ 94%] Building C object zephyr/CMakeFiles/zephyr_pre0.dir/misc/empty_file.c.obj
[build] [ 94%] Linking CXX executable zephyr_pre0.elf
[build] /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/libstdc++_nano.a(atexit_arm.o): in function `__aeabi_atexit':
[build] atexit_arm.cc:(.text.__aeabi_atexit+0x0): multiple definition of `__aeabi_atexit'; arch/arch/arm/core/aarch32/libarch__arm__core__aarch32.a(__aeabi_atexit.c.obj):/workspaces/daughterboard-application/build/zephyr/arch/arch/arm/core/aarch32/../../../../../../../platforms/zephyr/zephyr_workspace/zephyr-core/arch/arm/core/aarch32/__aeabi_atexit.c:26: first defined here
[build] /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/libm_nano.a(lib_a-acosl.o): in function `acosl':
[build] acosl.c:(.text.acosl+0x0): multiple definition of `acosl'; /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/libstdc++_nano.a(math_stubs_long_double.o):math_stubs_long_double.cc:(.text.acosl+0x0): first defined here
[build] /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/libm_nano.a(lib_a-asinl.o): in function `asinl':
[build] asinl.c:(.text.asinl+0x0): multiple definition of `asinl'; /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/libstdc++_nano.a(math_stubs_long_double.o):math_stubs_long_double.cc:(.text.asinl+0x0): first defined here
[build] /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/libm_nano.a(lib_a-atan2l.o): in function `atan2l':
[build] atan2l.c:(.text.atan2l+0x0): multiple definition of `atan2l'; /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/libstdc++_nano.a(math_stubs_long_double.o):math_stubs_long_double.cc:(.text.atan2l+0x0): first defined here
[build] /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/12.1.0/../../../../arm-zephyr-eabi/bin/ld.bfd: /opt/zephyr-sdk-0.15.0/arm-zephyr-eabi/arm-zephyr-eabi/lib/thumb/v7e-m/nofp/libm_nano.a(lib_a-atanl.o): in function `atanl':
``` 
...

I am using prj.conf flags:

```
CONFIG_CPLUSPLUS=y
CONFIG_LIB_CPLUSPLUS=y
CONFIG_NEWLIB_LIBC=y
```