zephyrproject-rtos/zephyr #35992: stm32f303k8 device tree missing DACs and pinmux for CAN alternate pins

stoberblog, 2021-06-06:
Hi,

I'm developing with a stm32f303k8 and I've found a few items missing from the device tree. These are both DAC peripherals (DAC1 and DAC2), CAN alternate pins (PB8 and PB9). I also had some issues with timer3 pwm channels.

I'm making a custom board which includes
#include <st/f3/stm32f303X8.dtsi>
#include <st/f3/stm32f303k(6-8)tx-pinctrl.dtsi>
which further intern includes 
#include <st/f3/stm32f3.dtsi>
#include <st/f3/stm32f303.dtsi>

Firstly both DACs are missing:
/ {
       soc {
		/*
		 * The DACs are not down as being in the included
		 * device trees, so we need full details of them
		 */
		dac1: dac@40007400 {
			compatible = "st,stm32-dac";
			reg = <0x40007400 0x400>; // At location, with length
			clocks = <&rcc STM32_CLOCK_BUS_APB1 0x20000000>; // clock, and bit that enables it
			status = "disabled";
			label = "DAC_1";
			#io-channel-cells = <1>;
		};
		
		dac2: dac@40009800 {
			compatible = "st,stm32-dac";
			reg = <0x40009800 0x400>;
			clocks = <&rcc STM32_CLOCK_BUS_APB1 0x04000000>;
			status = "disabled";
			label = "DAC_2";
			#io-channel-cells = <1>;
		};
        };
};

Secondly there are pin-muxes missing from #include <st/f3/stm32f303k(6-8)tx-pinctrl.dtsi>
/ {
       soc {
		pinctrl: pin-controller@48000000 {
			// The below seem to not be in the include (yet are in the datasheet)
			can1_rx_pb8: can1_rx_pb8 {
				pinmux = <STM32_PINMUX('B', 8, AF9)>;
				bias-pull-up;
			};
			can1_tx_pb9: can1_tx_pb9 {
				pinmux = <STM32_PINMUX('B', 9, AF9)>;
			};
		};
        };
};

I also couldn't get my the timer3 pwm pinmuxes to come through so I had to add those to the pinctrl:
/ {
       soc {
		pinctrl: pin-controller@48000000 {
			// These seemed to not be included
			tim3_ch2_pwm_pa7: tim3_ch2_pwm_pa7 {
				pinmux = <STM32_PINMUX('A', 7, AF2)>;
			};
			tim3_ch3_pwm_pb0: tim3_ch3_pwm_pb0 {
				pinmux = <STM32_PINMUX('B', 0, AF2)>;
			};
			tim3_ch4_pwm_pb1: tim3_ch4_pwm_pb1 {
				pinmux = <STM32_PINMUX('B', 1, AF2)>;
			};
		};
        };
};

stoberblog, 2021-06-06:
For reference, the same MCU is used on the Nucleo F303K8 which is a supported board and also what I based my custom board off.

erwango, 2021-06-07:
@stoberblog We' have a PR in preparation to fix the missing CAN nodes. This should be available in few days.
About the pinmux, *-pinctrl.dtsi files are generated from official data base (https://github.com/STMicroelectronics/STM32_open_pin_data). So we need to see if the issue comes from the database or the generation script.

erwango, 2021-06-07:
I just had a look to STM32F303K8 datasheet (DS9866.pdf)
Please note that 'he first letter after 303 refers to Pin count (cf "Chapter 8, Ordering information")
In current case, 'K' means 32 pins.
In that package, PB8 and PB9 are not available, which explain why matching CAN_RX and CAN_TX pinctrl config are not available.

Regarding the tim3 pin configs, I've been able to find them here: 
https://github.com/zephyrproject-rtos/hal_stm32/blob/f8ff8d25aa0a9e65948040c7b47ec67f3fa300df/dts/st/f3/stm32f303k(6-8)tx-pinctrl.dtsi#L246

stoberblog, 2021-06-07:
Ah yes, my mistake for not triple checking as your correct. I had must have copied my configuration from another board (as my schematic is correct) and not realised as the CAN pins matched the pin out section for the PB8 and PB9 in the datasheet for STM32F03x6/x8 (DS9866).

I guess that means the only real issue is the DACs missing in the stm32f3.dtsi or stm32f303.dtsi files then.

PR #36006 (2021-06-10): Add DAC1 & DAC2 in stm32f3
These commits adds DAC1 support in dtsi for stm32f3 series. Some
stm32f3 series has DAC2 nodes for in dtsi for the same where applicable.
DAC2 is present in stm32f303x8 and stm32f373xc.

Fixes #35992
closed 2021-06-10 by galak