zephyrproject-rtos/zephyr #36014: Missing closing parenthesis for soc prototype

koalaByte, 2021-06-07:
https://github.com/zephyrproject-rtos/zephyr/blob/a41fc5165acce3f1cee4d4f754844ad31ba48f0b/dts/arm/st/g0/stm32g050.dtsi#L10

west tool output:
```bash
Error: nucleo_g070rb.dts.pre.tmp:329.1-2 syntax error
FATAL ERROR: Unable to parse input tree
CMake Error at ../zephyr/cmake/dts.cmake:214 (message):
  command failed with return code: 1
```

There's a missing closing parenthesis for **soc** prototype in this file, which causes compilation to fail.
Fixing this should solve the issue. (tested locally)

carlescufi, 2021-06-07:
@koalaByte thanks for the report. Could you send a Pull Request fixing this please?

koalaByte, 2021-06-07:
> @koalaByte thanks for the report. Could you send a Pull Request fixing this please?

Yes.

closed 2021-06-08 by koalaByte