zephyrproject-rtos/zephyr #36026: wolfssl / wolfcrypt

kevanpigott, 2021-06-07:
## Origin

wolfSSL
https://github.com/wolfSSL/wolfssl

## Purpose

Integration of wolfSSL into zephyr as an optional crypto library. has more options than tinyCrypt

## Mode of integration

Updating the auto install script found in the wolfSSL repo under https://github.com/wolfSSL/wolfssl/tree/master/IDE/zephyr
to fit the current working tree of zephyr.

## Description

Has anyone been able to get wolfSSL to work within zephyr lately? the auto install script found in wolfSSL/wolfssl/IDE/zephyr/setup.sh seems to be outdated. I would ask this within the wolfcrypt page, but they seem to be unresponsive. how should I update the paths defined in setup.sh and lib/install_lib.sh to work with the current zephyr version?

# Dependencies

wolfSSL

galak, 2021-06-16:
@kevanpigott You might want to ask this on the #general channel in slack.  I'm not aware of anyone that has integrated Zephyr and WolfSSL.

closed 2021-08-30 by github-actions[bot]

gojimmypi, 2022-06-05:
I'm a little late to the party here, but just a heads-up that Zephyr is fully supported in [wolfSSL](https://www.wolfssl.com/). The codebase is highly configurable and can have a considerably smaller footprint.

See [wolfSSL Zephyr  issues](https://github.com/wolfSSL/wolfssl/issues?q=is%3Aissue+zephyr) and [PRs](https://github.com/wolfSSL/wolfssl/issues?q=is%3APR+zephyr+). The most recent PR was merged just a couple of weeks ago.

Please [open a wolfSSL issue](https://github.com/wolfSSL/wolfssl/issues/new/choose) or contact wolfSSL at [support@wolfssl.com](mailto:support@wolfssl.com) with any questions, bug fixes, or suggested feature additions.