zephyrproject-rtos/zephyr #36030: West warnings (and others?) are ignored when building documentation

marc-hb, 2021-06-07:
**Describe the bug**

West warnings (and others?) are ignored when building documentation

**To Reproduce**
Break the following file like this:

```diff
--- a/scripts/west_commands/runners/bossac.py
+++ b/scripts/west_commands/runners/bossac.py
@@ -5,6 +5,9 @@
 
 '''bossac-specific runner (flash only) for Atmel SAM microcontrollers.'''
 
+
+import missing_stuff
+
```

Build the documentation:
```
cd zephyr
make -C doc
```

The following warning appears yet the documentation build completes successfully:

```
The module for runner "bossac" could not be imported (No module named 'missing_stuff'). This most likely means it is not handling its dependencies properly. Please report this to the zephyr developers.
```

**Expected behavior**
The build should fail because sphinx and other warnings are either explicitly filtered or treated as errors.

**Impact**
Hard to tell, depends on which warnings are ignored.

**Logs and console output**
See above.

**Environment (please complete the following information):**
Documentation for Zephyr commit `v2.6.0-rc3` built on Linux.

**Additional context**
Commit 442f631af1f6 and before may be related.

Note you will most likely miss that warning (which is part of the problem!) because the documentation build is quite verbose. To see it better and test it faster, trigger an incremental build like this:
```
printf '\n' >> doc/guides/west/build-flash-debug.rst # makes sure bossac.py is needed again
make -C doc/
```

See more info in PR #33448

gmarull, 2021-06-08:
This is an expected behavior. If west outputs a Python warning (not a Sphinx warning) when it fails to import we can’t do much about it. If we want to make docs fail if this happens then the runners init file should have a “fail mode” than can be activated. Since the installed requirements for docs have not changed, this warning would have been catched by the old mechanism if supported, so I don’t think it is a regression, and possibly should not be considered a bug (I can’t find docs derived from runners docstrings).

mbolivar-nordic, 2021-06-08:
> I can’t find docs derived from runners docstrings

There are none for individual runners as those are implementation details. Only the runner core has docstrings pulled into here: https://docs.zephyrproject.org/latest/guides/west/build-flash-debug.html#hacking

There would be a build error if that didn't work.

So I agree with @gmarull that this isn't really a docs bug.

closed 2021-06-08 by mbolivar-nordic