zephyrproject-rtos/zephyr #36033: Mere warnings slow down incremental documentation build from seconds to minutes

marc-hb, 2021-06-07:
**Describe the bug**

Making a typo and trying to build the documentation make it last minutes instead of seconds. Even a simple warning seems to break the incremental build.

I don't remember this performance issue happening in the past. Regression caused by #33934 maybe?

**To Reproduce**
```
$ cd zephyr
$ time make -C doc # a few minutes as expected
$ time make -C doc # a couple seconds as expected: incremental build.
```
```
$ printf '\n\nsome text\n' >> doc/guides/west/build-flash-debug.rst
$ time make -C doc # incremental build takes seconds, nice
```
Now insert a warning in /doc/guides/west/build-flash-debug.rst (or some other file)
```
$ printf '`missing quote\n' >> doc/guides/west/build-flash-debug.rst
$ time make -C doc # fails instantly the first time, nice
$ time make -C doc # takes a few minutes again, not nice
```
Fixing the typo immediately after the first failure does not help: it does not make the second build fast.

**Expected behavior**

Making a typo when writing documentation should no take minutes.

**Impact**

Serious loss of time and strong incentive to cut documentation corners.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
Linux

 - Commit SHA or Version used
v2.6.0-rc3

gmarull, 2021-06-08:
I have checked and I can reproduce the behavior:
```
$ time make -C doc
make -C doc  1289.84s user 21.03s system 654% cpu 3:20.32 total
...
$ printf '\n\nsome text\n' >> doc/guides/west/build-flash-debug.rst 
$ time make -C doc
...
make -C doc  11.55s user 0.60s system 111% cpu 10.898 total
$ printf '`missing quote\n' >> doc/guides/west/build-flash-debug.rst
$ time make -C doc
...
Warning, treated as error:
/home/gmarull/ws/teslabs/zephyr/zephyr/doc/_build/src/guides/west/build-flash-debug.rst:621:Inline interpreted text or phrase reference start-string without end-string.
...
make -C doc  5.47s user 0.40s system 124% cpu 4.712 total
$ time make -C doc
...
make -C doc  1293.58s user 22.02s system 637% cpu 3:26.36 total
```

I can confirm that the external_content extension is doing its job properly, that is, only copying the changed file. What is likely happening is that Sphinx clears the environment before building and so the previous state is lost when the failure happens. I'd report an issue upstream, we can't do much about that.

marc-hb, 2021-06-08:
@gmarull I unfortunately can't date this but I'm pretty sure this is a (IMHO serious) regression. I didn't test this specifically but I can't imagine I never made any `.rst` mistakes in the past.

gmarull, 2021-06-08:
> @gmarull I unfortunately can't date this but I'm pretty sure this is a (IMHO serious) regression. I didn't test this specifically but I can't imagine I never made any `.rst` mistakes in the past.

I think I have found the reason: the usage of the `-W` flag. Now it is set by default, meaning that on first warning the build will error and so environment will be _lost_. We can enforce this flag just on CI, but not set it as default. There is also the option to activate `-W` but let the build finish, but that's not the best option for CI, where we want to fail fast.

PR #36045 (2021-06-08): doc: only enable -W option on CI
The usage of -W may lead to the loss of the Sphinx build environment
even for small typos. Remove this option from the defaults but still
enable it on CI, where the fail-fast behavior given by -W is desired.

Fixes #36033
commit a140a249 (2021-06-08): doc: only enable -W option on CI

The usage of -W may lead to the loss of the Sphinx build environment
even for small typos. Remove this option from the defaults but still
enable it on CI, where the fail-fast behavior given by -W is desired.

Fixes #36033
closed 2021-06-08 by galak