zephyrproject-rtos/zephyr #36035: struct devices should be allocated in ROM, not RAM

mbolivar-nordic, 2021-06-08:
**Somebody please prove me wrong by showing me I'm missing something obvious...**

Pull request #24873 marked all devices const as a fix for issue #27399. The problem description says:

> At this point all device instance could be made constant to save RAM.

However, despite being const, devices are still linked into RAM, at least on Arm Cortex-M:

```
$ west build -b nrf52840dk_nrf52840 samples/hello_world/ 

$ arm-none-eabi-gdb build/zephyr/zephyr.elf -q -ex 'x __device_start' -ex q
Reading symbols from build/zephyr/zephyr.elf...
0x20000040 <__device_dts_ord_26>:	0x0000406a
```

Note 0x20000000 is the base address of SRAM in the SoC memory map.

It seems that #24873 did not actually save any RAM for devices. 

Since the device related linker section is in include/linker/common-ram.h:

https://github.com/zephyrproject-rtos/zephyr/blob/efccfcbfc91501cac2ede0541a48c338f2b0cae4/include/linker/common-ram.ld#L25

I'm guessing this bug applies to most if not all targets.

**Describe the solution you'd like**

I'm not sure if this is just an oversight or if there's something else blocking the fix, but I moved the linker section to ROM and it seems to work in basic testing, and produces the RAM savings I expected: https://github.com/zephyrproject-rtos/zephyr/pull/36032.

dleach02, 2021-06-08:
Generic linker scripts usually put .text and .ro together in the same section or ROM in this case. Looking at your PR, I'm guessing our linking is overriding that because of where these are defined. I wonder how many other const data items are ending up in RAM?

closed 2021-06-09 by carlescufi

mbolivar-nordic, 2021-06-09:
> Looking at your PR, I'm guessing our linking is overriding that because of where these are defined.

Yes, that's correct.

> I wonder how many other const data items are ending up in RAM?

They would have to be const and specifically placed into a linker section that ends up in RAM. "Normal" static const data without an explicit section attribute ends up in ROM.

You'd have to audit the linker scripts to find the answer, which is probably nontrivial due to Zephyr's linker snippets CMake API.

PR #36032 (2021-06-09): allocate struct devices in ROM
Move the linker sections where devices end up into ROM, saving RAM since the devices themselves are all const qualified and therefore should not be mutated.

Unless I'm missing something ...

Fixes: #36035

Results of:

```
$ west build -b nrf52840dk_nrf52840 samples/hello_world/
```

Without this patch:

```
Memory region         Used Size  Region Size  %age Used
           FLASH:       16860 B         1 MB      1.61%
            SRAM:        5600 B       256 KB      2.14%
        IDT_LIST:          0 GB         2 KB      0.00%
```

With this patch:

```
Memory  [... cut]