zephyrproject-rtos/zephyr #36037: bt_init returning success when Bluetooth initialization does not get finalized.

pigeonhands, 2021-06-08:
**Describe the bug**
`bt_enable` is returning success when not initialized.
Caused by [`bt_init` returning early](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/bluetooth/host/hci_core.c#L3435) and [not calling `bt_finalize_init`](https://github.com/zephyrproject-rtos/zephyr/blob/38eaabc9c75c9a0932bbd272952dc81a2f2939ef/subsys/bluetooth/host/hci_core.c#L3441) if `CONFIG_BT_SETTINGS=y` is and `bt_dev.id_count` is 0.

**To Reproduce**
Steps to reproduce the behavior:
1. set `CONFIG_BT_SETTINGS=y`
2. call `bt_enable` whoult calling `settings_load`

**Expected behavior**
Either returning a non-zero error code if this is an error condition, or finalizing the Bluetooth initializing if it is not.

**Impact**
It causes functions like `bt_le_ext_adv_start` to return `EAGAIN` because the `BT_DEV_READY` [does not get set](https://github.com/zephyrproject-rtos/zephyr/blob/38eaabc9c75c9a0932bbd272952dc81a2f2939ef/subsys/bluetooth/host/hci_core.c#L3407) even though it seems that Bluetooth got enabled successfully.

**Environment (please complete the following information):**
 - OS: Windows
 - Toolchain: NRF Sdk
 - v1.5.0

joerchan, 2021-06-16:
@pigeonhands This is the intended behavior, If you enable BT_SETTINGS you have to call settings_load.
Otherwise the stack does not have any identities, which is needed to start the advertiser.

pigeonhands, 2021-06-16:
> @pigeonhands This is the intended behavior, If you enable BT_SETTINGS you have to call settings_load.
> Otherwise the stack does not have any identities, which is needed to start the advertiser.

Should'nt bt_init return an error if called before settings_load then?

joerchan, 2021-06-16:
> Should'nt bt_init return an error if called before settings_load then?

No, calling settings_load after bt_enable is supported, and what most of the samples do.
If you see the documentation for `bt_id_create` you can see that the order of these matters when it comes to the handling of the identities.

PR #36287 (2021-06-17): Bluetooth: host: Document settings_load is required to finalize init
Document that calling settings_load is required when the application is
not creating and managing the identities of the stack itself.
The application will not be able to use all features of the stack before
the identities have been loaded.

Fixes: #36037
closed 2021-06-17 by galak