zephyrproject-rtos/zephyr #36038: iotdk: the testcase samples/modules/nanopb can't build

Zhaoningx, 2021-06-08:
**Describe the bug**
The testcase can't build on iotdk platform, and it works to run others cases.

**To Reproduce**
Steps to reproduce the behavior:
twister -p iotdk --device-testing --device-serial /dev/ttyUSB0 -T samples/modules/nanopb/

**see error**
Including boilerplate (Zephyr base): /home/ztest/zephyrproject/zephyr/cmake/app/boilerplate.cmake
-- Application: /home/ztest/zephyrproject/zephyr/samples/modules/nanopb
-- Zephyr version: 2.6.99 (/home/ztest/zephyrproject/zephyr), build: zephyr-v2.6.0-41-g9143f4fd8c35
-- Found Python3: /usr/bin/python3.9 (found suitable exact version "3.9.0") found components: Interpreter
-- Found west (found suitable version "0.10.0", minimum required is "0.7.1")
-- Board: iotdk
-- Cache files will be written to: /home/ztest/.cache/zephyr
-- Using toolchain: zephyr 0.12.4 (/home/ztest/zephyr-sdk-0.12.4)
-- Found dtc: /home/ztest/zephyr-sdk-0.12.4/sysroots/x86_64-pokysdk-linux/usr/bin/dtc (found suitable version "1.5.0", minimum required is "1.4.6")
-- Found BOARD.dts: /home/ztest/zephyrproject/zephyr/boards/arc/iotdk/iotdk.dts
-- Generated zephyr.dts: /home/ztest/zephyrproject/zephyr/twister-out/iotdk/samples/modules/nanopb/sample.modules.nanopb/zephyr/zephyr.dts
-- Generated devicetree_unfixed.h: /home/ztest/zephyrproject/zephyr/twister-out/iotdk/samples/modules/nanopb/sample.modules.nanopb/zephyr/include/generated/devicetree_unfixed.h
-- Generated device_extern.h: /home/ztest/zephyrproject/zephyr/twister-out/iotdk/samples/modules/nanopb/sample.modules.nanopb/zephyr/include/generated/device_extern.h
Parsing /home/ztest/zephyrproject/zephyr/Kconfig
Loaded configuration '/home/ztest/zephyrproject/zephyr/boards/arc/iotdk/iotdk_defconfig'
Merged configuration '/home/ztest/zephyrproject/zephyr/samples/modules/nanopb/prj.conf'
Configuration saved to '/home/ztest/zephyrproject/zephyr/twister-out/iotdk/samples/modules/nanopb/sample.modules.nanopb/zephyr/.config'
Kconfig header saved to '/home/ztest/zephyrproject/zephyr/twister-out/iotdk/samples/modules/nanopb/sample.modules.nanopb/zephyr/include/generated/autoconf.h'
-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ztest/zephyr-sdk-0.12.4/arc-zephyr-elf/bin/arc-zephyr-elf-gcc
CMake Error at ../../../modules/nanopb/CMakeLists.txt:11 (message):
  'protoc' not found, please ensure protoc is installed and in path.  See
  https://docs.zephyrproject.org/latest/samples/modules/nanopb/README.html

-- Configuring incomplete, errors occurred!
See also "/home/ztest/zephyrproject/zephyr/twister-out/iotdk/samples/modules/nanopb/sample.modules.nanopb/CMakeFiles/CMakeOutput.log".
See also "/home/ztest/zephyrproject/zephyr/twister-out/iotdk/samples/modules/nanopb/sample.modules.nanopb/CMakeFiles/CMakeError.log".

Environment
OS: Fedora33
Toolchain:zephyr-sdk-0.12.4
Commit ID: 9143f4fd8c3588

galak, 2021-06-08:
It looks like you don't have `protoc` installed

From scripts/requirements-extras.txt

```
# used by nanopb module to generate sources from .proto files
protobuf
```

carlescufi, 2021-06-08:
```
sudo apt install protobuf-compiler
```

closed 2021-06-08 by carlescufi

Zhaoningx, 2021-06-09:
> ```
> sudo apt install protobuf-compiler
> ```

Thanks, it works normall.