zephyrproject-rtos/zephyr #36048: Cannot establish ISO CIS connection properly after ACL disconnected several times

rick1082, 2021-06-08:
**Describe the bug**

**To Reproduce**
Steps to reproduce the behavior:
1. Pull the current master bfd271932228f8ce33b68b82ffee5ee3b2386a17
2. Prepare two nRF5340-DK
3. Modify the ```device_found``` function in central_iso for connecting the specific peripheral, which is easier to test
```
char *target_mac = "EA:98:D0:EE:D9:94"; //TODO: Please replace the MAC address
static void device_found(const bt_addr_le_t *addr, int8_t rssi, uint8_t type,
                         struct net_buf_simple *ad)
{
        char addr_str[BT_ADDR_LE_STR_LEN];
        int err;

        if (default_conn) {
                return;
        }

        /* We're only interested in connectable events */
        if (type != BT_GAP_ADV_TYPE_ADV_IND &&
            type != BT_GAP_ADV_TYPE_ADV_DIRECT_IND) {
                return;
        }

        bt_addr_le_to_str(addr, addr_str, sizeof(addr_str));
        if(strncmp(addr_str, target_mac, 17) == 0){
                printk("GOT target device\n");

                if (bt_le_scan_stop()) {
                        return;
                }

                err = bt_conn_le_create(addr, BT_CONN_LE_CREATE_CONN,
                                        BT_LE_CONN_PARAM_DEFAULT, &default_conn);
                if (err) {
                        printk("Create conn to %s failed (%u)\n", addr_str, err);
                        start_scan();
                }
        }
}
```
4. Program the first nRF5340-DK with central_iso example (zephyr/samples/bluetooth/central_iso)
5. Program the second nRF5340-DK with peripheral_iso example (zephyr/samples/bluetooth/peripheral_iso)
6. Wait central connected with peripheral, and established ISO channel for sending dummy data
7. Reset the peripheral during central sending data, the central will try to connect with peripheral
8. For the first disconnected and then connected, we saw the following log in central 
```
GOT target device                                                               
Connected: EA:98:D0:EE:D9:94 (public)                                           
ISO Channel 0x20002044 connected                                                
ISO Channel 0x20002044 disconnected (reason 0x08)                               
Disconnected: EA:98:D0:EE:D9:94 (public) (reason 0x08)                          
Scanning successfully started                                                   
[00:00:09.733,856] <wrn> bt_hci_core: opcode 0x206f status 0x12                 
GOT target device                                                               
Connected: EA:98:D0:EE:D9:94 (public)                                    
```
9. And after several trials, we can see the following log, and we cannot establish the ISO channel anymore
```
Scanning successfully started                                                   
GOT target device                                                               
Connected: EA:98:D0:EE:D9:94 (public)                                           
Failed to bind iso to connection (-5)                                           
[00:00:16.895,599] <wrn> bt_hci_core: opcode 0x206f status 0x0c                 
[00:00:16.895,904] <wrn> bt_hci_core: opcode 0x206f status 0x12                 
[00:00:17.223,297] <wrn> bt_hci_core: opcode 0x2062 status 0x0c                 
Disconnected: EA:98:D0:EE:D9:94 (public) (reason 0x08)                                      
```
10. If we only reset central, then we will see the following log from peripheral. we will see the following log
[... 91 lines cut ...]
[00:00:54.255,584] <dbg> bt_iso.bt_iso_cleanup: 0x20000dd8                                                                              
[00:00:54.577,697] <wrn> bt_hci_core: opcode 0x2062 status 0x0c                                                                         
[00:00:54.577,697] <dbg> bt_iso.bt_iso_cleanup: 0x20000dd8                                                                              
Disconnected: EA:98:D0:EE:D9:94 (public) (reason 0x3e)                                                                                  
Scanning successfully started                                                                                                           
GOT target device                                                                                                                       
Connected: EA:98:D0:EE:D9:94 (public)                                                                                                   
Failed to bind iso to connection (-12)                                                                                                  
Disconnected: EA:98:D0:EE:D9:94 (public) (reason 0x3e)                                                                                  
Scanning successfully started                                                                                                           
GOT target device                                                                                                                       
Connected: EA:98:D0:EE:D9:94 (public)                                                                                                   
Failed to bind iso to connection (-12)                                                                                                  
Disconnected: EA:98:D0:EE:D9:94 (public) (reason 0x3e)                                                                                  
Scanning successfully started                                                                                                           
GOT target device           
```

**Environment (please complete the following information):**
 - OS: 18.04.1-Ubuntu x86_64
 - Toolchain: arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
 - hash number: bfd271932228f8ce33b68b82ffee5ee3b2386a17

**Additional context**
Sometimes this issue can be replicated easily by reseting peripheral for only once, but most of time we need to reset the peripheral for ~10 times to replicate this issue.

Thalley, 2021-06-11:
Hi @rick1082 I've been looking into this issue and have managed reproduce the issue when we get an error when creating the CIG after connection. 

The issue appears if the central creates a connection and sets up a CIS, but where the connection actually isn't established yet (because the central will get the connection complete event regardless). 

Apparently this triggers some issue in the controller, as central believes that we have an ACL and an ISO, but where neither is seemingly actually established. 

I think this issue should be forwarded to the controller's implementers (which in this case is not Zephyr related). 

Exert from my HCI log that shows that the connection failed to be established and that causes issue for the CIG creation afterwards:

```
< HCI Command: LE Create Connection (0x08|0x000d) plen 25                                             #229 20.220500
        Scan interval: 60.000 msec (0x0060)
        Scan window: 60.000 msec (0x0060)
        Filter policy: White list is not used (0x00)
        Peer address type: Random (0x01)
        Peer address: 70:F4:5C:18:DE:1B (Resolvable)
        Own address type: Public (0x00)
        Min connection interval: 30.00 msec (0x0018)
        Max connection interval: 50.00 msec (0x0028)
        Connection latency: 0 (0x0000)
        Supervision timeout: 4000 msec (0x0190)
        Min connection length: 0.000 msec (0x0000)
        Max connection length: 0.000 msec (0x0000)
> HCI Event: Command Status (0x0f) plen 4                                                             #230 20.224300
      LE Create Connection (0x08|0x000d) ncmd 1
        Status: Success (0x00)
> HCI Event: LE Meta Event (0x3e) plen 19                                                             #231 20.795100
      LE Connection Complete (0x01)
        Status: Success (0x00)
        Handle: 0
        Role: Master (0x00)
        Peer address type: Random (0x01)
        Peer address: 70:F4:5C:18:DE:1B (Resolvable)
        Connection interval: 30.00 msec (0x0018)
        Connection latency: 0 (0x0000)
        Supervision timeout: 4000 msec (0x0190)
        Master clock accuracy: 0x07
= Note: Connected: 70:F4:5C:18:DE:1B (random)                                                              20.798300
< HCI Command: LE Set Connected Isochronous Group Parameters (0x08|0x0062) plen 24                    #232 20.803000
        CIG ID: 0x03
        Master to Slave SDU Interval: 10000 us (0x002710)
        Slave to Master SDU Interval: 10000 us (0x002710)
        SCA: 201 - 500 ppm (0x00)
        Packing: Sequential (0x00)
        Framing: Unframed (0x00)
        Master to Slave Maximum Latency: 10 ms (0x000a)
        Slave to Master Maximum Latency: 10 ms (0x000a)
        Number of CIS: 1
        CIS ID: 0x00
        Master to Slave Maximum SDU Size: 251
        Slave to Master Maximum SDU Size: 0
        Master to Slave PHY: LE 2M (0x02)
        Slave to Master PHY: LE 2M (0x02)
        Master to Slave Retransmission attempts: 0x01
        Slave to Master Retransmission attempts: 0x00
> HCI Event: Command Complete (0x0e) plen 8                                                           #233 20.806900
      LE Set Connected Isochronous Group Parameters (0x08|0x0062) ncmd 1
        Status: Success (0x00)
        CIG ID: 0x03
        Number of Handles: 1
        Connection Handle: 2
< HCI Command: LE Create Connected Isochronous Stream (0x08|0x0064) plen 5                            #234 20.808900
        Number of CIS: 1
        CIS Handle: 2
        ACL Handle: 0
> HCI Event: Command Status (0x0f) plen 4                                                             #235 20.810800
      LE Create Connected Isochronous Stream (0x08|0x0064) ncmd 1
        Status: Success (0x00)
< HCI Command: LE Read Remote Used Features (0x08|0x0016) plen 2                                      #236 20.812500
[... 103 lines cut ...]
        Framing: Unframed (0x00)
        Master to Slave Maximum Latency: 10 ms (0x000a)
        Slave to Master Maximum Latency: 10 ms (0x000a)
        Number of CIS: 1
        CIS ID: 0x00
        Master to Slave Maximum SDU Size: 251
        Slave to Master Maximum SDU Size: 0
        Master to Slave PHY: LE 2M (0x02)
        Slave to Master PHY: LE 2M (0x02)
        Master to Slave Retransmission attempts: 0x01
        Slave to Master Retransmission attempts: 0x00
> HCI Event: Command Complete (0x0e) plen 8                                                           #258 21.192000
      LE Set Connected Isochronous Group Parameters (0x08|0x0062) ncmd 1
        Status: Command Disallowed (0x0c)
        CIG ID: 0x04
        Number of Handles: 1
        Connection Handle: 0
< HCI Command: LE Remove Connected Isochronous Group (0x08|0x0065) plen 1                             #259 21.194000
        CIG ID: 0x04
> HCI Event: Command Complete (0x0e) plen 5                                                           #260 21.195500
      LE Remove Connected Isochronous Group (0x08|0x0065) ncmd 1
        Status: Success (0x00)
        CIG ID: 0x04
= Note: Failed to bind iso to connection (-5)                                                              21.197300
```

carlescufi, 2021-06-17:
@Thalley is the issue here on the Host or Controller side?

Thalley, 2021-06-17:
@carlescufi My assumption is that it is on the controller side. We are currently in dialog with the controller supplier for the controller used here, to fully determine the issue, but at least from the HCI logs it looks like a controller-related issue.

Thalley, 2021-06-24:
This has been confirmed as a 3rd party controller issue. 

TBD if we need to handle this in the Zephyr host. 

@Vudentz @joerchan @jhedberg 
This issue is the one mentioned in slack a few days ago, where we get the ISO and ACL disconnect events in an order not properly handled by the Zephyr host. The core spec doesn't explicitly state an order but implies that the ISO disconnect events should come first (but may not), but also states that if the ACL disconnect event comes first, the host can assume that connected ISO (CIS) are disconnected. 

If we get an ACL disconnect event, we cleanup everything related to ACL and CIS, including removing the CIG. The above issue comes from us removing the CIG before we get the CIS disconnect events, and the host is not allowed to remove a CIG with active (non-disconnected) CIS.

joerchan, 2021-06-28:
@Thalley So what can be done about this? Should we just handle the it when the events comes in the unexpected order? Or should we raise an issue with specification on this?

Thalley, 2021-06-28:
> @Thalley So what can be done about this? Should we just handle the it when the events comes in the unexpected order? Or should we raise an issue with specification on this?

Excellent question :D 

Based on the specification, the there's no explicit order, and while there's a recommended order (described by a `should`), there's an MSC showing the ACL disconnect event coming first. 

One interesting part is that the controller that causes this issue considers the ISO connected when it sends the ACL disconnect to the host, which I think is arguably wrong for them to do (they are looking into it). 

We, as the Zephyr host, could implement support to handle the orders in any event, as well as waiting to remove the CIG until after we have received the disconnect event from all CIS. We have already tested with a quick workaround that simply does nothing to the ISO channels on an ACL disconnect event, which seems to take care of this specific issue, but in reality we should be marking the ISO channels as being "Disconnecting" such that any request from the application would be denied. 

So I propose we do implement something like the following when the ACL disconnect event comes before the last ISO disconnect event:

1. Mark the remaining ISO channels as disconnected
2. Call the disconnected callback on the remaining ISO channels
3. Once all ISO channels in the CIG has be disconnected via the HCI disconnect events, then remove the CIG

One thing to consider for the above, is that we may not be able to create a new CIG before the previous CIG was properly disconnected. The current API works with individual channels and does not have a concept for a group (CIG), so we might need a way to tell the above layers when the CIG has been removed an another one can be created.

rick1082, 2021-07-09:
The behavior from Zephyr host is correct, we fixed this issue in controller side. Thanks for the great comments :)

closed 2021-07-09 by rick1082