zephyrproject-rtos/zephyr #36057: Zephyr Shell Console and Logging Targeting Isolated Different Device Interfaces

bob2oneil, 2021-06-08:
**Describe the bug**
I am unable to separate Zephry logging and the Zephyr Shell/Console to different physical interfaces.  Since a long running log can obscure the shell console, I need these to be on separate pathways.

Interface separation by device could potentially be 1 UARTS, 1 UART and a Telnet Session, or 2 Telnet sessions on different IPs.

I used the Telnet example as a starting point with the goal of having the Telnet session be the shell interface for user interaction, and the UART as simply the output log stream.

Despite having what I believe were the correct LOG target settings in the Project's configuration file, I wound up having both interfaces stream both the logging and the shell content.

This project initially targets the native-posix-64 board target.

Small changes to the Telnet main() source code is listed below as well as the project configuration file.

**To Reproduce**
Steps to reproduce the behavior:
1. west build project
2. Use minicom to establish serial interface to running instance at provided pseudoterminal
3. Use Telnet client to establish connection.
3. Observer output of logging and Zephyr shell console interface

**Expected behavior**
Isolation of output by interface.

With the following settings, I expected the Shell only under Telnet, and log output on UART only, not combined on both interfaces.

CONFIG_SHELL_BACKEND_TELNET=y
CONFIG_LOG_BACKEND_UART=y

**Impact**
Problem

**Logs and console output**

**Environment (please complete the following information):**
 - OS: (e.g. Ubuntu Linux 20)
 - Toolchain (e.g Zephyr SDK, Native Posix)
 - Zephyr Version 2.6 RC3

**Additional context**

The Telnet sample does a good job of showing the shell being targeted towards a Telnet session.
However, I am unable to get logging to isolate device target as the UART interface without it also targeting the Telnet session using this sample and making a few modifications as follows:

```
void main(void)
{
	struct net_if *iface = net_if_get_default();

	LOG_INF("Starting Telnet sample");

	setup_ipv4(iface);

	setup_dhcpv4(iface);

	setup_ipv6(iface);

	for (unsigned int i = 0; i < (unsigned)-1; i++)
	{
		printk("[%d] Telnet sample log output from printk()\n", i);

		LOG_INF("[%d] Telnet sample log output", i);

		k_msleep(SLEEPTIME);
	}
}
```

The kernel configuration as follows, and this effort targets the native_posix_64 board target.
```

CONFIG_PRINTK=y

CONFIG_NETWORKING=y
CONFIG_NET_IPV6=y
CONFIG_NET_IPV4=y
CONFIG_NET_ARP=y
CONFIG_NET_UDP=y
CONFIG_NET_TCP=y

CONFIG_NET_DHCPV4=n

CONFIG_NET_IF_UNICAST_IPV6_ADDR_COUNT=3
CONFIG_NET_IF_MCAST_IPV6_ADDR_COUNT=2
CONFIG_NET_MAX_CONTEXTS=10
CONFIG_NET_PKT_TX_COUNT=20
CONFIG_NET_BUF_TX_COUNT=50

CONFIG_TEST_RANDOM_GENERATOR=y

CONFIG_INIT_STACKS=y

CONFIG_NET_BUF_LOG=n
CONFIG_NET_LOG=n

CONFIG_NET_CONFIG_SETTINGS=y
CONFIG_NET_CONFIG_MY_IPV6_ADDR="2001:db8::1"
CONFIG_NET_CONFIG_MY_IPV4_ADDR="192.0.1.1"

CONFIG_NET_SHELL=y
CONFIG_KERNEL_SHELL=y
CONFIG_SHELL_BACKEND_TELNET=y
CONFIG_LOG=y
CONFIG_LOG_BACKEND_UART=y
CONFIG_NET_STATISTICS=y

# Debug Options for debugging
CONFIG_DEBUG_OPTIMIZATIONS=y

```
Walking through the code into LOG_INF() into method log_generic() the <shell_backend_telnet> device/value is positive allowing all log output to likewise target this interface..

manoj153, 2021-06-08:
I might not be that helpful, but I had faced a similar issue while isolating shell and logging to their respective uarts, below what works for me

dt_changes:
```
chosen {		
    zephyr,console = &usart2;		
    zephyr,shell-uart = &usart6;		
};
```

prj.conf

```
CONFIG_LOG_BACKEND_UART=y
CONFIG_SHELL_LOG_BACKEND=n
```

so perhaps `CONFIG_SHELL_LOG_BACKEND=n` might help you?

bob2oneil, 2021-06-08:
Thanks, I will give it a try with these changes to the kernel configuration.  I need this to work with a single UART and a Telnet session.  I will post my results in a moment.

bob2oneil, 2021-06-08:
That last kernel setting fixed it for me, thanks for you help and your contribution to the community.

closed 2021-06-08 by bob2oneil