zephyrproject-rtos/zephyr #36061: Undefined reference to `z_priq_rb_lessthan(rbnode*, rbnode*) when using k_timer_start in cpp file.

Zzzwierzak, 2021-06-08:
**Describe the bug**
Undefined reference to `z_priq_rb_lessthan(rbnode*, rbnode*) when using k_timer_start in cpp file. 

**To Reproduce**
Steps to reproduce the behavior:
1. Prepare cpp source file with call of k_timer_start() function and add kernel.h header. 
```
///
/// @file main.cpp
///
#include <kernel.h>

static void startTimerHandler(struct k_timer *timer)
{
    ARG_UNUSED(timer);
}

K_TIMER_DEFINE(startTimer, startTimerHandler, NULL);

void main(void)
{
    k_timer_start(&startTimer, K_MSEC(500), K_MSEC(0));
}
```

2. Ensure that CONFIG_WAITQ_SCALABLE is set
3. cmake -DBOARD=qemu_cortex_m3
3. ninja
4. See error:
/opt/zephyr-sdk/arm-zephyr-eabi/bin/../lib/gcc/arm-zephyr-eabi/10.2.0/../../../../arm-zephyr-eabi/bin/ld.bfd: app/libapp.a(main.cpp.obj):(._k_timer.static.startTimer+0x1c): undefined reference to `z_priq_rb_lessthan(rbnode*, rbnode*)'

**Expected behavior**
Build successful. 

**Environment (please complete the following information):**
 - Zephyr SDK 12.4, 
 - Zephyr 2.6.99

**Additional context**
After adding extern "C" to /include/kernel_structs.h  I'm able to build project.

andyross, 2021-06-09:
Yeah, for sure that's just some missing ABI hygiene.  Care to submit the patch? :)

Zzzwierzak, 2021-06-09:
@andyross, PR opened;)

closed 2021-06-18 by MaureenHelm