zephyrproject-rtos/zephyr #36078: coredump.logging_backend: lpcxpresso55s28: test failure assertion fail

hakehuang, 2021-06-09:
**Describe the bug**
     lpcxpresso55S28
    ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
    	ESF could not be retrieved successfully. Shall never occur.
    ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
    	ESF could not be retrieved successfully. Shall never occur.
    

**To Reproduce**
Steps to reproduce the behavior:
1. mkdir build; cd build
2. cmake -DBOARD=lpcxpresso55s28 ..
3. make; make flash
4. See error

**Expected behavior**
test pass

**Impact**
logging for core dump

**Logs and console output**
```
*** Booting Zephyr OS build zephyr-v2.6.0-90-gefccfcbfc915  ***
Coredump: lpcxpresso55S28
ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
	ESF could not be retrieved successfully. Shall never occur.
ASSERTION FAIL [esf != ((void *)0)] @ WEST_TOPDIR/zephyr/arch/arm/core/aarch32/cortex_m/fault.c:993
	ESF could not be retrieved successfully. Shall never occur.
    
```

**Environment (please complete the following information):**
 - OS: (e.g. Linux )
 - Toolchain (e.g Zephyr SDK)
 - Commit SHA or Version used: zephyr-v2.6.0-90-gefccfcbfc915

hakehuang, 2021-06-09:
@MaureenHelm , @dleach02 , found in this weekly

henrikbrixandersen, 2021-06-10:
@galak Sorry, but I don't have access to an lpcxpresso55s28 board (only lpcxpresso55s16).

galak, 2021-06-10:
> @galak Sorry, but I don't have access to an lpcxpresso55s28 board (only lpcxpresso55s16).

Oops, couldn't remember which one you had.  I guessed wrong ;).

dleach02, 2021-06-25:
I have this board

dleach02, 2021-08-17:
#37725 fixed this

closed 2021-08-17 by dleach02