zephyrproject-rtos/zephyr #36084: Arduino Nano 33 BLE: USB gets disconnected after flashing

nageshshamnur, 2021-06-09:
**Describe the bug**
Once Arduino Nano 33 BLE board is connected to host ubuntu machine via USB and flash samples such as ```samples/basic/blinky``` or ```samples/philosophers```, USB gets disconnected and port /dev/ttyACM0 (to which board is connected to becomes unavailable.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect Arduino Nano 33 BLE via USB to host ubuntu machine
2. Check the port and board connect status via dmesg
```
$ sudo dmesg
[318819.137425] usb 1-2: new full-speed USB device number 78 using xhci_hcd
[318819.288999] usb 1-2: New USB device found, idVendor=2341, idProduct=005a, bcdDevice= 0.11
[318819.289013] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[318819.289018] usb 1-2: Product: Arduino Nano 33 BLE
[318819.289023] usb 1-2: Manufacturer: Arduino
[318819.289027] usb 1-2: SerialNumber: 0000000000000000768275C7726D618B
[318819.291711] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
```
3. set permission to port via ```sudo chmod a+rw /dev/ttyACM0```
4. build sample ```west build -p auto -b arduino_nano_33_ble samples/philosophers```
5. flash via ```west flash --bossac=<arduino_variant_bossac>"
6. dmesg will show that USB is disconnected and access to port ```/dev/ttyACM0`` is not accessible anymore.
```
$ sudo dmesg
[318819.137425] usb 1-2: new full-speed USB device number 78 using xhci_hcd
[318819.288999] usb 1-2: New USB device found, idVendor=2341, idProduct=005a, bcdDevice= 0.11
[318819.289013] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[318819.289018] usb 1-2: Product: Arduino Nano 33 BLE
[318819.289023] usb 1-2: Manufacturer: Arduino
[318819.289027] usb 1-2: SerialNumber: 0000000000000000768275C7726D618B
[318819.291711] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[318889.384145] usb 1-2: USB disconnect, device number 78
[318889.693520] usb 1-2: new full-speed USB device number 79 using xhci_hcd
[318889.849229] usb 1-2: New USB device found, idVendor=2341, idProduct=005a, bcdDevice= 0.11
[318889.849243] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[318889.849248] usb 1-2: Product: Arduino Nano 33 BLE
[318889.849253] usb 1-2: Manufacturer: Arduino
[318889.849257] usb 1-2: SerialNumber: 0000000000000000768275C7726D618B
[318889.851946] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[318889.895099] usb 1-2: USB disconnect, device number 79
```
```
$ sudo minicom -D /dev/ttyACM0
minicom: cannot open /dev/ttyACM0: No such file or directory
```
**Expected behavior**
Once flashing is completed, board needs to be accessible via USB and port ```/dev/ttyACM0``` should be accessible for minicom.

**Impact**
Once flashing is done, USB is not accessible any more. Need to access the board via UART which is connected by USB-TTL cable to board.

**Logs and console output**
As attached in steps to reproduce section

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04.2 LTS 
 - Toolchain: zephyr-sdk-0.12.3
 -  Commit SHA: 457a28bf78a4cdd2b17f3983c50a0183f2f71700

**Additional context**
Add any other context about the problem here.

nageshshamnur, 2021-06-09:
@nandojve : I am not sure, if this issue was introduced after fixing #33352 but for sure, it came out after the issue was fixed.

nageshshamnur, 2021-06-09:
@jfischer-no: Just trying to understand more. Same philosophers sample when flashed for 96b_nitrogen board, i am able to use the same port ```/dev/ttyACM0``` to watch the console via minicom ``` $minicom -D /dev/ttyACM0```.

nageshshamnur, 2021-06-09:
steps done for 96b_nitrogen:
1) connect 96b_nitrogen via USB
```
$ sudo dmesg
[342257.436253] usb 1-2: USB disconnect, device number 82
[342257.449006] FAT-fs (sda): unable to read boot sector to mark fs as dirty
[342260.396510] usb 1-2: new full-speed USB device number 83 using xhci_hcd
[342260.548444] usb 1-2: New USB device found, idVendor=0d28, idProduct=0204, bcdDevice= 1.00
[342260.548458] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[342260.548464] usb 1-2: Product: MBED CMSIS-DAP
[342260.548469] usb 1-2: Manufacturer: MBED
[342260.548473] usb 1-2: SerialNumber: 9009022103BB586AFE65379B
[342260.553801] usb-storage 1-2:1.0: USB Mass Storage device detected
[342260.554481] scsi host1: usb-storage 1-2:1.0
[342260.555102] cdc_acm 1-2:1.1: ttyACM0: USB ACM device
[342260.558208] hid-generic 0003:0D28:0204.000B: hiddev1,hidraw2: USB HID v1.00 Device [MBED MBED CMSIS-DAP] on usb-0000:00:14.0-2/input3
[342261.557201] scsi 1:0:0:0: Direct-Access     MBED     microcontroller  1.0  PQ: 0 ANSI: 2
[342261.557866] sd 1:0:0:0: Attached scsi generic sg0 type 0
[342261.558579] sd 1:0:0:0: [sda] 16512 512-byte logical blocks: (8.45 MB/8.06 MiB)
[342261.559031] sd 1:0:0:0: [sda] Write Protect is off
[342261.559041] sd 1:0:0:0: [sda] Mode Sense: 03 00 00 00
[342261.559380] sd 1:0:0:0: [sda] No Caching mode page found
[342261.559393] sd 1:0:0:0: [sda] Assuming drive cache: write through
[342261.583088]  sda:
[342261.585306] sd 1:0:0:0: [sda] Attached SCSI removable disk
```
2) flash philosophers 
```$ west build -p auto -b 96b_nitrogen samples/philosophers```
3) flash the board
```$ west flash
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner pyocd
-- runners.pyocd: Flashing file: /home/ohos-bld/zephyrproject/zephyr/build/zephyr/zephyr.hex
[====================] 100%
0004220:INFO:loader:Erased 0 bytes (0 sectors), programmed 0 bytes (0 pages), skipped 28672 bytes (7 pages) at 8.89 kB/s
```
4) watch the console via same USB
```$ sudo minicom -D /dev/ttyACM0

Philosopher 0 [P: 3]        STARVING       
Philosopher 1 [P: 2]  THINKING [  375 ms ] NKING [  25 ms  EATING  [  650 ms ]                                                         
Philosopher 2 [P: 1]    HOLDING ONE FORK   ]                                                                                           
Philosopher 3 [P: 0]   EATING  [  725 ms ]                                                                                             
Philosopher 4 [C:-1]        STARVING        [P: 2]        STARVING                                                                     
Philosopher 5 [C:-2]   EATING  [  275 ms ]  [  25 ms ]                                                                                 
                                                                                                                                       
Press CTRL-A Z for help on special keys 
```

nageshshamnur, 2021-06-09:
> It i not an issue, you are using it wrong. Neither samples/basic/blinky nor samples/philosophers enables USB, they are not supposed to do it either. Check out samples/subsys/usb/console

@jfischer-no : Like you mentioned, i had already checked the source code under ```samples/subsys/usb/cdc_acm``` and i had seen ```enable_usb( )``` which i added to philosophers sample for testing purpose and i see the USB is intact even after flashing. but i was still not able to use that port ```/dev/ttyACM0```

jfischer-no, 2021-06-09:
> @jfischer-no: Just trying to understand more. Same philosophers sample when flashed for 96b_nitrogen board, i am able to use the same port `/dev/ttyACM0` to watch the console via minicom ` $minicom -D /dev/ttyACM0`.

96b_nitrogen board has on board debug adapter (probably DAPLink) and is based on nRF52832 SoC which does not support USB. It is different to Arduino Nano 33 BLE board. Nano 33 is not equipped with a debug adapter and uses bootloader (BOSSA) to upload firmware. You can also see it when you read your own bug description:
>build sample west build -p auto -b arduino_nano_33_ble samples/philosophers
flash via ```west flash --bossac=<arduino_variant_bossac>"

See also relevant section in the boards documentation https://docs.zephyrproject.org/latest/boards/arm/arduino_nano_33_ble/doc/index.html#programming-and-debugging, https://docs.zephyrproject.org/latest/boards/arm/96b_nitrogen/doc/index.html#flashing-zephyr-onto-96boards-nitrogen

rzr, 2022-08-23:
@joaodullius hi any update ? on this problem

note that @nageshshamnur 's patch wont be apply on current code base since  CONFIG_BOARD_ARDUINO_NANO_33_BLE_EN_USB_CONSOLE has been removed in:

https://github.com/zephyrproject-rtos/zephyr/pull/44023/commits/69707dfb4bd91b513fb3e1503d76265c857f18b7#diff-93ba35382d86f5dce92d22c0f8c1a2649ba17786dca3bff30c5b760cbe8b9128L8

I'll investigate a bit more and report a new issue if it helps