zephyrproject-rtos/zephyr #36088: intel_adsp_cavs25: secondary boot fails in arch_start_cpu()

kv2019i, 2021-06-09:
**Describe the bug**
SOF test cases using multiple DSP cores fail on cAVS25 hardware (TGL).

**To Reproduce**
Steps to reproduce the behavior:
Compile SOF application with Zephyr for cavs25 target and load SOF Linux kernel driver using a DSP topology that has at least one pipeline element assigned to non-primary core. This will result in SOF application to call arch_start_cpu(1).

Tracing the problem, zephyr/soc/xtensa/intel_adsp/common/soc_mp.c:arch_start_cpu() never gets pass
»       while (!start_rec.alive)
»       »       ;

**Expected behavior**
SOF test cases with multiple cores should pass.

**Impact**
Showstopper for SOF usage on TGL hw when using Zephyr as OS.

**Logs and console output**
n/a

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Zephyr SDK 0.12.4
 - Commit SHA: 1f0b783bbb2cd0546d0fee12d72cccc694847915

**Additional context**
Related SOF bug https://github.com/thesofproject/sof/issues/4298

kv2019i, 2021-06-09:
FYI @nashif @lgirdwood @lyakh

lgirdwood, 2021-06-29:
@andyross I will try and run this on Qemu and see if we are programming different IP to SOF

kv2019i, 2021-08-30:
Removing the stale label as this is actively worked on, with latest fix candidate in https://github.com/zephyrproject-rtos/zephyr/pull/37647

andyross, 2021-09-08:
Fixed and merged.  Closing this.  Seems like in those 19 patches or whatever that I might have remembered to include a "Fixed:" header, but apparently I didn't.

closed 2021-09-08 by andyross