zephyrproject-rtos/zephyr #36089: intel_adsp_cavs25: support more than 2 DSP cores

kv2019i, 2021-06-09:
**Is your enhancement proposal related to a problem? Please describe.**
Current cavs25 implementation is limited to max 2 DSP cores while we need to support SKUs upto 4 cores:

zephyr/soc/xtensa/intel_adsp/common/soc_mp.c:arch_start_cpu()
```
»       __ASSERT(cpu_num == 1, "Only supports only two CPUs!");
```

**Describe the solution you'd like**
Support full set of 4 DSP cores.

**Describe alternatives you've considered**
n/a

**Additional context**
n/a

kv2019i, 2021-06-09:
FYI @nashif @andyross @lgirdwood @lyakh

andyross, 2021-09-08:
Closing this.  Support for all four cores is present and works on TGL through all the core tests.  The default CONFIG_MP_NUM_CPUS setting is still 2 though, out of conservatism (not wanting to expose bugs in the test suite late in a release cycle), and the fact that existing SOF topologies only want to exercise two cores anyway.

closed 2021-09-08 by andyross