zephyrproject-rtos/zephyr #36094: BLE wrong connections intervals on multible connections

hf81-ble, 2021-06-09:
**Describe the bug**
If defining a strict connection interval to all peripherals to have solid timing inside each connection interval, completly different connection intervals a used. 

**To Reproduce**
1. Use 10 or more peripherals 
2. Use a connection interval were all peripherals can fit with their worst event length
10*1,4 => 15ms

3. 
4. 

**Expected behavior**
All peripherals should equally be distributed inside the connection intervall
(1) 1,5 => (2) 3ms => (3) 4,5ms => ...... => (10) 15ms

**Impact**
-Datarate is not consistent to all peripherals equally.
-Performance is dropped
-unpredictable datarates

**Logs and console output**
=> no HW to log all connections simulaneous.
=> would it be helpfull to log single connections?
=> is there a way to read out the real (internal) and not the requested connection intervals?

**Environment (please complete the following information):**

**Additional context**

cvinayak, 2021-06-09:
> To Reproduce
>
>    Use 10 or more peripherals
>
>    Use a connection interval were all peripherals can fit with their worst event length
    10*1,4 => 15ms
>
>    Wait till all peripherals are connection.
>
>    Transmit equal packets sizes
>
>    Monitor transmission times and calc back the "real" connection interval (longterm not to track single transmission errors)

Please share your application, preferably in a draft PR. This will speed up analysis. Also, provide complete details of zephyr commit hash used, toolchain, development environments, platforms/boards used, steps to build, flash and test your application.

carlescufi, 2021-06-17:
@hf81-ble could you please provide the additional information requested by @cvinayak?

carlescufi, 2021-07-15:
Closing due to inactivity after the response from the maintainer.

closed 2021-07-15 by carlescufi

CodingGhost, 2024-02-20:
I just found this and it seems like this is the same problem I have described in #68336