zephyrproject-rtos/zephyr #36107: ehl_crb: Multiple tests are failing and board is not booting up.

yerabolu, 2021-06-09:
**Describe the bug**
On EHL_CRB, Board is not booting up/stuck after flashing the build.

What have you tried to diagnose or workaround this issue?
Tried git bisect and commit: 9cb8dcbf841484ccb925b6032775e9db0bc2150c  is identified where the board is getting stuck/booting up.

**To Reproduce**
Steps to reproduce the behavior:
1. twister -W -p ehl_crb --device-testing --device-serial /dev/ttyUSB0 -T tests --west-flash=/home/ztest/EHL_X86_PXE.sh -j 1
2. Board is not booting up/stuck.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Impact**
Pass rate dropped to ~38%.

**Logs and console output**
```
DEBUG   - DEVICE: NBP filename is zephyr.efi
DEBUG   - DEVICE: NBP filesize is 6539951 Bytes
DEBUG   - DEVICE: Downloading NBP file...
DEBUG   - DEVICE:
DEBUG   - DEVICE: NBP file downloaded successfully.
DEBUG   - DEVICE: *** Zephyr EFI Loader ***
DEBUG   - DEVICE: Zeroing 23871488 bytes of memory at 0x12d000
DEBUG   - DEVICE: Copying 32768 data bytes to 0x1000 from image offset
DEBUG   - DEVICE: Copying 184320 data bytes to 0x100000 from image offset 32768
DEBUG   - DEVICE: Copying 6316576 data bytes to 0x17f1000 from image offset 217088
DEBUG   - DEVICE: Jumping to Entry Point: 0x112b (48 31 c0 48 31 d2 48)
DEBUG   - DEVICE: *** Booting Zephyr OS build zephyr-v2.6.0-119-g5719e545af4d
```

**Environment (please complete the following information):**
 - OS: Fedora
 - Toolchain: Zephyr SDK 12.4
 - Commit: 9143f4fd8c35

andyross, 2021-06-10:
> Jumping to Entry Point: 0x112b

This looks suspicious.  When I build hello_world for this platform, the entry point shows as 0x1000, which is where it should be for all x86 builds right now (because acrn couldn't easily handle the fact that our entry point moves around, we pin it to the beginning of .locore for everyone now).

Which project did you build to get this output?

andyross, 2021-06-10:
Oh, wait.  Now I see what's happening.  EFI entry happens in **64 bit** mode, so it skips the new code at __start32 that fixes up the jump at the start of .locore, so the first SMP CPU to launch hits that jump while in real mode and the world blows up.   Will fix.

PR #36191 (2021-06-14): arch/x86/zefi: Fix entry-nop hack for EFI entry
commit 5e9c583c24c1 ("arch/x86_64: Terrible, awful hackery to
bootstrap entry") introduced a terrible trick which begins execution
at the bottom of .locore with a jump, which then gets replaced with
NOP instructions for the benefit of 16 bit real mode startup of the
other CPUs later on.

But I forgot that EFI enters in 64 bit code natively, and so never
hits that path.  And moving it to the 64 bit setup code doesn't work,
because at that point when we are NOT loaded from EFI, we already have
the Zephyr page tables in place that disallow writes to [... cut]
commit b651aa9f (2021-06-14): arch/x86/zefi: Fix entry-nop hack for EFI entry

commit 5e9c583c24c1 ("arch/x86_64: Terrible, awful hackery to
bootstrap entry") introduced a terrible trick which begins execution
at the bottom of .locore with a jump, which then gets replaced with
NOP instructions for the benefit of 16 bit real mode startup of the
other CPUs later on.

But I forgot that EFI enters in 64 bit code natively, and so never
hits that path.  And moving it to the 64 bit setup code doesn't work,
because at that point when we are NOT loaded from EFI, we already have
the Zephyr page tables in place that disallow writes t [... cut]
commit d0dca157 (2021-06-29): arch/x86/zefi: Fix entry-nop hack for EFI entry

commit 5e9c583c24c1 ("arch/x86_64: Terrible, awful hackery to
bootstrap entry") introduced a terrible trick which begins execution
at the bottom of .locore with a jump, which then gets replaced with
NOP instructions for the benefit of 16 bit real mode startup of the
other CPUs later on.

But I forgot that EFI enters in 64 bit code natively, and so never
hits that path.  And moving it to the 64 bit setup code doesn't work,
because at that point when we are NOT loaded from EFI, we already have
the Zephyr page tables in place that disallow writes t [... cut]
closed 2021-06-14 by nashif