zephyrproject-rtos/zephyr #36117: toolchain: The added abstraction for llvm, breaks builds with off-tree llvm based toolchains

daor-oti, 2021-06-10:
**Describe the bug**
The abstraction added for llvm, in commit [eca4618](https://github.com/zephyrproject-rtos/zephyr/pull/34370/commits/eca46188c3cdb665d74bfc8c00afa6165f82edb6) under PR #34370, adds a check for definition of `__llvm__` to the `include/toolchain.h` file.

For off-tree toolchains based upon llvm, this means that the `include/toolchain/llvm.h`, which includes `include/toolchain/gcc.h`, will be included and thus the `include/toolchain/other.h` will never be included, as it ought to have been.

**To Reproduce**
This is only reproducible when having an off-tree toolchain which is based upon llvm.

**Expected behavior**
This would be that the `include/toolchain/other.h` gets included for all off-tree toolchains, no matter what they are based upon.

**Impact**
Breaks build for off-tree toolchains based upon llvm, and as such is a showstopper for those usages.

**Logs and console output**
NA.

**Environment (please complete the following information):**
 - OS: Linux
 - Toolchain: Off-tree llvm frontend based
 - Commit SHA: [eca4618](https://github.com/zephyrproject-rtos/zephyr/pull/34370/commits/eca46188c3cdb665d74bfc8c00afa6165f82edb6) or later

nashif, 2021-06-10:
> This would be that the `include/toolchain/other.h` gets included for all off-tree toolchains, no matter what they are based upon.

Not sure how something like this would work, either you are llvm or not. If your 'other' compiler has some unique identifier the solution would be to add a check for that compiler and include a specific header for this 'other' compiler. Depending on the fallback for compiler support will always lead to issues IMO.

daor-oti, 2021-06-16:
@nashif As we want 'any' off-tree toolchain to get the `other.h` included, and since it makes little sense for all of the off-tree toolchains to have to update the toolchains.h file, I suggest we come up with a common define to be set for off-tree toolchains, which is then checked first and which includes `other.h`. At the `else`, we then put an error, stating that the toolchain is unsupported and/or unusable. 
This will solve the problem for off-tree toolchains and also eliminate the issues with fallback compiler/toolchain support.
How does that sound?