zephyrproject-rtos/zephyr #36131: Occasionally unable to scan for extended advertisements when connected

Thalley, 2021-06-10:
**Describe the bug**
If connected to another BLE device, scanning for extended advertisements occasionally does not work, in the sense that the host does not receive any advertisement reports from the device advertising extended advertisements. 

**To Reproduce**
This requires 3 devices and be reproduced using the BT shell:

First device (legacy advertiser)
1. Build and flash the BT shell and run the following commands:
2. `bt init`
3. `bt advertise on`

Second device (extended advertiser)
1. Build and flash the BT shell and run the following commands:
2. `bt init`
3. `bt adv-create nconn-nscan ext-adv name`
4. `bt adv-start`

Third device (scanner/initiator)
1. Build and flash the BT shell and run the following commands:
2. `bt init`
3. `bt scan on`
4. Verify that advertisements from both advertisers are visible
5. `bt scan off`
6. Connect to legacy advertiser
7. `bt scan on`
8. Verify that advertisements from the extended advertiser is visible

Step 8 occasionally fails, and it can be verified that it does in fact advertise by scanning with the legacy advertiser device, or a 4th device. If the advertisements from the extended advertiser does not show, then they will continue to not showing even after disconnecting with the legacy advertiser. Step 8 fails roughly 30% of the time, but can fail 100% in a BabbleSIM test implying that it's a timing issue (as the issue can be remove in a BabbleSIM test by introducing a delay using k_sleep). 

**Expected behavior**
Expected behavior is that extended advertisements from the second device will always reach the host. 

**Impact**
Low as this is still experimental. 

**Logs and console output**
N/A

**Environment (please complete the following information):**
 - Toolchain: Zephyr SDK 0.12.4
 - Commit SHA or Version used: 5719e545af4d70262654c7b1d385911eb052925f

**Additional context**
N/A

Thalley, 2021-06-10:
What @asbjornsabo says is likely the case, but any subtle timing changes may also be the cause why we start seeing it now, even if the issue is older. It's only recently that we have implemented a test that requires scanning for extended advertising while connected.

Thalley, 2021-06-11:
@cvinayak and @andrzej-kaczmarek perhaps this is an issue for one of you guys?

cvinayak, 2021-06-15:
@Thalley I tried both upstream main 7ba52d8f82 and the reported 5719e54, I am not able to reproduce the issue on `nrf52840dk_nrf52840` board with the `tests/bluetooth/shell` application with `CONFIG_BT_CTLR_EXT_ADV=y` enabled.

[image]

Thalley, 2021-06-15:
@cvinayak If you have access to https://github.com/zephyrproject-rtos/zephyr-bt-sig you can easily reproduce it by changing `-rs=24` to `-rs=23` in https://github.com/zephyrproject-rtos/zephyr-bt-sig/blob/le-audio/tests/bluetooth/bsim_bt/bsim_test_audio/test_scripts/bass.sh and run that test. 

Please let me know if you have issues with the above.

cvinayak, 2021-06-21:
The reception of primary channel PDU and auxiliary PDU are being cancelled by the connection event. Connection interval of 50 ms, scan interval of 60ms and a window of 30 ms unfortunately is not able to receive unless the scanning is kept active for a longer duration.

@Thalley The suggested solution is to use continuous scanning, please try an interval of 2.5ms and a window of 2.5 ms.

Thalley, 2021-06-21:
@cvinayak That is a workaround though, isn't it (which I already have)?

The test is attempting to simulate a real life scenario using the default scan parameters of Zephyr. 

If I understood correctly, the scan interval and window are basically in conflict with the connection interval, right? 
Is this something that can be detected by the controller, and if so, would it not be possible to offset the scan window by a couple milliseconds?

cvinayak, 2021-06-21:
> @cvinayak That is a workaround though, isn't it (which I already have)?
> 
> The test is attempting to simulate a real life scenario using the default scan parameters of Zephyr.
> 

This is not a work around, advertising primary PDU and auxiliary PDU can be anywhere within the advertising interval duration. The recommended BT Spec value are, `TGAP(scan_fast_interval) 30 ms to 60 ms`, and `TGAP(scan_fast_window) 30 ms`. Using continuous scanning (30 ms interval, 30 ms window) will ensure quicker reception of advertising reports.
I suggested 2.5ms, as it will have less overlap with other states/roles.

> If I understood correctly, the scan interval and window are basically in conflict with the connection interval, right?
> Is this something that can be detected by the controller, and if so, would it not be possible to offset the scan window by a couple milliseconds?

In the implementation, if scan interval and connection interval are similar, then it does by design place the scan window after the connection event. But in the test's case, the intervals are not similar, and hence will drift to collide/overlap.

Thalley, 2021-06-21:
> This is not a work around, advertising primary PDU and auxiliary PDU can be anywhere within the advertising interval duration. The recommended BT Spec value are, `TGAP(scan_fast_interval) 30 ms to [...]

Changing from the recommended values to continuous scanning due to the connection parameters (which may change from peripheral to peripheral) sounds like a workaround to me, as we should be able to scan with the recommended values (which we clearly can by changing the timing of the device just slightly). 

I don't think I would accept "continuous scan" as a solution to this issue. 

> But in the test's case, the intervals are not similar, and hence will drift to collide/overlap.
If that is the case, wouldn't they at some point drift to avoid collision/overlap again?

cvinayak, 2021-06-21:
> I don't think I would accept "continuous scan" as a solution to this issue.

Interval of 30 ms and a window of 30 ms (continuous scanning) is in the range of recommended value in the BT Spec.

> If that is the case, wouldn't they at some point drift to avoid collision/overlap again?

Your sample is using 100 ms advertising interval, connection interval of 50ms, scan interval of 60 ms and scan window of 30ms. Yes, they should drift, avoid overlaps and eventually be received by the scanner.

I have been debugging the test and notice that the issue was due to a buffer leak under race condition. Race condition being, any connection established to another device between primary channel PDU reception and auxiliary PDU reception of another device.

I will send a PR shortly.

PR #36439 (2021-07-02): Bluetooth: Controller: Fix auxiliary scan context leak
Fix an auxiliary scan context leak when initiator
establishes a connection while there is another pending
auxliary PDU scheduled to be received. In this case, the
pending auxliary scan LLL context does an early abort
without generating a scan aux done event. Missing scan aux
done event caused the auxiliary scan context leak.

Fixes #36131.
commit 3cc8f317 (2021-07-02): Bluetooth: Controller: Fix auxiliary scan context leak

Fix an auxiliary scan context leak when initiator
establishes a connection while there is another pending
auxliary PDU scheduled to be received. In this case, the
pending auxliary scan LLL context does an early abort
without generating a scan aux done event. Missing scan aux
done event caused the auxiliary scan context leak.

Fixes #36131.
closed 2021-07-02 by nashif