zephyrproject-rtos/zephyr #36172: net: ieee802154: LL src/dst address is lost from received net_pkt (when using 6LO)

niflostancu, 2021-06-11:
Hello,
I am trying to read the MAC addresses for received IPv6 + UDP packets but this info disappears when reaching the net context layer (I use a custom receive callback into the context).
Specifically, after receiving the IEEE802.15.4 frame (with 6LO is enabled), `pkt->lladdr_src` and `pkt->lladdr_dst` point to the wrong memory after passing through the [`ieee802154_manage_recv_packet`](https://github.com/zephyrproject-rtos/zephyr/blob/75a2ef8fc02a417f0d26307105f80b136a666dc8/subsys/net/l2/ieee802154/ieee802154.c#L124). I'll detail it below, together with my observations and personal opinions. 
Note: I am not a Zephyr developer, most of its code is at first sight.
Also, note that I'm using a TI CC1352-based board with the `ieee802154_cc13xx_cc26xx_subg` driver, but I don't think it matters (since this layer has already finished doing its processing).

What I did:
- I have a UDP client and a UDP server receiving packets using a `net_context_recv` callback;
- Layer 3+ communication works as expected, but I need the L2 address of the sender device :( 
- I used gdb and found out that the `pkt->lladdr_src` contains the correct L2 MAC address BEFORE executing the `ieee802154_manage_recv_packet` function mentioned above.
- After digging deeper, we see that the `lladdr` offsets are backed up using [the following code](https://github.com/zephyrproject-rtos/zephyr/blob/75a2ef8fc02a417f0d26307105f80b136a666dc8/subsys/net/l2/ieee802154/ieee802154.c#L150):
```c
	/** Uncompress will drop the current fragment. Pkt ll src/dst address
	 * will then be wrong and must be updated according to the new fragment.
	 */
	src = net_pkt_lladdr_src(pkt)->addr ?
		net_pkt_lladdr_src(pkt)->addr -
		(net_pkt_data(pkt) - hdr_len) : 0;
```
- When using GDB to read the pointers, `pkt->frags->__buf` points to `net_buf_data_rx_bufs` (i.e., first buffer).
- Then the function calls `ieee802154_reassemble` -> `net_6lo_uncompress` -> `uncompress_IPHC_header`, where the main cause of the problem [is here](https://github.com/zephyrproject-rtos/zephyr/blob/75a2ef8fc02a417f0d26307105f80b136a666dc8/subsys/net/ip/6lo.c#L1365):
```c
	if (net_buf_tailroom(pkt->buffer) >= diff) {
            // ...
	} else { // this is triggered...
		NET_DBG("Not enough tailroom. Get new fragment");
		cursor =  pkt->buffer->data;
		frag = net_pkt_get_frag(pkt, NET_6LO_RX_PKT_TIMEOUT);
``` 
i.e., a new fragment buffer is allocated for the packet and replaced [in here](https://github.com/zephyrproject-rtos/zephyr/blob/75a2ef8fc02a417f0d26307105f80b136a666dc8/subsys/net/ip/6lo.c#L1465):
```c
	if (frag != pkt->buffer) {
		/* Insert the fragment (this one holds uncompressed headers) */
		net_pkt_frag_insert(pkt, frag);
	}
```
- note: the `pkt->frags` is replaced with the new allocated buffer, the old one being available at `pkt->frags->frags`;
- when execution returns back to [`ieee802154_manage_recv_packet`](https://github.com/zephyrproject-rtos/zephyr/blob/75a2ef8fc02a417f0d26307105f80b136a666dc8/subsys/net/l2/ieee802154/ieee802154.c#L169), the `lladdr` pointers are restored using this new buffer at `net_pkt_data(pkt)` which, unfortunately, does not contain the correct L2 addresses (coincidentally, it's filled with 0x00 for me):
```c
	net_pkt_lladdr_src(pkt)->addr = src ?
		(net_pkt_data(pkt) - hdr_len) + src : NULL;
```

**Expected behavior**
I need the source IEEE802.15.4 Link Layer MAC address :( 

**Impact**
I have a workaround, which hopefully doesn't affect other functions, though I want an official opinion / proper fix :) 

**GDB output with memory output**
```
pwndbg> bt
#0  ieee802154_manage_recv_packet (iface=<optimized out>, hdr_len=15, pkt=0x200072ec <_k_mem_slab_buf_rx_pkts+204>) at ***/zephyr/zephyr/subsys/net/l2/ieee802154/ieee802154.c:158
...
pwndbg> p pkt->frags->__buf
$1 = (uint8_t *) 0x200075c0 <net_buf_data_rx_bufs>
pwndbg> x/8xb pkt->lladdr_src.addr
0x200075c7 <net_buf_data_rx_bufs+7>:    0x38    0x9c    0x83    0x24    0x00    0x4b 0x12     0x00
```
later, after executing the `uncompress_IPHC_header` and restoring the LL address pointers (note the `lladdr_src` contents):
```
pwndbg> p pkt->frags->__buf
$2 = (uint8_t *) 0x200078c0 <net_buf_data_rx_bufs+768>
pwndbg> p pkt->frags->frags->__buf
$3 = (uint8_t *) 0x200075c0 <net_buf_data_rx_bufs>
pwndbg> x/8xb pkt->lladdr_src.addr
0x200078b8 <net_buf_data_rx_bufs+760>:  0x00    0x00    0x00    0x00    0x00    0x00 0x00     0x00
```

**Environment (please complete the following information):**
 - OS: Arch Linux
 - Toolchain: arm-none-eabi-gcc (Arch Repository) 11.1.0
 - Commit SHA: 39354d92bb

Note that it _WorksForMe_ if I comment the lines "restoring" the `lladdr` after the decompression call (I don't think the old fragment is deallocated?).
I think that the real fix would be to also copy the L2 header into the new fragment's buffer, though, as I said, I am pretty new to Zephyr's codebase :( 
I can implement the fix and come with a PR if you would tell me the correct approach ;) 

Thanks!

alexanderwachter, 2021-06-13:
The problem is that the IPHC does not know anything about the L2 address, but the L2 address is stored in the buffer that was pulled from the beginning.
The same problem hit me when I was developing the CAN L2.
That is why i reworked the IPHC to reuse the old buffer whenever possible. Storing the L2 in the packet an pull it is like doing a malloc, write data, free it and hope that nobody reuses the memory.

alexanderwachter, 2021-06-13:
Ping @jukkar

jukkar, 2021-06-14:
The idea of the lladdr pointer in net_pkt, was to save some memory because the data is there already in first net_buf. For IPHC this is not really true as Alexander pointed out. I just wonder would it be possible to restore the ll addresses after the uncompress and then point the lladdr to those.

niflostancu, 2021-06-14:
From how I understand it, the IPHC decompression algorithm retains the old fragment's pointer at `pkt->frags->frags`, so its buffer won't get freed (right?). If this is the case, the `lladdr_*` pointers are wrongly modified [in here](https://github.com/zephyrproject-rtos/zephyr/blob/75a2ef8fc02a417f0d26307105f80b136a666dc8/subsys/net/l2/ieee802154/ieee802154.c#L169)?
If there is another code path leading to the first buffer being deallocated, then the LL address restoration should be conditioned to this.

rlubos, 2021-06-16:
Hi @niflostancu @alexanderwachter @jukkar 

After digging into the issue a bit, I think that the simple fix proposed by @niflostancu might actually be a good one. Looking how things worked before https://github.com/zephyrproject-rtos/zephyr/pull/16794, the `uncompress_IPHC_header()` function could indeed deallocate the packet (it called `net_pkt_compact()` which could do this). So perhaps backing up the offset and restoring it afterwards made some sense. It's still unclear to me though whether it actually worked correctly, as the 15.4 header information is stripped off with net_pkt with [net_buf_pull()](https://github.com/zephyrproject-rtos/zephyr/blob/main/subsys/net/l2/ieee802154/ieee802154.c#L227) so it wouldn't be copied anyway, but there's little point in analyzing the past state, as it works differently now.

As Alexander pointed out, the 6lo layer will now retain the original `net_buf` allocated for the packet. We do trim the 15.4 header with `net_buf_pull()` but the data is still there (of course provided that upper layers won't overwrite it, but it seemed to be an assumption from the beginning). So in my opinion all that is needed to do in this case, is to drop the backing-up and restoring the LL address pointers, and leave them intact.

rlubos, 2021-06-17:
@niflostancu I've submitted #36359 with a fix, please give it a try.

niflostancu, 2021-06-17:
I've been compiling Zephyr with those lines commented for a couple of days now.
Just tested on my boards, it works ;) I can finally get rid of the manual patch :D

PR #36359 (2021-07-09): net: l2: ieee802154: Fix LL address handling on frame reception
6lowpan module can swap the original buffer with a newly allocated one
during decompression in case the decompressed header would not fit into
the original buffer. Therefore, storing the LL address offset and
restoring the pointer after decompression as it is done today is not
correct, as the new packet with decompressed IPv6 header will not
contain the LL header.

As the 6lowpan module doesn't deallocate the original buffer and
doesn't overwrite the LL header, its fine to use the original
pointers as they are.

Fixes #36172
closed 2021-07-09 by cfriedt