zephyrproject-rtos/zephyr #36185: CMP0116 related warnings

cederom, 2021-06-11:
**Describe the bug**
With version v2.6.0 Zephyr and `CONFIG_DEBUG=y` I can see build warning that come from Zephyr source tree (output provided in logs section). This seems low priority but may be nice to fix in a free moment :-)

**To Reproduce**
Steps to reproduce the behavior:
1. Set ` CONFIG_DEBUG=y` in `prj.conf`.
2. Run `west build`.
3. See error

**Expected behavior**
I guess expected policy needs to be set.

**Impact**
No impact noticed.

**Logs and console output**
```
CMake Warning (dev) at /XXX/zephyr/cmake/linker/ld/target.cmake:51 (add_custom_command):
  Policy CMP0116 is not set: Ninja generators transform DEPFILEs from
  add_custom_command().  Run "cmake --help-policy CMP0116" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  /XXX/zephyr/CMakeLists.txt:1198 (configure_linker_script)
This warning is for project developers.  Use -Wno-dev to suppress it.

CMake Warning (dev) at /XXX/zephyr/cmake/linker/ld/target.cmake:51 (add_custom_command):
  Policy CMP0116 is not set: Ninja generators transform DEPFILEs from
  add_custom_command().  Run "cmake --help-policy CMP0116" for policy
  details.  Use the cmake_policy command to set the policy and suppress this
  warning.
Call Stack (most recent call first):
  /XXX/zephyr/CMakeLists.txt:1253 (configure_linker_script)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
```

**Environment (please complete the following information):**
 - OS: FreeBSD
 - Toolchain GNU ARM GCC + BSD.
 - Zephyr v2.6.0.
 - `arm-none-eabi-gcc (GNU Arm Embedded Toolchain) 10.2.1 20201103 (release)`, `cmake version 3.20.2`, `ninja version 1.10.2
`.

de-nordic, 2021-06-14:
@tejlmand ?

de-nordic, 2021-06-14:
@cederom Can you take a look at https://github.com/zephyrproject-rtos/zephyr/pull/36223?

cederom, 2021-06-14:
Thank you @de-nordic moving conversation into https://github.com/zephyrproject-rtos/zephyr/pull/36223 :-)

PR #36223 (2021-06-16): cmake: set CMP0116 policy to old, Ninja DEPFILE behaviour
Fixes: #36185

CMake >=3.20 now supports relative paths in Ninja depfiles.

As Zephyr still supports the use of CMake 3.13, and the current use of
DEPFILE in linker/ld/target.cmake is already absolute path, then the
safest choice is to use old behavior.
This can be changed when Zephyr increases minimum required CMake to be
3.20 or newer.
commit 2f7ede5a (2021-06-16): cmake: set CMP0116 policy to old, Ninja DEPFILE behaviour

Fixes: #36185

CMake >=3.20 now supports relative paths in Ninja depfiles.

As Zephyr still supports the use of CMake 3.13, and the current use of
DEPFILE in linker/ld/target.cmake is already absolute path, then the
safest choice is to use old behavior.
This can be changed when Zephyr increases minimum required CMake to be
3.20 or newer.
closed 2021-06-16 by carlescufi