zephyrproject-rtos/zephyr #36211: usb: stm32: not implemented functionality

JonathanVanWin, 2021-06-13:
**USB subsys functions use unimplemented functions in stm32 USB driver**
There are three functions that are currently not implemented in the stm32
USB driver (from the LOG's that I can see):

- **usb_dc_ep_flush**
- **usb_dc_detach**
- **usb_dc_reset**

The problem is that when you use functions such as usb_deconfig (from the USB subsys), they call the USB driver
and those will print that they are not implemented.

**Finish the USB stm32 driver implementation**
As far as I know, the other USB drivers implemented all the necessary behaviour, used by the subsys and by the driver itself,
and by looking at their own implementations, it seems that the detach and reset functions are pretty trivial.

**Started implementing those functions**
I took the ideas from the other drivers and understood that the detach needs to update the attached state (to check if attached before calling other functions), disable USB IRQs, and soft disconnect (some also disable clocks).
I tried doing this, by writing directly to the **USB_BASE_ADDRESS** + 0x804, and enable BIT(1), that is the soft disconnect;
and also disabled this bit at the end of the usb_dc_stm32_init function (tried before and after the call to irq_enable).
The problem is that the first time you call usb_enable, it works, then you call usb_disable, and it also works (using my changes),
but it seems that after that I cannot enable it again.
The reset function looks that is to clear the data struct and reset the peripheral, but didn't try it.

erwango, 2021-06-14:
@JonathanVanWin Thanks for having had a try on this.
Would you mind sharing your code as a draft PR and I can try to get someone looking into it ?

FRASTM, 2023-02-21:
usb_dc_ep_flush : to be implemented targetting release 3.4.0
usb_dc_detach : exists in drivers/usb/device/usb_dc_stm32.c
usb_dc_reset : to be implemented with reset register,  targetting release 3.4.0

FRASTM, 2023-02-21:
usb_dc_reset : 
is toggling the reset register enough to reset the peripheral:
```
	LL_APB1_GRP1_ForceReset(LL_APB1_GRP1_PERIPH_USB);
	LL_APB1_GRP1_ReleaseReset(LL_APB1_GRP1_PERIPH_USB);
```

erwango, 2023-10-12:
Only `usb_dc_reset` remains to be implemented;
According to API, this is stlightly unclear what operations should be performed : 
```
/**
 * @brief Reset the USB device
 *
 * This function returns the USB device and firmware back to it's initial state.
 * N.B. the USB PLL is handled by the usb_detach function
```

@jfischer-no Can you clarify the expectation here?
Would it be ok to put the device back to a state similar to a post-`usb_dc_stm32_init` ?

jfischer-no, 2023-10-25:
> Only `usb_dc_reset` remains to be implemented; According to API, this is stlightly unclear what operations should be performed :
>
> ```
> /**
>  * @brief Reset the USB device
>  *
>  * This function returns [...]

The only place it is called is https://github.com/zephyrproject-rtos/zephyr/blob/e364f29bffdfb122fead4b7dad082428f2f6969c/subsys/usb/device/usb_device.c#L1322-L1345
The description leaves a lot of room for interpretation, but I think the most important thing is to put the device in a state from which it can be made operational again using (usb_set_config()) usb_enable().

erwango, 2023-10-26:
> The description leaves a lot of room for interpretation, but I think the most important thing is to put the device in a state from which it can be made operational again using (usb_set_config()) usb_enable().

Thanks, much clearer.

mathieuchopstm, 2025-10-01:
Closing this issue as the USB DC driver is now legacy/obsolete and no new features will be accepted. Please open a new issue if similar issues are present on UDC driver.

closed 2025-10-01 by mathieuchopstm